letscontrolit/ESPEasy #2978: ESP watchdog reboots every 30 sec if syslog is enabled and no WiFi connection could be established

vortigont, 2020-03-30:
esp8266 reboots every 30 seconds if the following conditions are met:
 - logging to syslog server has been configured
 - WiFi client mode with credentials has been set
 - ESP has been powered-on and configured WiFi AP can not be found

I've hit this bug with one espeasy relays continuosly clicking every 30 sec if powered on when my WiFI router was offline. With try and errors found that syslog server IP configuration leads to this bug.

Was able to reproduce it on several boards.

### System configuration
Hardware:
blank Wemos D1 board

ESP Easy version: 
ESP_Easy_mega-20200305_normal_ESP8266_4M1M.bin  

ESP Easy settings/screenshots: 
blank config, only WiFi credentials and syslog server IP configured

ESP serial logging output follows:

```
INIT : Booting version: mega-20200305 (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
72 : Info  : INIT : Free RAM:32552
73 : Info  : INIT : Warm boot #9 Last Task: Background Task Last systime: 6347 - Restart Reason: External System
75 : Info  : FS   : Mounting...
101 : Info  : FS   : Mount successful, used 75802 bytes of 957314
499 : Info  : CRC  : program checksum       ...OK
536 : Info  : CRC  : SecuritySettings CRC   ...OK
647 : Info  : INIT : Free RAM:29336
650 : Info  : INIT : I2C
651 : Info  : INIT : SPI not enabled
739 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
845 : Info  : WIFI : Set WiFi to STA
949 : Info  : WIFI : Connecting EMi9T attempt #0
954 : Info  : Time set to 6347.000 Time adjusted by 6346047.00 msec. Wander: 1762.79 msec/second
957 : Info  : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
960 : Info  : Local time: 1970-01-01 01:45:47
1066 : Info  : WIFI : Connecting EMi9T attempt #1
2595 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 25088 WiFiStatus 6
4062 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2852 ms
4180 : Info  : WIFI : Connecting EMi9T attempt #2
4301 : Info  : WIFI : Connecting EMi9T attempt #3
7291 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2852 ms
7408 : Info  : WIFI : Connecting EMi9T attempt #4
7528 : Info  : WIFI : Connecting EMi9T attempt #5
10504 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2853 ms
10621 : Info  : WIFI : Connecting EMi9T attempt #6
10741 : Info  : WIFI : Connecting EMi9T attempt #7
13719 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2853 ms
13836 : Info  : WIFI : Connecting EMi9T attempt #8
13956 : Info  : WIFI : Connecting EMi9T attempt #9
16935 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2854 ms
17052 : Info  : WIFI : Connecting EMi9T attempt #10
17172 : Info  : WIFI : Connecting EMi9T attempt #11
20160 : Info  : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2854 ms
20277 : Info  : WIFI : Connecting EMi9T attempt #12
20397 : Info  : WIFI : Connecting EMi9T attempt #13
20744 : Info  : WIFI : Set WiFi to AP+STA
20845 : Info  : WIFI : Set WiFi to AP
20945 : Info  : WIFI : Set WiFi to AP
21046 : Info  : WIFI : Set WiFi to AP
21146 : Info  : WIFI : Set WiFi to AP
21246 : Info  : WIFI : Set WiFi to AP
21347 : Info  : WIFI : Set WiFi to AP
21447 : Info  : WIFI : Set WiFi to AP
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
U72 : Info  :

INIT : Booting version: mega-20200305 (ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)

```

kretzp, 2020-04-26:
I could confirm this. I had this issue the last days, too. And I didn't know why. Now I kow. Perhaps, I will take a look in the source code and try to fix it.

TD-er, 2020-06-28:
I just merged a commit that may reduce the needed amount of stack memory for sending out syslog log entries.
Hopefully that also fixes these issues.

Please let me know if it does fix these issues.

vortigont, 2020-06-30:
@TD-er unfortunately it does not :( Still same issue, same conditions, fresh build with 2da08358938

21354 : Info  : WIFI : Set WiFi to AP
21454 : Info  : WIFI : Set WiFi to AP
Exception (28):
epc1=0x4027f1d5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000020 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4027f1d5 in Print::write(char) at ??:?

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

Stack trace below

0x402734c9 in String::operator=(char const*) at ??:?
0x4022e641 in process_serialWriteBuffer() at ??:?
0x40235a33 in addToSerialBuffer(char const*) at ??:?
0x40100c26 in millis at ??:?
0x40238a3b in addToLog(unsigned char, char const*) at ??:?
0x40273784 in String::concat(char const*, unsigned int) at ??:?
0x40273728 in String::String(__FlashStringHelper const*) at ??:?
0x402cbc48 in chip_v6_unset_chanfreq at ??:?
0x40238af0 in addToLog(unsigned char, String const&) at ??:?
0x402627cb in setWifiMode(WiFiMode) at ??:?
0x402a6f4c in cont_ret at cont.S.o:?
0x4020974c in ESP8266WiFiGenericClass::getMode() at ??:?
0x40262932 in setAP(bool) at ??:?
0x402636cd in WiFiConnected() at ??:?
0x402775fd in uart_write_char at ??:?
0x40261f50 in NetworkConnected() at ??:?

TD-er, 2020-06-30:
Thanks for the decoded stack :)
That's helpful.

vortigont, 2021-01-27:
Looks like it has been fixed somewhere :)
Can't reproduce with recent head  4747dd81a

closed 2021-01-27 by vortigont

TD-er, 2021-01-27:
Yep this is indeed fixed.
There have been several changes in the log handling, for example when trying to send log strings from flash.
Also there was some kind of loop possible where setting the wifi mode also could trigger a log which could not be sent to the syslog server.

vortigont, 2021-01-27:
Looks like it's time for an upgrade and enabling syslog al last!
Well done, thanks a lot @TD-er !