letscontrolit/ESPEasy #2980: Rule-set change not working - JavaScript error

psi-4ward, 2020-03-31:
When I select "Rule set 2" from the dropdown on the _old engine_ I get the JS Error:

```
rules:4 Uncaught ReferenceError: wide is not defined
    at HTMLSelectElement.onchange (rules:4)
```

seems that class and onchange are interchanged:

```html
<select class='return dept_onchange(frmselect)' name='set' id='set' onchange='wide'>
  <option value=1 selected>Rules Set 1</option>
  <option value=2>Rules Set 2</option>
  <option value=3>Rules Set 3</option>
  <option value=4>Rules Set 4</option>
</select>
```

Flashed: `ESP_Easy_mega-20200328_normal_ESP8266_4M1M.bin`

TD-er, 2020-03-31:
The last commits (not yet included in a nightly build) do fix this issue.
So I will close this one, as it already has been fixed.

closed 2020-03-31 by TD-er

ghtester, 2020-04-13:
Well, there's a strange issue still - can't access Rules Set 2-4. The logon window appears instead and even though the correct username and password is selected, the logon window reappears until I navigate to another ESP_Easy web page and authenticate there succesfully. Tested with Chromium and Firefox with the same result on my custom build:

Build:⋄ | 20106  - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 5511180c, NONOS SDK 2.2.2-dev(bb83b9b), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Apr 11 2020 10:05:33
Plugin Count:⋄ | 36 [Normal]
Build Md5: | bc161e1b8b7984d07379d96b34972be5
Md5 check: | passed.
Build Time:⋄ | Apr 11 2020 10:07:20
Binary Filename:⋄ | ESP_Easy_20200411_vagrant_custom_beta_ESP8266_4M1M.bin