letscontrolit/ESPEasy #2981: No online COM ports found

user-hash, 2020-03-31:
Hey, today I have normaly flashed few D1 minis and everything went ok.
I then tried to flash another module, but I got a messege that No online COM ports was found.
Its strange, because ports are visible in arduino IDE and device manager, I just cant find them in ESP flasher tool. 
The tool was working today, So I have no clue what is going on. I have tried to restarted my pc, but no help.
Any ideas?

1. Flashed few modules with normal bin, everything OK
2. Next module I try to flash I get No online COM ports found error, without any other change
3. I can see the open ports in Arduino IDE in device manager and I can flash this same device in Ardiuno IDE with blink sketch

TD-er, 2020-03-31:
Try plugging the module in another USB port?
@Grovkillen Maybe this is something for you to look into?

I also have a module which has a CP210x USB to serial on it, that never would like to work with the ESPEasy Flasher.
Problem with that specific unit is, it is my ESP12F flasher so I use it to flash blank ESP12F/E/S modules before soldering them on the PCB. :(

Maybe Windows has changed something here? Or these are 'fake' USB to serial chips and they have again done something to the drivers of those?

user-hash, 2020-03-31:
hmm, I found the problem. The module I had problems with flashing is a D1 mini pro.
I than switched to another ordinary D1 mini and I can flash It normaly, as COM port is found.
I will close this issue now

closed 2020-03-31 by TD-er