letscontrolit/ESPEasy #2983: [new req] add ticker box to activate/deactivate each rule

tparvais, 2020-04-03:
Idea of evolution:
add a thicker box close to each rule to activate or deactivate the execution of it.

It's very useful when developing to perform tests etc... and should not be complex

TD-er, 2020-04-03:
It has been requested a few times before.
And indeed it may be a useful feature.

uzi18, 2020-04-03:
This could be possible by check if line ends with something special like * or - sign, other idea is to store list of file number+line number

Budman1758, 2020-04-03:
https://github.com/letscontrolit/ESPEasy/issues/659

TD-er, 2020-04-04:
Wel we have a lot of options here.
- Store in the settings the flag to ignore a rules file
- rename the file to `_rules1.txt` to disable (not sure how well that will work on 1M modules)
- Place marker around some `on ... do` blocks (e.g. `/*` and `*/` )
- something else.

Personally I do like the block comment idea as it is more flexible

tparvais, 2020-04-04:
Personnally I prefer the flag less risk of error and quicker to use

TD-er, 2020-04-04:
Well these options are not all XOR