letscontrolit/ESPEasy #2984: Compile error...  Failed to recurse into submodule path 'tools/sdk/lwip2/builder'

cmorrow101, 2020-04-03:
Release mega-20200328
I cannot complie the do to an error 
Fetched in submodule path 'tools/sdk/lwip2/builder/lwip2-src', but it did not contain 159e31b689577dbf69cf0683bbaffbd71fa5ee10. **Direct fetching of that commit failed.**

Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', '2.6.3', 'https://github.com/esp8266/Arduino.git', 'C:\\Users\\Cottage-Proto2\\.platf_installing-r_uu61dx-package']

Someone had a similar issue which was resolved by updating to the latest git, but that did not solve my problem. I have not edited the release in any way, just tried to complie "out of the box", with no luck. I tried to clone the framework-arduinoespressif8266 repository and place it in the platformio build directory but it seems the complier want to download a version anyway. I don't know if there is a way to have the compiler use the copy I cloned. 

I am using visual studio code with platformIO, I have also tried Atom with platformIO and get the same error on both. 

I could really use some help here. I have spent 2 days trying to sort this out with no success.

uzi18, 2020-04-03:
@cmorrow101 @TD-er confirmed - have same problem

uzi18, 2020-04-03:
```
$ platformio run -e normal_ESP8266_1M
*********************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
*********************************************************************************************************************************************************************************************************************************

Processing normal_ESP8266_1M (platform: espressif8266@2.4.0; board: esp01_1m; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing espressif8266 @ 2.4.0
espressif8266 @ 2.4.0 has been successfully installed!
The platform 'espressif8266@2.4.0' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.40802.191122 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266
git version 2.26.0
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package'...
remote: Enumerating objects: 1647, done.
remote: Counting objects: 100% (1647/1647), done.
remote: Compressing objects: 100% (1401/1401), done.
remote: Total 1647 (delta 201), reused 841 (delta 78), pack-reused 0
Receiving objects: 100% (1647/1647), 23.62 MiB | 291.00 KiB/s, done.
Resolving deltas: 100% (201/201), done.
Note: switching to '3d128e5c785cbe2096a0def394554d1d8091601d'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.                                                                                                                                                              
                                                                                                                                                                                                                                 
If you want to create a new branch to retain commits you create, you may                                                                                                                                                         
do so (now or later) by using -c with the switch command. Example:                                                                                                                                                               
                                                                                                                                                                                                                                 
  git switch -c <new-branch-name>                                                                                                                                                                                                
                                                                                                                                                                                                                                 
Or undo this operation with:                                                                                                                                                                                                     
                                                                                                                                                                                                                                 
  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Submodule 'libraries/ESP8266SdFat' (https://github.com/earlephilhower/ESP8266SdFat.git) registered for path 'libraries/ESP8266SdFat'
Submodule 'libraries/LittleFS/lib/littlefs' (https://github.com/ARMmbed/littlefs.git) registered for path 'libraries/LittleFS/lib/littlefs'
Submodule 'libraries/SoftwareSerial' (https://github.com/plerup/espsoftwareserial.git) registered for path 'libraries/SoftwareSerial'
Submodule 'tools/esptool' (https://github.com/espressif/esptool.git) registered for path 'tools/esptool'
Submodule 'tools/pyserial' (https://github.com/pyserial/pyserial.git) registered for path 'tools/pyserial'
Submodule 'lwip2' (https://github.com/d-a-v/esp82xx-nonos-linklayer.git) registered for path 'tools/sdk/lwip2/builder'
Submodule 'tools/sdk/ssl/bearssl' (https://github.com/earlephilhower/bearssl-esp8266.git) registered for path 'tools/sdk/ssl/bearssl'
Cloning into '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package/libraries/ESP8266SdFat'...
remote: Enumerating objects: 3534, done.        
[... cut ...]
Fetched in submodule path 'tools/sdk/lwip2/builder/lwip2-src', but it did not contain 159e31b689577dbf69cf0683bbaffbd71fa5ee10. Direct fetching of that commit failed.
Submodule path 'tools/sdk/ssl/bearssl': checked out '89454af34e3e61ddfc9837f3da5a0bc8ed44c3aa'
Failed to recurse into submodule path 'tools/sdk/lwip2/builder'
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', '2.6.3', 'https://github.com/esp8266/Arduino.git', '/home/users/linaja/.platformio/packages/_tmp_installing-bgoetgw6-package']

```
`.platformio` deleted - without help

TD-er, 2020-04-04:
OK, will look into it.
Maybe it refers to a SHA or tag that no longer exists?

cmorrow101, 2020-04-04:
I successfully installed the arduino framework with the Git for Windows 2.16.2. Git version 2.26.0 did not work.

closed 2020-04-04 by cmorrow101

TD-er, 2020-04-04:
I use `git version 2.24.1.windows.2` and that's working fine also.

uzi18, 2020-04-04:
@TD-er @cmorrow101 thx for hints, checked several versions and git 2.25.2 is last working.
maybe we should report it into core

TD-er, 2020-04-05:
OK, so it seems a bug in Git itself?

uzi18, 2020-04-05:
@TD-er yes, so for now avoid new version :)

TD-er, 2020-04-06:
> @TD-er yes, so for now avoid new version :)

Have you made a bug report at the Git repository?

cmorrow101, 2020-04-06:
I have not made a bug report.

arkhnchul, 2020-05-02:
mess with submodules was fixed in 2.7.0 version of esp8266 arduino platform, so building test_beta_ESP8266_4M1M env works fine with newer git.