letscontrolit/ESPEasy #2985: WifiConnect command doesn't work

tiomny, 2020-04-03:
**Steps to reproduce:**

1. Set WiFi key or WiFi AP to connect to as incorrect so device isn't able to connect to any AP;
2. Reboot device;
3. Run following commands via terminal or web (tools?cmd=<command>):
`WifiAPKey,<wifiapkey>`
`WifiKey,<wifikey>`
`Save`
`WifiConnect`

**Actual result:**
Device saves _WifiAPKey_ and _WifiKey_ but the `WifiConnect` command doesn't make device connect to saved AP.
After running the `Reboot` command device connects to _WifiAPKey_ successfully.

Expected result:
Device should connect to _WifiAPKey_ when running the `WifiConnect` command

TD-er, 2020-04-04:
Do you have any special characters in the ssid or key?
(special as in other than A-Za-z0-9)

tiomny, 2020-04-04:
No special characters.

TD-er, 2020-04-04:
Is the "other" key also still present?
The problem may be that a scan for AP's can sometimes miss an AP.
So maybe I should perform a new scan if the specified SSID's are not found.

The scan now uses 200 msec per channel, based on the idea that a beacon interval usually takes 102.4 msec. But that's not a fixed value. Each manufacturer of access points may use a different value and if there is a lot of activity it may miss an interval.
So either I should run a new scan if it was not found or make this beacon interval configurable.
A lot of other stuff is based on this assumed interval.

tiomny, 2020-04-04:
Fallback SSID isn't configured.

TD-er, 2020-04-12:
I was looking into the current WiFi related issues again and re-read your start post.
I notice you are using the `WifiAPKey` command, but I wonder why.

This command sets the password for the node when it is running in AP mode.
See the [Command reference](https://espeasy.readthedocs.io/en/latest/Reference/Command.html#commands-listed)

So you need to use `WiFiSSID` to set the SSID and `WiFiKey` to set the password. 

If you meant to use `WiFiAPKey`, please explain why as I don't understand what you are trying to achieve.

tiomny, 2020-04-12:
Sorry, I meant WiFiKey.
This is the code from my app:
```
  List<String> cmdList = [
    'WifiSSID,$connectToApSsid',
    'WifiKey,$connectToApPassword',
    'Save',
    'WifiConnect'
  ];
```

TD-er, 2020-04-12:
Maybe you need to have a little delay between the commands?
How do you send the commands?
Via HTTP GET, or the serial port?

tiomny, 2020-04-12:
May I? I can try if it would help.
I use HTTP GET.

TD-er, 2020-04-12:
How much time is there between calling the commands via HTTP GET?
I think one or more is just missed on the ESP as it takes time to process them.

tiomny, 2020-04-12:
Commands are sent to node one by one:
```
  List<String> cmdList = [
    'WifiSSID,$connectToApSsid',
    'WifiKey,$connectToApPassword',
    'Save',
    'WifiConnect'
  ];

    for (String cmd in cmdList) {
      var response = await client.get(cmdAddress + cmd);
      if (response.statusCode != 200) return false;
    }
```
Serial monitor in Platformio displays all commands proceed.

TD-er, 2020-04-12:
```c++
void WiFiConnectRelaxed() {
  if (!wifiConnectAttemptNeeded) {
    return; // already connected or connect attempt in progress need to disconnect first
  }
  if (!processedScanDone) {
    // Scan is still active, so do not yet connect.
    return;
  }

  // Start connect attempt now, so no longer needed to attempt new connection.
  wifiConnectAttemptNeeded = false;

  if (!prepareWiFi()) {
    addLog(LOG_LEVEL_ERROR, F("WIFI : Could not prepare WiFi!"));
    last_wifi_connect_attempt_moment = millis();
    wifi_connect_attempt             = 1;
    return;
  }

  if (wifiSetupConnect) {
    // wifiSetupConnect is when run from the setup page.
    RTC.lastWiFiSettingsIndex     = 0; // Force to load the first settings.
    RTC.lastWiFiChannel = 0; // Force slow connect
    wifi_connect_attempt = 0;
    wifiSetupConnect     = false;
  }

  // Switch between WiFi credentials
  if ((wifi_connect_attempt != 0) && ((wifi_connect_attempt % 2) == 0)) {
    if (selectNextWiFiSettings()) {
      // Switch WiFi settings, so the last known BSSID cannot be used for a quick reconnect.
      RTC.lastBSSID[0] = 0;
    }
  }
  const char *ssid       = getLastWiFiSettingsSSID();
  const char *passphrase = getLastWiFiSettingsPassphrase();

  if (loglevelActiveFor(LOG_LEVEL_INFO)) {
    String log = F("WIFI : Connecting ");
    log += ssid;
    log += F(" attempt #");
    log += wifi_connect_attempt;
    addLog(LOG_LEVEL_INFO, log);
  }
  last_wifi_connect_attempt_moment = millis();
  wifiConnectInProgress            = true;

  // First try quick reconnect using last known BSSID and channel.
  bool useQuickConnect = RTC.lastBSSID[0] != 0 && RTC.lastWiFiChannel != 0 && wifi_connect_attempt < 3;

  if (useQuickConnect) {
    WiFi.begin(ssid, passphrase, RTC.lastWiFiChannel, &RTC.lastBSSID[0]);
  } else {
    WiFi.begin(ssid, passphrase);
  }
  ++wifi_connect_attempt;
  logConnectionStatus();
}
```
Well I guess it will not actually perform the `WiFiConnect` command as the value for `wifiConnectAttemptNeeded` is not true.
So maybe we should not actually call this function, but just set this flag to true.
Then it will try to reconnect in the background.

tiomny, 2020-04-12:
I ran this command from the serial monitor manually, but it didn't make any affect.
Besides, I recalled that I ran all these commands manually one by one with some delay but without success as well.

TD-er, 2020-04-12:
I created a pull request: https://github.com/letscontrolit/ESPEasy/pull/3011
Can you compile it yourself to test?
Or do you need a test build from me? If so, what build do you need?

tiomny, 2020-04-12:
Sure, I will test.

PR #3011 (2020-04-12): [WiFi] Fix WiFiConnect command should force (re)connect (#2985)
Fixes: #2985
commit 64ed9f7c (2020-04-12): [WiFi] Fix WiFiConnect command should force (re)connect (#2985)

Fixes: #2985
closed 2020-04-12 by TD-er