letscontrolit/ESPEasy #2986: mDNS doesn't work

tiomny, 2020-04-03:
**System info:**
"Build":20105,
"Git Build":"Plantware: Apr  3 2020 15:31:05",
"System Libraries":"ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support",
"Plugin Count":4,
"Plugin Description":"[Normal]",

**Custom.h contains following line:**
`#define FEATURE_MDNS`

**System log:**
```
717 : Info  : WIFI : Connecting HobbitHole attempt #0
722 : Info  : Webserver: start
723 : Info  : WIFI : Starting mDNS...
724 : Info  : WIFI : mDNS failed
```

TD-er, 2020-04-04:
Hmm that's an exceptionally fast wifi connect. (700 msec from boot)
What is the secret here to make such a quick wifi connect?

tiomny, 2020-04-04:
I didn't perform any special actions for this - perhaps, small amount of plugins makes it connect so fast?

TD-er, 2020-04-04:
The last known BSSID + channel is stored in RTC memory, so at a warm boot it may perform a much faster reconnect.
I have not seen it below 900 msec at my tests, usually around 1.2 sec.
After this connect it also must wait for a DHCP result, which typically takes at least 20 msec, but often a multiple of 100 msec.
So it may be possible the mDNS is started before things are fully initialized.

tonhuisman, 2023-09-02:
This seems to be outdated, can be closed.

TD-er, 2023-09-02:
Close this as mDNS does now work....
However it has still been disabled for ESP8266 since it does have a memory leak which we can't fix in ESPEasy.

closed 2023-09-02 by TD-er