letscontrolit/ESPEasy #2988: my sonoff reboot randomly

ccostaccosta, 2020-04-06:
Hello team,

For some time now, I have my shutter button in MEGA which reboots randomly. This action makes me lower my shutter ...
I flashed my button to zero, still the same.

I have follow this site /
https://www.monjeedom.fr/domotiser-ses-volets-roulant-electrique-avec-un-sonoff-t1/#comment-151

Version of my sonoff: mega-20191003

I specify that the sonoff reboot by itself

some log :
152236: WD : Uptime 3 ConnectFailures 33 FreeMem 22320 WiFiStatus 3
182236: WD : Uptime 3 ConnectFailures 33 FreeMem 22320 WiFiStatus 3
212236: WD : Uptime 4 ConnectFailures 33 FreeMem 22288 WiFiStatus 3
Failed to fetch <<
22686: EVENT: Rules#Timer=1
22732: ACT : gpio,5,0
22736: SW : GPIO 5 Set to 0
22739: ACT : gpio,12,0
22743: SW : GPIO 12 Set to 0
22775: SW : GPIO=5 State=0 Output value=0
22778: EVENT: volet-D#Switch-D=0.00
22889: HTTP : C008 connection failed (23/0)
23012: HTTP : C008 connection failed (24/0)
23132: HTTP : C008 connection failed (25/0)
23254: HTTP : C008 connection failed (26/0)
23374: HTTP : C008 connection failed (27/0)
23500: HTTP : C008 connection failed (28/0)
23620: HTTP : C008 connection failed (29/0)
23743: HTTP : C008 connection failed (30/0)
23965: HTTP : C008 connection failed (31/0)
24085: HTTP : C008 connection failed (32/0)
24207: HTTP : C008 connection failed (33/0)
32236: WD : Uptime 1 ConnectFailures 33 FreeMem 22216 WiFiStatus 3

Any Ideas ?
Thanks team !

TD-er, 2020-04-06:
Some ideas:
- Try a newer build
- Check WiFi setup (fixed channel, Force B/G, no WiFi sleep, check RSSI value)
- Check controller settings (lower number of messages in queue, check timeout, maybe check wait for reply)

In the screenshot on the page you linked, GPIO-9 is used.
This can only be used on ESP8285 based units, not on ESP8266 as that GPIO is used for the (external) flash.
So what is the complete setup you're using?

ccostaccosta, 2020-04-08:
Hi,

I put the new version to the ESP8285. And I watch what's going on.

Thanks for your time !!!

ccostaccosta, 2020-04-12:
untill now it's work.
thanks for your time

TD-er, 2020-04-12:
You're welcome :)

closed 2020-04-12 by TD-er