letscontrolit/ESPEasy #2990: Rules#Timer is fired one second before the correct time

giig1967g, 2020-04-07:
Hi,
I noticed that one of my rules wasn't working properly and discovered that the reason was that the Rules#Timer event was firing one second earlier than expected. See log:
```
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: EVENT: Clock#Time=Tue,05:57
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: ACT  : asyncEvent,sceneSet
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: Command: asyncEvent
```

Is this a wanted behavior, or it's a bug?

TD-er, 2020-04-08:
Looks like you send the logs to an external logserver.
Make sure that server is properly NTP synchronized before we can draw conclusions :)

What may be happening here is that the internal clock of the ESP is not really accurate.
It only queries the set NTP server every N minutes (60 I think) and if that fails for some reason it may be longer.
Typically the drift I see is in the order of 0.05 msec/sec, but I guess on some modules it may be worse.
With a drift of 0.05 msec/sec, the internal clock can be off by one second after roughly 5.5 hours.

giig1967g, 2020-04-08:
I have checked and my server is NTP synchronized, even if if difficult to tell looking at both if they are perfectly in synch.

Nevertheless, these are my rules:
At certain times, I run the general event called `sceneSet`. Inside the event `sceneSet` I recheck the time and react accordingly. But as the `Clock#Time` fired one second earlier, the scene selected inside `sceneSet` is not the correct scene.

Example: assume `%sunrise-1h%` is 5:57. It means that at 5:57, the scene selected inside `sceneSet` should be `S_OFF_ALL`. But because  the Clock#Time fired at 5:56:59 the scene selected was `S_OFF_ExceptRampa` because the condition `%systm_hm%<%sunrise%` was TRUE (i.e. 5:56 < 5:57).
And infact I am getting wrong scene selection.

```
On Clock#Time=All,%sunset% do
  asyncEvent,sceneSet
endon

On Clock#Time=All,%sunset+1h% do
  asyncEvent,sceneSet
endon
 
On Clock#Time=All,23:59 do
  asyncEvent,sceneSet
endon

On Clock#Time=All,2:00 do
  asyncEvent,sceneSet
endon

On Clock#Time=All,%sunrise-1h% do
  asyncEvent,sceneSet
endon

on sceneSet do
 if %systm_hm%>=%sunset+1h%
  asyncEvent,S_ON_GardenFull
 elseif %systm_hm%>=%sunset%
  asyncEvent,S_ON_pool
 elseif %systm_hm%<%sunrise-1h%
  asyncEvent,S_OFF_ExceptRampa
 else
  asyncEvent,S_OFF_ALL
 endif	
endon

```
And looking at the extended log you will see the behavior. At 5:57 (=%sunrise-1h%) the scene selected is not `S_OFF_ALL` as it should be, but it's `S_OFF_ExceptRampa`:
```
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: EVENT: Clock#Time=Tue,05:57
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: ACT  : asyncEvent,sceneSet
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: Command: asyncEvent
2020-04-07 05:56:59	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: EVENT: sceneSet
2020-04-07 05:57:00	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: ACT  : asyncEvent,S_OFF_ExceptRampa
2020-04-07 05:57:00	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: Command: asyncEvent
2020-04-07 05:57:00	Kernel.Notice	10.8.0.2	ESPT2 EspEasy: EVENT: S_OFF_ExceptRampa

```