letscontrolit/ESPEasy #2991: Feature request: display %syshour% %sysmin% %syssec% as integers with leading 0

bccrew, 2020-04-07:
Hello,

I like to have my LCD2004 in dutch language and for time annotation we like to use "15u00" for 3PM.

According to the documentation https://www.letscontrolit.com/wiki/inde ... _Variables the notation should be like this:

%syshour% Current hour (hh)
%sysmin% Current minute (mm)

When I use the following line in the LCD2004 config:

%sysweekday_s% %sysday%/%sysmonth%/%sysyear% %syshour%u%sysmin%

the notation is 0u1 instead of 00u01 what I'm expecting.

Suggestion: display hours, minutes, and seconds under value of 10 as 01, 02 , 03, 04 etc.

CFR: https://www.letscontrolit.com/forum/viewtopic.php?p=43729#p43729

TD-er, 2020-04-08:
The variables you want, are already present as can be seen on the system variables page:

%sysyear%  // %sysyear_0% | 2020 // 2020 | 2020 // 2020
-- | -- | --
%sysyears% | 20 | 20
%sysmonth% // %sysmonth_0% | 4 // 04 | 4 // 04
%sysday%   // %sysday_0% | 8 // 08 | 8 // 08
%syshour%  // %syshour_0% | 9 // 09 | 9 // 09
%sysmin%   // %sysmin_0% | 3 // 03 | 3 // 03
%syssec%   // %syssec_0% | 27 // 27 | 27 // 27
%syssec_d% | 32607 | 32607
%sysweekday% | 4 | 4
%sysweekday_s%cv | Wed | Wed

For example `%syssec_0%`

bccrew, 2020-04-08:
Hi TD-er,

Ok, so just the documentation needs to be updated then. Thanks!

https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

thomastech, 2020-04-08:
_Ok, so just the documentation needs to be updated then._

[x] Done. 
I updated the wiki to include the missing system vars. 

BTW, the %sysyear_0% var is amusing. Would have been useful a few centuries ago.

- Thomas

bccrew, 2020-04-08:
OK, thanks Thomas!

closed 2020-04-08 by bccrew

TD-er, 2020-04-08:
> BTW, the %sysyear_0% var is amusing. Would have been useful a few centuries ago.

Well we plan on a long running project :)
At least it is more useful than the code present at one of my previous jobs, where there was actual code for milli bits, micro bits and nano bits.

uzi18, 2020-04-08:
> milli bits, micro bits and nano bits

how to add 10micro bits to 100 nano bits ? :)

TD-er, 2020-04-08:
> > milli bits, micro bits and nano bits
> 
> how to add 10micro bits to 100 nano bits ? :)

Or even better, multiply them :)