letscontrolit/ESPEasy #2992: Plugin P145 won't compile

piano-thomas, 2020-04-07:
Hi All,

I'm trying to include the P145 ITHO plugin in the latest release of ESP Easy Mega, but I cannot get it working. I tried compiling using Arduino IDE and Atom/PlatformIO, but I get a lot of errors that I can't solve. Is there anyone who could give it a try and see if it could be succesfully done?

I'm trying to compile the build normal_ESP8266_4M1M for a NodeMCU V3.

Hope you guys can help me out!

Best,

Thomas

TD-er, 2020-04-08:
Can you post the build errors?

piano-thomas, 2020-04-08:
Hi TD-er,

Here is a screenshot of the errors I see when the compiling process stops:
What I did:
- define_plugin_sets.h, I added plugin P145 to the list of normal/stable plugins
- ESPEasy-Globals.h, I kept the #define PLUGIN_BUILD_NORMAL command active, and did not change

[image]

Is this info sufficient?

Thanks for helping out!

TD-er, 2020-04-08:
All `addForm...` functions now no longer need to have the `string` variable.
So if you remove the first parameter (`string`) from the function calls, you will at least have those issues fixed.
Not sure what's there to do with the warning about ignoring the typedef.

piano-thomas, 2020-04-08:
Thanks! That fixes most of it, except the last one about 'WebServer' was not declared in this scope. 
Maybe the plugin is a bit outdated compared to the latest release of ESPEasy?

piano-thomas, 2020-04-08:
I disabled a few lines from the plugin code regarding this webserver, and got it working now. NO idea what happened, but at least it works! Thanks for your help @TD-er!

closed 2020-04-08 by piano-thomas

TD-er, 2020-04-08:
Replace `WebServer` with `web_server`
I was renamed recently because of a name conflict with ESP32, which has a class with the same name