letscontrolit/ESPEasy #2993: Generic HTTP Advanced causes __yield() panic

pkoster, 2020-04-07:
Pushing sensor data to Influxdb using the Generic HTTP Advanced Controller causes constant reboots. The functionality works, but every now and then, typically in the order of minutes apart, an error occurs:
> User exception (panic/abort/assert)
> Panic core_esp8266_main.cpp:121 __yield
> >>>stack>>>
> ctx: cont

Basic steps to reproduce:
* dummy device (e.g. every 15s, similar results with other sensor devices)
* generic http advanced controller (user + password + POST + write?db=test + Content-Type: application/x-www-form-urlencoded + dummy val=%val1% )

Version: ESP_Easy_mega-20200328_test_ESP8266_4M1M_VCC.bin 
(also tried ESP_Easy_mega-20200328_dev_ESP8266_4M1M.bin )
Hardware: nodemcu v3

- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [X] Tested on other node to make sure hardware isn't defective.
- [X] Verified if the problem is limited to a single plugin/controller

TD-er, 2020-04-08:
Can you also try a build with _beta_ in the name?
That one uses the latest esp8266/Arduino core.

pkoster, 2020-04-08:
Done, but problem persists
Version file: ESP_Easy_mega-20200328_test_beta_ESP8266_4M1M.bin

With respect to reproduce reboots seem to occur much faster with a larger HTTP BODY like 'dummy state=6,val=%val1%,uptime=%uptime%' instead of just the earlier 'dummy val=%val1%'.

TD-er, 2020-04-08:
The reason I asked is because `__yield()` panics can only happen in callback functions and as far as I know we don't use callback functions in HTTP Advanced.
What other plugins do you have active?

pkoster, 2020-04-09:
Captured serial output for the beta firmware:
```
825899 : Debug More: HTTP after replacements:
825900 : Debug More: dummy state=6,val=0.00,uptime=14
840835 : Info  : Dummy: value 1: 0.00
User exception (panic/abort/assert)
Panic core_esp8266_main.cpp:121 __yield

>>>stack>>>

ctx: sys
sp: 3fff2580 end: 3fffffb0 offset: 0000
3fff2580:  00000064 3fff2ad0 00000020 4010171f
```
Interestingly, here ctx is sys instead of cont.

The only plugins configured are:
- Generic HTTP (disabled, but problem also observed with it completely absent on blank NodeMCU)
- Generic HTTP Advanced [TESTING]
- Generic - Dummy Device (but problem also identified with just Gases - CO2 MH-Z19 or Analog Input - ADS1115)

pkoster, 2020-04-18:
Is there something that could help debugging / fixing this? Full stacktrace / log / test / ... ?

TD-er, 2020-04-18:
Well if you can give some (decoded) stack trace that would be great.
Sadly decoding stacks is quite hard in PlatformIO, so I have to make a build in ArduinoIDE in order to decode it (and let you run that code)

pkoster, 2020-04-18:
For some reason capturing a stack trace is a problem by itself: no stacktrace, stacktrace with end missing, stacktrace interrupted with another stacktrace, etc.

This was the first complete stacktrace I could make:
```
User exception (panic/abort/assert)
Panic core_esp8266_main.cpp:121 __yield

>>>stack>>>

ctx: cont
sp: 3fff2680 end: 3fff32b0 offset: 0000
3fff2680:  00000064 3fff2bd0 00000020 40101873
3fff2690:  000000fe 00000000 00000000 00000000
3fff26a0:  00000000 00000000 00000000 3fff333c
3fff26b0:  3fff7fd8 00000000 3fff81ec 00000000
3fff26c0:  0f377601 3fff2bd0 00000001 4028f336
3fff26d0:  3ffefc58 0f377602 3fff7850 4028f36f
3fff26e0:  00000029 3fff3538 3fff7850 4028ec49
3fff26f0:  0f377601 3fff2bd0 00000001 40259e00
3fff2700:  4029b2c8 0a01a8c0 3fff27a0 4028fc8e
3fff2710:  40206b0c 00000001 3fff7ca4 402843c8
3fff2720:  00000000 ffffffff 3fff3a5c 3fff27a0
3fff2730:  3fff2bd0 00000002 3fff7ca4 4028443c
3fff2740:  4029b2c8 0a01a8c0 80007461 3fff7ca4
3fff2750:  0000000b 3fff2bd0 3fff7ca4 4028182a
3fff2760:  00000000 402f1d94 3fff27a0 4028d928
3fff2770:  0f377538 0000000b 3fff27a0 4028d958
3fff2780:  402f1d94 00000400 3fff2bd0 3fff7ca4
3fff2790:  00000000 0000000b 3fff2bd0 4028188b
3fff27a0:  50545448 00203a20 87ff2bd0 4028d28d
3fff27b0:  3fff7ca4 0016001f 3fff3170 4028d2ac
3fff27c0:  00000000 3fff2bd0 3fff3170 4026a568
3fff27d0:  54534f50 00000000 00000000 00000000
3fff27e0:  74697277 62643f65 7561703d 7365746c
3fff27f0:  00000074 00000000 00000000 00000000
3fff2800:  00000000 00000000 00000000 00000000
3fff2810:  00000000 00000000 00000000 00000000
3fff2820:  00000000 00000000 00000000 00000000
3fff2830:  00000000 00000000 00000000 00000000
3fff2840:  00000000 00000000 00000000 00000000
3fff2850:  00000000 00000000 00000000 00000000
3fff2860:  00000000 00000000 00000000 00000000
3fff2870:  00000000 00000000 00000000 00000000
3fff2880:  00000000 00000000 00000000 00000000
3fff2890:  00000000 00000000 00000000 00000000
3fff28a0:  00000000 00000000 00000000 00000000
3fff28b0:  00000000 00000000 00000000 00000000
3fff28c0:  00000000 00000000 00000000 00000000
3fff28d0:  746e6f43 2d746e65 65707954 7061203a
3fff28e0:  63696c70 6f697461 2d782f6e 2d777777
3fff28f0:  6d726f66 6c72752d 6f636e65 0d646564
3fff2900:  0000000a 00000000 00000000 00000000
3fff2910:  00000000 00000000 00000000 00000000
3fff2920:  00000000 00000000 00000000 00000000
3fff2930:  00000000 00000000 00000000 00000000
3fff2940:  00000000 00000000 00000000 00000000
3fff2950:  00000000 00000000 00000000 00000000
3fff2960:  00000000 00000000 00000000 00000000
3fff2970:  00000000 00000000 00000000 00000000
3fff2980:  00000000 00000000 00000000 00000000
3fff2990:  00000000 00000000 00000000 00000000
3fff29a0:  00000000 00000000 00000000 00000000
3fff29b0:  00000000 00000000 00000000 00000000
3fff29c0:  00000000 00000000 00000000 00000000
3fff29d0:  2c326f63 61636f6c 6e6f6974 6f6f773d
3fff29e0:  6d616b6e 63207265 253d326f 316c6176
3fff29f0:  00000025 00000000 00000000 00000000
3fff2a00:  00000000 00000000 00000000 00000000
3fff2a10:  00000000 00000000 00000000 00000000
3fff2a20:  00000000 00000000 00000000 00000000
[... 217 lines cut ...]
0x4028e95c: operator new(unsigned int) at ??:?
0x4028d28d: String::invalidate() at ??:?
0x4028d2ac: String::~String() at ??:?
0x402868a3: CPluginCall(unsigned char, CPlugin::Function, EventStruct*, String&) at ??:?
0x40260177: validUserVar(EventStruct*) at ??:?
0x402f1438: chip_v6_set_sense at ??:?
0x4026024d: sendData(EventStruct*) at ??:?
0x4028d928: String::operator=(__FlashStringHelper const*) at ??:?
0x4028d28d: String::invalidate() at ??:?
0x402f1308: chip_v6_set_sense at ??:?
0x4026043e: SensorSendTask(unsigned char) at ??:?
0x40288d90: usecPassedSince(unsigned long) at ??:?
0x4022a8ac: setNewTimerAt(unsigned long, unsigned long) at ??:?
0x40287258: getDeviceIndex_from_TaskIndex(unsigned char) at ??:?
0x40260492: process_task_device_timer(unsigned long, unsigned long) at ??:?
0x4028fc8e: delay at ??:?
0x4027e73c: handle_schedule() at ??:?
0x4027e76e: handle_schedule() at ??:?
0x40257d9c: WiFiConnected() at ??:?
0x4027e867: loop at ??:?
0x40100891: esp_schedule at ??:?
0x4028ed1c: loop_wrapper() at core_esp8266_main.cpp:?
0x401007d5: cont_wrapper at ??:?
```
Based on self built test_beta_ESP8266_4M1M. Stack decoded with https://github.com/janLo/EspArduinoExceptionDecoder .

TD-er, 2020-04-18:
I don't think the first decoded stack trace is complete.
But just guessing, based on that stack trace, you could try to remove the delay in this function:
```c++
boolean ControllerSettingsStruct::checkHostReachable(bool quick) {
  if (!WiFiConnected(10)) {
    return false; // Not connected, so no use in wasting time to connect to a host.
  }
  delay(1);       // Make sure the Watchdog will not trigger a reset.

  if (quick && ipSet()) { return true; }

  if (UseDNS) {
    if (!updateIPcache()) {
      return false;
    }
  }
  return hostReachable(getIP());
}
```

uzi18, 2020-04-18:
or change to delay(0)

pkoster, 2020-04-19:
Uncommenting delay(1) or setting it to delay(0) does not solve the problem. It still reboots.

tonhuisman, 2023-09-02:
This issue is outdated, can be closed.

closed 2023-09-02 by TD-er