letscontrolit/ESPEasy #2994: PCF8574 issue when all pin settled as output

Fapibo, 2020-04-08:
Hello all,

First thanks to all developpers for this amazing esp easy SW.

# Setup:
* Git Build: | mega-20191208
* plug-in: "Switch input - PCF8574"
* HW: wemos D1 mini + PCF8574  + 8 relay board. I connected the relay board as recommended (sink current). To make sure this issue is not linked to supply, I added some 100nF capacitor on PCF8574 power supply and I unplugged all relay coils (only LEDs from relay board are connected) 
* I use all PCF8574 pins (8) as output

# Issue description:
* when I set output 6 or 8 to 0V, outputs are stuck to 0: no possible to get 3V3 anymore. I have to perform ESPeasy reset (SW or HW) to get back control.
* when I drive other outputs, it's working fine

# bash script to confirm behavior:
```
# 1-switch all to ON
curl http://192.168.31.203/control?cmd=PCFGPIO,1,0 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,2,0 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,3,0 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,4,0 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,5,0 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,6,0 && sleep 0.3 # Not good: both output 6 and 8 are switched on.
curl http://192.168.31.203/control?cmd=PCFGPIO,7,0 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,8,0 && sleep 0.3 # Not good: already ON
#2-switch all to OFF
curl http://192.168.31.203/control?cmd=PCFGPIO,1,1 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,2,1 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,3,1 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,4,1 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,5,1 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,6,1 && sleep 0.3 # still ON
curl http://192.168.31.203/control?cmd=PCFGPIO,7,1 && sleep 0.3 # OK
curl http://192.168.31.203/control?cmd=PCFGPIO,8,1 && sleep 0.3 # still ON
#3-reset
curl http://192.168.31.203/?cmd=reboot > /dev/null && sleep 10
curl http://192.168.31.203/control?cmd=PCFGPIO,1,1 && sleep 0.3 # ok: all OFF
```
My feeling is plug-in issue: (because reboot fix it). What do you think?

uzi18, 2020-04-08:
is it always same behaviour?
when turning on 6th you got also 8th?

tonhuisman, 2020-04-08:
Are you sure your hardware isn't broken?

Fapibo, 2020-04-09:
> is it always same behaviour? 
> when turning on 6th you got also 8th?
Yes but I checked there are no short between 6 and 8

> Are you sure your hardware isn't broken?
It was my first hypothesis but how to explain that ESP reboot fixes the issue? PCF8574 is connected through i2C: its behaviour shouldn't be affected by ESP SW reset? 
As soon as I perform SW reboot (as in the bash script), I am able to get pin 6 and 8 high again.

May be my plug-in configuration is wrong: device name is "Switch input - PCF8574" and I use only output! I did not see any setting to configure all pin as output...

uzi18, 2020-04-09:
If you want to control all 8 pins at once you can use this trick:
/tools?cmd=wdconfig,32,1,x
where x is value from 0-255 to send to PCF

in such case you even not need to configure tasks as switch input :)

uzi18, 2020-04-09:
@Fapibo show you "devices" configuration screenshot

Fapibo, 2020-04-09:
@uzi18 
Thanks for the trick. I can drive now pin 6 and 8 to high without reboot. Nevertheless, it seems still linked... I adapted the bash script, actual behaviour described in comments:
```
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,255 && sleep 0.3 # 0xFF, ALL OFF (OK)
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,253 && sleep 0.3 # 0xFD, pin 2 ON  (OK)
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,255 && sleep 0.3 # 0xFF, ALL OFF (OK)
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,127 && sleep 3 # 0x7F, PIN 6 and 8 on  (should be 8 only)
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,255 && sleep 0.3 # 0xFF, ALL OFF (OK)
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,223 && sleep 3 # 0xDF, PIN 6 and 8 on (should be 6 only)
curl http://192.168.31.203/tools?cmd=wdconfig,32,1,255 && sleep 0.3 # 0xFF, ALL OFF (OK)
```

log output when I run above script (seems no details)
```
Logging: Debug Dev (9)
35772: Webserver args: 0: 'cmd' length: 17
35784: Command: wdconfig
36648: Webserver args: 0: 'cmd' length: 17
36658: Command: wdconfig
37529: Webserver args: 0: 'cmd' length: 17
37541: Command: wdconfig
38402: Webserver args: 0: 'cmd' length: 17
38412: Command: wdconfig
41984: Webserver args: 0: 'cmd' length: 17
41995: Command: wdconfig
42852: Webserver args: 0: 'cmd' length: 17
42863: Command: wdconfig
46450: Webserver args: 0: 'cmd' length: 17
46462: Command: wdconfig
```
Screenshots:
[image]
[image]

giig1967g, 2020-04-09:
> If you want to control all 8 pins at once you can use this trick:
> /tools?cmd=wdconfig,32,1,x
> where x is value from 0-255 to send to PCF
> 
> in such case you even not need to configure tasks as switch input :)

Hi @uzi18 
what is the syntax of wdconfig?
Can it be used in rules?

uzi18, 2020-04-09:
it is designed for external watchdog configuration, it sends here 2 bytes under address 32 on the bus, here 1 and x as next, because pcf8574 has got only one byte buffer only x remains there and it drives outputs

giig1967g, 2020-04-09:
@uzi18 does it work for MCP too?
what is the syntax?

uzi18, 2020-04-09:
do you mean syntax for mcp? 
I think we change address before write to register there so wdconfig will not work as it writes everytime from register 0 in i2c device memory.

giig1967g, 2020-04-09:
so the wdconfig command works only for PCF?
I am not sure I understood this command :(

uzi18, 2020-04-09:
@giig1967g that's why said it;sa trick, because wdconfig has different propose, it sends 2 bytes at adress 0 of i2c device buffer. I use it for pcf to set all 8bits when needed and no configuration in task is needed.
Have application where AVR uC sends to serial port of espeasy command via wifi  to launch this command in different espeasy ;)

Fapibo, 2020-04-11:
> 
> 
> it sends here 2 bytes under address 32 on the bus, here 1 and x as next, because pcf8574 has got only one byte buffer only x remains there and it drives outputs

If I understood well, wdconfig directly sends values to I2C without involving the pcf plug in? 
If you saw my last script, the issue still occurs even with wdconfig... Should I consider it's HW failure? Did you understannd why I should reboot after PCFGPIO?

uzi18, 2020-04-11:
yes, if wdconfig also turn on wrong gpio for you please change pcf module, because it is hardware failure.

Fapibo, 2020-04-11:
OK thanks. I will try.

Fapibo, 2020-04-16:
Confirmed. It works with another HW. Thanks for your support

closed 2020-04-16 by Fapibo