letscontrolit/ESPEasy #2997: [ESP32] Static IP does not seem to work

TD-er, 2020-04-10:
As mentioned here: https://github.com/letscontrolit/ESPEasy/issues/2941#issuecomment-605083896

TD-er, 2020-04-10:
Some log of what happens on ESP32 with static IP:

```
INIT : Booting version:  (ESP32 SDK v3.2.3-14-gd3e562907)
38 : Info  : INIT : Free RAM:291304
38 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
46 : Info  : FS   : Mounting...
80 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
99 : Info  : CRC  : SecuritySettings CRC   ...OK
172 : Info  : INIT : Free RAM:286256
172 : Info  : INIT : I2C
173 : Info  : INIT : SPI not enabled
367 : Info  : INFO : Plugins: 80 [Normal] [TEST ESP32] (ESP32 SDK v3.2.3-14-gd3e562907)
468 : Info  : WIFI : Set WiFi to STA
667 : Info  : WIFI  : Start network scan
669 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
675 : Info  : WIFI : Connecting Lurch3 attempt #0
680 : Info  : Webserver: start
683 : Info  : OTA  : Arduino OTA enabled on port 3232
933 : Info  : SYS  : 214652.00
945 : Info  :  Domoticz: Sensortype: 1 idx: 139 values: 214652.00
1815 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 255.255.255.0   duration: 1755 ms
1840 : Info  : SYS  : 100.00
1850 : Info  :  Domoticz: Sensortype: 1 idx: 195 values: 100.00
1865 : Info  : SYS  : 0.00
1875 : Info  :  Domoticz: Sensortype: 1 idx: 492 values: 0.00
2032 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 212060 WiFiStatus 3
10685 : Info  : WIFI  : Start network scan
10687 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
10694 : Info  : WIFI : Connecting Lurch3 attempt #1
10698 : Info  : WIFI : Static IP: 0.0.0.0 (ESP-Easy-32) GW: 0.0.0.0 SN: 0.0.0.0   duration: 10694 ms
10918 : Info  : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 2 ms
16697 : Error : WiFi  : Scan failed
16801 : Info  : WIFI : Set WiFi to STA
16911 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
16916 : Info  : WIFI : Connecting MikroTik2 attempt #2
20806 : Info  : WIFI : Set WiFi to AP+STA
21643 : Info  : WIFI : AP Mode ssid will be ESP_Easy_32 with address 192.168.4.1
32431 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 208908 WiFiStatus 6
37024 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
37029 : Info  : WIFI : Connecting MikroTik2 attempt #3
57208 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
57213 : Info  : WIFI : Connecting Lurch3 attempt #4
57567 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 255.255.255.0
60290 : Info  : SYS  : 208284.00
60301 : Info  :  Domoticz: Sensortype: 1 idx: 139 values: 208284.00
60330 : Info  : SYS  : 75.20
60340 : Info  :  Domoticz: Sensortype: 1 idx: 195 values: 75.20
60379 : Info  : SYS  : 1.00
60390 : Info  :  Domoticz: Sensortype: 1 idx: 492 values: 1.00
62030 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 208456 WiFiStatus 3
77219 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
77225 : Info  : WIFI : Connecting Lurch3 attempt #5
77229 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 255.255.255.0
77342 : Info  : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 3 ms
77457 : Info  : WIFI : Set WiFi to AP+STA
77559 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
77564 : Info  : WIFI : Connecting MikroTik2 attempt #6
77919 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 255.255.255.0
92032 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 206656 WiFiStatus 3
97574 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
97581 : Info  : WIFI : Connecting MikroTik2 attempt #7
97585 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 0.0.0.0
97698 : Info  : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 3 ms
97813 : Info  : WIFI : Set WiFi to AP+STA
97915 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
97920 : Info  : WIFI : Connecting Lurch3 attempt #8
98275 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 255.255.255.0
117929 : Info  : IP   : Static IP : 192.168.10.100 GW: 192.168.10.254 SN: 255.255.255.0 DNS: 192.168.88.1
117935 : Info  : WIFI : Connecting Lurch3 attempt #9
117939 : Info  : WIFI : Static IP: 192.168.10.100 (ESP-Easy-32) GW: 192.168.10.254 SN: 255.255.255.0
```

After a while the ESP32 does reply on the network (browser shows a page and Domoticz does manage to get connected to the broker.
But then it retries to connect so all connectivity is lost again.

ironataerial, 2020-04-10:
Michal mentioned the same behavior on the ETH interface as well.  Sign that points away from the interface(s) ?

TD-er, 2020-04-10:
I did find some logic errors in the handling of WiFi events, especially for ESP32.
Also it does seem like we don't get WiFi events if using static IP.
I am looking into it and already have it working here on my node, but now have to pinpoint what actually happens so I can be sure to fix the right thing and not just fighting symptoms.

PR #2999 (2020-04-10): [WiFi] Fix problem with static IP and reconnect ESP32 (#2997)
Fixes: #2997
commit fe195c60 (2020-04-10): [WiFi] Fix problem with static IP and reconnect ESP32 (#2997)

Fixes: #2997
closed 2020-04-10 by TD-er