letscontrolit/ESPEasy #3019: PulseCounter "setpulsecountertotal" behavior is unexpected when using formula

TD-er, 2020-04-14:
See [forum discussion](https://www.letscontrolit.com/forum/viewtopic.php?p=43976#p43976)
In short, when using a formula, its behavior is unexpected as it is not calling the same code as a `PLUGIN_READ`

So maybe it is best to make sure to call `PLUGIN_READ` (or equivalent code) after setting the values (and resetting it)

uzi18, 2020-04-15:
@TD-er maybe some plugins needs oportunity to store own internal structs in RTCmem instead of UserVars?

TD-er, 2020-04-15:
> @TD-er maybe some plugins needs oportunity to store own internal structs in RTCmem instead of UserVars?

Yep, but that may require quite some rewrite as you need to implement some specific plugin calls to deliver the binary dump of the plugin's representation of what needs to be stored.
Also it may fluctuate in size, or we should demand it to fit in 16 bytes per task.
For now, this will probably do, but later we may need to have a look at it.
Also the call to store the data in RTC is writing way more than needed, so I guess a simple optimization can be to let only the changed task write the update of those bytes (N.B. 32 bit writes at once to RTC)

PR #3020 (2020-04-18): [Pulse Counter] Call PLUGIN_READ after set counter (#3019)
Fixes: #3019
Only setting the internal counter values is not enough.
The output values must also be set and a call to any optional formula must also be made.
A call to `PLUGIN_READ` will also store the values in RTC.
commit fc8c0b56 (2020-04-18): [Pulse Counter] Call PLUGIN_READ after set counter (#3019)

Fixes: #3019
Only setting the internal counter values is not enough.
The output values must also be set and a call to any optional formula must also be made.
A call to `PLUGIN_READ` will also store the values in RTC.
closed 2020-04-18 by TD-er