letscontrolit/ESPEasy #3022: Request: Firmware Release in Plugin

fraeggle, 2020-04-17:
i have a question. In my iobroker i have an overview of my ESP. At the moment I enter there the SW state "Manual". Would it be possible to integrate it into the Generic System Info Plugin?

[image]

Regards Peter

TD-er, 2020-04-17:
I don't understand what you mean to ask here.
What "SW state" do you mean?
The build name? Or build date?

fraeggle, 2020-04-17:
sorry ....
It receives the Data via mqtt
I will use following from the ESP main Page 
[image]
it would be nice if this could be selected via the SystemInfo, like for ex. Uptime.
[image]

fraeggle, 2020-04-17:
> Also, how does IO broker receive the data from the ESP?
> I guess almost all this information is present in the JSON output.
hmm how do i get this?

TD-er, 2020-04-17:
The build information like `Git Build` or the build date are strings, they are not numerical values.
What can be done is that I create the build time as a Unix timestamp (nr. of seconds since Jan 1st 1970, GMT)
But then you have to convert that yourself in your application.

The JSON information can be retrieved via the `http://<ip>/json` URL, but I do see this information is missing there.
But as you do fetch the data via MQTT, I guess it doesn't matter whether this information is present in the JSON anyway.

I suggest you add some rules to publish the relevant info yourself.
See the system variables page (link on the tools page) to see all available variables.

Then the publish rule can be something like this:

```
on Clock#Time=All,**:** do //will every minute
 publish,"topic",%sysbuild_date%
endon
```

Of course you need to change the topic yourself

fraeggle, 2020-04-17:
You are amazing.
Only thing, must use quotes:
Publish,%sysname%/software/build,"%sysbuild_date%"
Otherwise you get:
Too many arguments, try using quotes!Command unknown: "Publish,ESP_1OG/software/build,Feb 21 2020"
But with quotes i receive in the broker: Feb 21 2020
Converting in other format is then no problem

Thanks for your help.
Will close this one  :-D

Regards Peter

closed 2020-04-17 by fraeggle