letscontrolit/ESPEasy #3023: ESP32 Analog input - internal, problem with GPIO ports

tomatos-st, 2020-04-17:
Hi
Recently I tested ESP32 and analog inputs (**Analog input - internal**) plugin **P002_ADC**
I noticed that in this plugin you can choose up to 32 GPIO, while there are only 18 analogue inputs.
8 for 1ADC (GPIO: 36,37,38,39,32,33,34,35) and 10 for 2ADC (GPIO: 4,0,2,15,13,12,14,27,25,26)
[https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf](url) , page 57 , Table 19

I tested the **ESP_Easy_mega-20200410_test_ESP32_4M316k** version.
Works correctly only 6 analog on GPIO ports: 36,39,34,35,33,32, i.e. ADC1
The remaining ports have fixed values: 0 or 4095

Tested equipment is:
ESP32 DEVKIT V1,
[https://randomnerdtutorials.com/esp32-pinout-reference-gpios/](url)
and
wemos D1 mini ESP32
[https://www.eelectronicparts.com/products/mh-et-live-d1-mini-esp32-esp-32-wifi-bluetooth-internet-of-things-development-board](url)

Can anyone confirm my tests?

[image]

TD-er, 2020-04-17:
Looks to be correct, as the ones that appear to work all belong to ADC1.

Also I don't know how well the ADC plugin is capable of running multiple instances of the same plugin.
So it is very well possible reading one task may affect the next task running the same (ADC) plugin.

I don't know how well the ESP32 can separate the readings from one gpio to the next.
Also it may be worth looking into to see which ADC is shared with other functions of the ESP.
For the ESP82xx, the ADC is also used during RF calibration. So reading it during an attempt to connect to WiFi may give strange readings.

tomatos-st, 2020-04-17:
Is there any way to run ADC2?
I cannot start any ADC2 reading at this time.
I need to read 16 ADC ports in my project :(, at the moment I can only 6 ...

I suggest blocking the selection of GPIOs that are not supported in the plugin

TD-er, 2020-04-17:
Right now there is no support to chose the other ADC on the ESP32.
I can add it, but not sure when I do have time for it.

tomatos-st, 2020-04-17:
Please, include this in your plans. I will wait.
For now, I will convert my project to 4 x ADS1115

uzi18, 2020-04-17:
@tomatos-st try to disable oversampling and it should work.

tomatos-st, 2020-04-17:
@uzi18 I don't use oversampling.
The GPIO port that supports ADC2 does not work.
As @TD-er  wrote, it looks like Right now there is no support to chose the other ADC on the ESP32.
[image]

tonhuisman, 2020-04-18:
Using the ADC2 ports is only possible on an ESP32 when WiFi is turned off. I don't expect many to make use of an ESP32 in that way effectively..., though you can still output data through the serial port, of course.
See the note in blue, in the paragraph about the ADC functions, on this Random Nerd Tutorials page: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Limiting the GPIO selection here to only ADC-capable GPIO's (32..39, att: many boards don't seem to have 37 and 38 connected to a pin) seems the most usable improvement.
Maybe ADC2 ports should only be available with WiFi turned off (not sure if that's possible/plausible), and then also limited to the GPIO's (0,2,4,12..15,25..27) connected to ADC2.

For now, using the currently available ports and adding ADS1115's for additional ADC ports seems the most plausible solution, but as the ADS1115 has a higher resolution than the internal ADC's, it might even be better/easier (less calculations) to only use ADS1115's.

TD-er, 2020-04-18:
> Maybe ADC2 ports should only be available with WiFi turned off (not sure if that's possible/plausible), and then also limited to the GPIO's (0,2,4,12..15,25..27) connected to ADC2.

Is the limitation about "WiFi turned off" or "Not in the process of making a WiFi connection to an AP" ?

tonhuisman, 2020-04-18:
Well, the note on that page says this:
``ADC2 pins cannot be used when Wi-Fi is used. So, if you’re using Wi-Fi and you’re having trouble getting the value from an ADC2 GPIO, you may consider using an ADC1 GPIO instead, that should solve your problem.``

closed 2020-04-27 by TD-er

Escrich, 2021-03-28:
dear all, even having the anaklog input working, there is something wrong in the readings, everitime you start to use bluetooth, i have most extensive information on the below link, could you please have any sugestions?
http://www.escrich-sanchez.com/20210219%20Problems%20on%20analog%20input%20of%20DOIT%20ESP32%20DEVKIT%20V1/

TD-er, 2021-03-28:
Do you use one of those pins which is connected to ADC2?
See [documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P002.html#wifi-activity-and-adc)

tonhuisman, 2021-03-28:
> everitime you start to use bluetooth

I don't think ESPEasy supports Bluetooth, at the moment...

What GPIO pins are you using for those analog inputs? As the ADC2 pins shouldn't be used when WiFi is enabled/used, maybe similar issues exist for a Bluetooth connection?

Escrich, 2021-03-28:
> 
> 
> Do you use one of those pins which is connected to ADC2?
> See [documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P002.html#wifi-activity-and-adc)

no, I'm using an input pin connected to ADC1, anyway I was tested analog pins onACD1, 2 and 3

Escrich, 2021-03-28:
> 
> 
> > everitime you start to use bluetooth
> 
> I don't think ESPEasy supports Bluetooth, at the moment...
> 
> (What GPIO pins are you using for those analog inputs?) As the ADC2 pins shouldn't be used when WiFi is enabled/used, maybe similar issues exist for a Bluetooth connection?

I'm currently using Bluetooth, please go to the link I published before, and get the program writen for Arduino environment, then, you'll see

TD-er, 2021-03-28:
Sorry, but I really don't get what the problem is you're having.
In your link I do see a number of screenshots and a link to an .ino file which has all in Spanish (I guess?), no explanation of what's going on and also it is just a stand-alone project so it seems. 
It doesn't seem like it is an extension to ESPEasy as a work-in-progress.  Or am I missing the ESPEasy part somewhere?  
Some part is marked in the screenshots but without any explanation of what is intended, I really don't get it.

Escrich, 2021-03-28:
> Sorry, but I really don't get what the problem is you're having.
> In your link I do see a number of screenshots and a link to an .ino file which has all in Spanish (I guess?), no explanation of [...]

I think we're losing the point, I'll resume it, if you are reading an analog input on ADC1, you can measure with no problems, just if you switch on the Bluetooth at this point, all the analog readings will change in more than 30 points, is there any way to avoid this?

TD-er, 2021-03-29:
Nope, ADC2 is also used by the 2.4 GHz radio, which is shared among WiFi and Bluetooth.
Not sure if by reading ADC2 you also affect the radio communication, but at least it does have an effect on ADC2 readings.

PR #3034 (2020-04-27): [ADC] Improve oversampling filtering + ESP32 support
Oversampling:
Values only added when not already the min or max value.
This way a number of erroneous min values or max. values (e.g. due to WiFi activity) will not affect the averaging function.

Perform first reading of ADC before any WiFi activity is taking place. This first reading is shared among VCC readings of sysinfo plugin and ADC plugin.

ESP32 improvements:
- Display detailed ADC pin capabilities/description for GPIO selection
- Only allow GPIOs to be selected for pins that can actually be used for ADC.
- Allow to read the interna [... cut]