letscontrolit/ESPEasy #3024: feature: get rid of special build for ADC_VCC

uzi18, 2020-04-18:
How about such change:
```diff
diff --git a/src/ESPEasy.ino b/src/ESPEasy.ino
index d5df0493..b588606e 100644
--- a/src/ESPEasy.ino
+++ b/src/ESPEasy.ino
@@ -119,9 +119,10 @@
 #include "src/Globals/Settings.h"
 #include "src/Globals/Statistics.h"
 
-#if FEATURE_ADC_VCC
-ADC_MODE(ADC_VCC);
-#endif
+int actual_adc_mode = ADC_TOUT_3V3;
+int __get_adc_mode(void) { 
+    return actual_adc_mode; 
+}
 
 // Get functions to give access to global defined variables.
 // These are needed to get direct access to global defined variables, since they cannot be defined in .h files and included more than once.

```
@TD-er what do you think?

TD-er, 2020-04-18:
`ADC_MODE(ADC_VCC);` has to be outside the scope of any function, so you cannot make it runtime configurable.

If that has changed I would love to see a link explaining it.

uzi18, 2020-04-18:
this macro ADC_MODE creates such function like in my patch but with constant return, my idea is to avoid it and return what we want to measure

TD-er, 2020-04-18:
The call to that macro is more like a compile-time call if it is included in the code, then the Arduino core library does set the ADC mode to TOUT.
It is like an override function definition of a weak function declaration.
So it is still compile time and not runtime, as we cannot access variables declared in the Arduino stack.
Or you should add core lib only code that will read something from RTC memory, which also forces a reboot to make it work if you do want to set it different from the 'default' value.
But I'm not sure if that code is able to access RTC memory reserved for "user" code.

uzi18, 2020-04-18:
so ... maybe it is possible to workaround this part of hardware, but we need 1 free byte in rtc mem :)

TD-er, 2020-04-18:
And we need to be able to read that part of RTC... We cannot use anything that's part of the Arduino stack, only what's available in the SDK.

uzi18, 2020-04-18:
@TD-er maybe unused1 or unused2 byte in RTCStruct?

TD-er, 2020-04-18:
You can use those unused.
You need to realize you should read the RTC data per 32 bit and also the address is per 32 bit, so you have to divide by 4 when addressing.
See the comments in the RTC struct to see what are we use.

uzi18, 2020-04-18:
yes, know but maybe will try tomorrow, priority is low so maybe afternon or late night will look at it, usually take bed at 3 or 4 morning :) 
why we have still no gitter.im channel??

TD-er, 2020-04-18:
Well I am connected to Slack.
You can get an invite for the ESPEasy channel if you like.

uzi18, 2020-04-18:
yes but slack is locked, if you can?