letscontrolit/ESPEasy #3025: P036 Framed OLED reboot deletes content lines

piano-thomas, 2020-04-18:
Hi all,

I'm experiencing a small issue with P036 Framed OLED. When i reboot my nodeMCU/Wemos, the content of the lines that I entered vanishes. Every reboot I have to enter my content again. I tried to use the rules tool to fill the lines on System#Boot, but I am using an MQTT value, that will be just filled with data of that moment.
For example, when I use this rule:

On System#Boot
OLEDFRAMEDCMD,1,"Testvalue = [MQTT1#Value1]"
EndOn

The value that appears in line 1 of my Framed OLED display will be the numerical value of MQTT1#Value 1 at that instant, in stead of an updating value.

Is there an easy way to fix this?

Cheers,

Thomas

v-a-d-e-r, 2020-04-18:
I would use MQTT#Connected instead of System#Boot. Do you use the MQTTimport PlugIn?

TD-er, 2020-04-18:
What do you mean by "Every reboot I have to enter my content again" ?
Do you mean on the lines of the plugin config?
Or data written to it via the command in the rules?
The data written to the display via the command is not stored in the settings, as that's volatile data (e.g. a status update)
If you need to store data, you should use the lines of the plugin config.

N.B. you can later update the content of the lines via the command, but that is lost after a reboot.
So you can store some initial value in the settings and later replace it via the command.

piano-thomas, 2020-04-18:
Thanks for your quick replies!
Yes, I am using the MQTTimport plugin, which works perfectly fine :)

I use the OLED Framed plugin to configure my OLED display, and write my text/values in the textboxes lines 1-12. However, when I reboot the NodeMCU, and look at those lines again, they are all empty. 

With my LCD display, that problem does not occur, so I was quite surprised. The only thing I can think of would be, that I changed the number of lines in the P036 plugin from 12 to 20 and recompiled it myself, but I think i had the same problem with the original plugin.

tonhuisman, 2020-04-18:
Hmm, I've been spending some time using the OLedFramed plugin lately, and worked a bit on that code, but I've never lost the settings during my tests. 
Increasing the number of lines beyond 12 may be problematic, as the space reserved for the settings is limited and fixed, it doesn't allow any extending, AFAICS.

What version of ESPEasy are you using here?

piano-thomas, 2020-04-18:
Hi Ton,

I am using the ESP Easy MEGA 20200410 4M release to compile. I imagine that my extension to 20 lines could be an issue, but it should work with 12 right?
I will try to use the normal_ESP8266_4M1 version tomorrow and have another try. 

The reason why I tried to use 20 lines was, that I have too much data for 1 display. I tried using two OLED displays at the same time (with two different I2C addresses ofcourse), but ESPeasy seems to get very confused. A workaround could be using the 'normal' OLED display plugin for one display, and the framed plugin for the other one, but for me the font in the 'normal' plugin is very small, and I did not find a way (yet) to decrease the number of lines, and increase the font size.. :)

TD-er, 2020-04-18:
Well increasing the number of lines is not always just simply changing a single value and recompile.
The data stored is often a struct which is created in memory and then written as binary dump to the settings file.
I did (very recently) make some changes where I do allow for a bit more resource friendly loading/saving of the strings of that plugin, so maybe it is working better on the latest code.

Another thing you need to keep in mind is that the data segments for settings is also restricted.
So you have to make sure the settings you wish to store do not overlap with data of another plugin.

tonhuisman, 2020-04-19:
> The reason why I tried to use 20 lines was, that I have too much data for 1 display. I tried using two OLED displays at the same time (with two different I2C addresses ofcourse), but ESPeasy seems [...]

It is true that having multiple OLEDFramed instances 'confuses' ESPEasy, I plan to try and fix that (I've taken some interest in these little displays) once I obtain a display that can be switched to the alternate address, as so far I've not been able to tweak mine (I have a couple of them)

In the meantime, would it be feasible to switch to a Nextion display?
They are larger than the OLED displays, have a higher resolution so more data can be displayed, and you can have much better control over font-size, positioning, etc., and they are available with a touch interface (but you don't have to use that). It does take more effort (and money, as they start at around $12 at AliExpress) to get started, but in the forum are several examples, and this wiki page is a good place to start https://www.letscontrolit.com/wiki/index.php/NEXTIONDisplay along with more tips in this wiki page https://www.letscontrolit.com/wiki/index.php/Nextion

TD-er, 2020-04-19:
Using multiple instances of the same plugin does require some changes in the plugin code.
For example, have a look at how the plugin data is handled for P085_AcuDC243.

```c++
struct P085_data_struct : public PluginTaskData_base {
...
```

This struct inherits from PluginTaskData_base and each time it is accessed, it is fetched from a vector of base pointers and casted into the right type to access the data.
This way, you can have multiple instances of the same plugin active that need to store some member data.

piano-thomas, 2020-04-20:
> In the meantime, would it be feasible to switch to a Nextion display?
> They are larger than the OLED displays, have a higher resolution so more data can be displayed, and you can have much better [...]

Hi Ton,

Thanks for the tip! I didn't know about the existence of these displays! I just ordered one, and will try if that's a good alternative for displaying all my data :).

In the meantime I tested the plugin again with the original build, which worked fine. I'll know have to get myself more acquainted with the plugin code to see if I can get 20 lines working, while getting rid of some plugins I will not use to have the desired amount of memory available.. Thanks for helping out!

closed 2020-04-20 by piano-thomas