letscontrolit/ESPEasy #3035: Easy ESP on esp8266 drops TCP connection to Serial server if there is another connection

std10k, 2020-04-26:
Easy ESP drops TCP connection to the serial server port if there is another connection to that port which times out or gets terminated otherwise. 

I have esp8266 with cc2530 zigbee coordinator connected over GPIO.  Zigbee2mqtt software connects to the serial port (1775 in my case) and works fine until another TCP connection occurs to that port. I noticed is as i had a monitoring system (PRTG) that was checking the port for availability every 5 minutes, and every 5 minutes (or 1 minute, depending on probe settings, but always consistently) Zigbee2mqtt's connection will get terminated causing it to restart.   Telnet on port 1775 from windows seems to have the same effect. 

### Expected behavior
Perhaps EasyESP should not kill all connections like this as there is no way to control access by individual port (i believe), which means that with this issue present, any random connection can destabilise EasyESP and cause the client software to crash (unless it can handle tcp reconnect, which zigbee shepherd for example cannot)

### Steps to reproduce
1) esp8266 with serial server
2) establish persistent connection to the serial server over TCP from the expected client
3) get another system like PRTG to connect to the TCP port of the serial server
4) the first connection gets dropped. 

am experiencing it with the latest mega build from April 2020, but previously had the same behavior with "stable build 120" so it seems to be carrying over. 

``
Build:⋄
20105  - Mega
System Libraries:⋄
ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20200410
Plugin Count:⋄
46 [Normal]
Build Time:⋄
Apr 10 2020 19:50:11
Binary Filename:⋄
ESP_Easy_mega-20200410_normal_ESP8266_4M1M.bin
``

TD-er, 2020-04-26:
Just a note of the first thing that comes to mind right now, so it is more of a reminder for me to check it.
There is a fixed number of max. open connections.
Maybe we're hitting this limit?

std10k, 2020-04-26:
How can i check that setting?  can't see anything obvious in the settings. 
It not that a new connection cannot be established, it can be, but it causes for the existing connection to get dropped. Though it could be that only the latest one survives.  In my environment there should be no other connections, so it would have to be set to '1' to cause the issue.

TD-er, 2020-04-26:
It is somewhere around 5 if memory serves me well.

But I can imagine if it is a connection made by a `WiFiClient` object, that it can only handle a single active connection.

uzi18, 2020-04-26:
first try change port to 80 in your monitoring system

sincze, 2024-04-01:
Do we have an idea about the amount of simultaneous sessions the esp8266 and Communication Server is able to provide with the current ESPEasy release??
(Asking for a friend :) )

TD-er, 2024-04-01:
Depends on other connections as well.
If I'm not mistaken, the number of open connections is 5.
But this is shared among all TCP connections. (maybe also UDP, not 100% sure)

sincze, 2024-04-01:
Ah those other connections. mmmm Any suggestions on how I can figure this out ;-)

tonhuisman, 2024-04-01:
I expect that the serial server, and other plugins that allow an IP connection from outside, are/act like point-to-point connections, accepting only 1 client simultaneously.

sincze, 2024-04-02:
Don't know if it is really the case.
Assuming it is based on Ser2Net, multiple connections should be possible
from Ser2Net 3.2+
https://github.com/cminyard/ser2net

https://github.com/cminyard/ser2net/blob/master/ser2net.yaml.5
On a port to allow more than one connection at a time to the same serial
port.
These connections will share all the settings.
You cannot have two separate TCP ports connect to the same port at the same
time.

So the happy flow would be in my case 2 IP's connecting to the same port
10004.

Sándor

TD-er, 2024-04-02:
As far as I can see, Ser2Net isn't for ESP/Arduino platforms.
We are limited by the limits of LWIP.

Not sure what should be done to let LWIP does support multiple active connections to the same port.
I know the async webserver does seem to be able to accept a new request while still serving another request, so technically it might be possible.
But I guess it might make this plugin quite a bit more complex as you also should handle separate write operations from each connected client.