letscontrolit/ESPEasy #3036: Request: Support for INA3221

satdj, 2020-04-26:
no issues, only request

TD-er, 2020-04-26:
Hmm I do see an issue here.
This module supports upto 3 inputs where both voltage and current can be measured.
So I guess you may want to output 2 or 3 values per input (power can also be useful)

This means you may need 8 - 12 values on the output of a plugin.
So I guess it should be made a bit similar to the Eastron plugin.
That one has the option to share a data collection object among multiple instances of the same plugin to be able to output all different outputs on different tasks.

satdj, 2020-04-26:
at first i want to use it as a battery monitor (6 big Cells lead acid) and just need the voltage. (2 modules = 6 voltages)
with ina219 i can only have 4 voltages

somehow I have to calculate that later. I don't know how i do that yet: D

the second step i will make a solar-log, then i need all. But there i can't use the ina's. At solar side i have up to 60VDC.

Solar Input (from Solar Cells)
Solar Output (from MPPT Charger)
Load

TD-er, 2020-04-26:
You could add a fixed voltage offset to shift the GND of the sensor.
For example using a 40V offset and a diode to protect the sensor if the input voltage is removed.

But I guess there may be better solutions which have the shunt in the GND line.

fredvduin, 2021-06-26:
I have build a INA3221 device for my LiFePO4 12.8 Volt battery.
In Node-Red I calculate the power.
For the battery cells I use a modified ADS1115 device that can read 4 values in one device .(and using Voltage Divider)
[image]

Code:
[_P227_3221.LOG](https://github.com/letscontrolit/ESPEasy/files/6720289/_P227_3221.LOG)

EDsteve, 2022-01-10:
That is exactly what i am looking for. But compiling it into a new bin file exceeds my expertise :)
Is there a chance that will be implemented in a test-release in the near future?

fredvduin, 2022-01-10:
I can give you my bin file, are you using ESP32 or ESP82xx

EDsteve, 2022-01-10:
Oh great. That would be a big help. I use ESP8266. You will upload it here?
Thanks so much

tonhuisman, 2022-01-10:
Sharing the plugin-source would be even more helpful, either via a PR or as source, so it can be added to ESPEasy...

EDsteve, 2022-01-10:
Thanks so much. I will try it.
If the ESP32 version is the newer firmware. I can easily switch to an ESP32. Bring it on :)

clumsy-stefan, 2022-03-18:
I'd also vote to include this at least in the playground or even better in the standard repository. There are one or two small changes necessary to make it compile with the actual version, just different definitions. I'm happy to support if I can.

@tonhuisman who is defining the plugin-nr?

@fredvduin 
> Where can I upload the source for testing/sharing, or do I create for each plugin a PR

see answer from @tonhuisman , easiest would be if you create a PR in the standard repository.

tonhuisman, 2022-03-18:
> who is defining the plugin-nr?

I have reserved **`P132`** for this plugin (and mentioned you in the issue that lists the WIP). Named it 'Energy - INA3221' for now, but if you use a different title I'll update that description.

clumsy-stefan, 2022-03-19:
As this is purely written by @fredvduin I dont want wo hijack his code... so let's wait for his feedback...

tonhuisman, 2022-03-19:
I've added his name to the plugin-is issue 😺

fredvduin, 2022-03-19:
Sorry it is new for me, You can use my code and I have not create a PR for every plugin (ADS1115 - 4 port, MCP3424 18 bit 4 port, INA219/INA226 and AD524x potmeter) I need some help for that but I can not find someone for making a PR

clumsy-stefan, 2022-03-19:
the plugins you mention already exist in the repository, did you provide them?

If It's OK for you I'm happy to create a PR out of your code (with somme small changes, eg. ID, etc.)! Of course I'll mention the credits to you.

Unfortunately I can't really test it yet, as I'm still waiting for the boards ;)

fredvduin, 2022-03-19:
Thanks,

No, my plugin (ADS1115 and MCP3424) can read 4 values in one plugin 
Very useful on ESP8266
[image]

clumsy-stefan, 2022-03-19:
ahh, cool... did you change/adapt the existing plugin or is it completely new?

I guess the problem is, if we're changing an existing one there could be compatibility issues...

But basically you will need to create a PR for each plugin sepparately, so it can be selectively tracked and merged..

fredvduin, 2022-03-19:
ADS1115 is basically original plugin and I add a loop. But I removed the calibration (code original plugin is more than 2 years old)
MCP3424 works different because for 18 bits it takes a very long time to measure and I do not want to use long wait state

clintinthecode, 2022-04-18:
Any update on this plugin? I am needing to use this board, if I can help with anything let me know

tonhuisman, 2022-04-18:
@fredvduin I've written a [beginners-guide for developers that intend to work on ESPEasy](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy), f.e. when writing a new plugin 😉. If you have issues when using this guide, or think more guidance would be helpful, then please ask/suggest.
If that's all over your head (which I doubt), either @clumsy-stefan or myself can step in and create a PR for you. I do feel [the code you shared](https://github.com/letscontrolit/ESPEasy/issues/3036#issuecomment-869017801) can use a bit of TLC though, to better fit with the current ESPEasy code-base, and some de-duplication in the I2C read/write code (there is a set of standard I2C helper functions available via `ESPEasy/src/src/Helpers/I2C_access.h`).

fredvduin, 2022-04-18:
@clintinthecode do you need INA3221 or ADS1115 ...

satdj, 2022-04-18:
Ohhh, activitiy

INA3221 whould be nice since 2020 ;D

i have any high-side boards (3 different Voltage/Current, mostly Black board) i not use at moment. 
die most boards have common Voltage, 3 different Current (mostly violett board)

by no time by work .. i have not use them since 2 years

clintinthecode, 2022-04-18:
@fredvduin i need INA3221. I want to design a power distribution module using espeasy. Will have 8 outputs each with their own current sensor. I need about 5 of these modules for my boat, 1 or 2 for my car, 1 or 2 for my trailer, so it would cost over $10,000 to buy PDM’s for this, so I want to make my own.

fredvduin, 2022-04-19:
@tonhuisman / @clumsy-stefan can you make a PR

See #3839 P132

tonhuisman, 2022-04-19:
> can you make a PR

I could, but where is the code...?

fredvduin, 2022-04-19:
[_P227_3221.LOG](https://github.com/letscontrolit/ESPEasy/files/6720289/_P227_3221.LOG)

Message commented [on 26 Jun 2021](https://github.com/letscontrolit/ESPEasy/issues/3036#issuecomment-869017801)

tonhuisman, 2022-04-19:
Aha, then it will take a couple of days (evenings). I want to do some work on it before creating a PR, as there are quite some duplicate strings that need to be avoided for (memory & code) efficiency, and I need to look at multi-instance use too. I have some other stuff, non-programming (I'm trying to have a life too 😛), to do also. But I assume it's not _that_ urgent 😃

tonhuisman, 2022-04-21:
The GH Actions build failed, so for those who want to test the PR: @clintinthecode @satdj @clumsy-stefan 😉 
[ESP_Easy_mega_20220421_energy_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/8534801/ESP_Easy_mega_20220421_energy_ESP8266_4M1M.zip)
[ESP_Easy_mega_20220421_energy_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/8534803/ESP_Easy_mega_20220421_energy_ESP32_4M316k.zip)

PR #4011 (2022-05-31): [P132] New plugin INA3221 3 channel DC Voltage/Current
Original Plugin created by @fredvduin

INA3221 DC Voltage/Current measurement, I2C.

Original features:
- Select 4 of 6 possible values (Current/Voltage for channel 1..3)
- Selectable shunt resistor value (0.1/0.01/0.005 ohm)

Added features:
- Selectable Averaging samples (1..1024 in 8 steps, as supported by the chip)
- Selectable Conversion rate (140 usec..8.244 msec in 8 steps, as supported by the chip)
- Update source to current standards, and using ESPEasy core I2C functions
- Split source into `PluginStructs` for optimized multi-inst [... cut]
closed 2022-05-31 by TD-er