letscontrolit/ESPEasy #3038: Problem with command "Controller Publish"

Tygers25, 2020-04-28:
Hi, 
I have a problem with ESP Easy 32, in Controller Settings, Controller Publish, I want to enter this command "device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%".
When I click Submit, the value of Controller Publish automatically changes to "device=%sysname%", the value is truncated and I don't know why. It is to communicate with Jeedom.
I have the ESPEasy32_R20100 version.

Thanks

TD-er, 2020-04-28:
> I have the ESPEasy32_R20100 version.

Have you tried a version that's newer?
The latest nightly reportedly has some issues when flashed, but for example build of 20200410 should work well.
I will now try it with that version on a node here, but please know that build is over a year old.

TD-er, 2020-04-28:
[image]
I saved it with an unit running ESP32 and there it is saved well as you can see in the screenshot.

TD-er, 2020-04-28:
As it is not a problem with newer builds, I will close this issue.

closed 2020-04-28 by TD-er