letscontrolit/ESPEasy #3041: progMemMD5check uses incorrect range for memcpy

TD-er, 2020-04-29:
See:
https://github.com/letscontrolit/ESPEasy/blob/637e14b5cbfd5e6e1c1dff9f1f44f60c6258dfa8/src/Misc.ino#L1037-L1046

The length of `calcBuffer` is less than the length used in memcpy.  (16)
So this is at least undefined behavior.

TD-er, 2020-04-29:
Also this part is also running out of range:
https://github.com/letscontrolit/ESPEasy/blob/637e14b5cbfd5e6e1c1dff9f1f44f60c6258dfa8/src/Misc.ino#L1049-L1051

PVS-Studio comment:

> [CWE-704](https://cwe.mitre.org/data/definitions/704.html?utm_source=platformio&utm_medium=piohome):The size of the '& CRCValues.compileTimeMD5[16 + l * 4]' buffer is not a multiple of the element size of the type 'uint32_t'.

TD-er, 2020-04-29:
@uzi18 I was hoping you could have a look at this function.
I do believe the current implementation is seriously flawed and may cause instabilities that differ from build to build, without touching this part of the code.  It just depends on what order the object files may be linked.

uzi18, 2020-04-30:
@TD-er what we see here is struct not defined properly:
https://github.com/letscontrolit/ESPEasy/blob/637e14b5cbfd5e6e1c1dff9f1f44f60c6258dfa8/src/src/DataStructs/CRCStruct.h#L22
we use here some chars as md5 sum (16) and rest of this array as a hidden place for pairs of start and end pointers to calculate sum, this should be corrected and code adjusted.
Need to think about it.

> 
> 
> See:
> 
> https://github.com/letscontrolit/ESPEasy/blob/637e14b5cbfd5e6e1c1dff9f1f44f60c6258dfa8/src/Misc.ino#L1037-L1046
> 
> The length of `calcBuffer` is less than the length used in memcpy. (16)
> So this is at least undefined behavior.

it is ok - uint32_t is 4 bytes so calcBuffer is 40 bytes long

uzi18, 2020-04-30:
https://github.com/letscontrolit/ESPEasy/blob/637e14b5cbfd5e6e1c1dff9f1f44f60c6258dfa8/src/src/DataStructs/CRCStruct.h#L22
maybe we should just simply remove `+1` from size ?

TD-er, 2020-04-30:
> it is ok - uint32_t is 4 bytes so calcBuffer is 40 bytes long

Ah OK, so it was good not to walk in there and change stuff, but simply make an issue for it :)

As discussed in another issue, maybe we don't need the CRC anymore, at least for ESP32 we don't.
And the filename we can define at compile time, so that can be set via a define and used in the binary.

uzi18, 2020-05-01:
it should be possible 
need to figure out how to pass env name to global define first
or will find a way to correct this crc in esp32 firmware section ...

TD-er, 2020-05-01:
You can create the env name in the pre: Python script and hand it over just like the `USES_Pxxx` defines.

tonhuisman, 2023-09-02:
Not sure if this is still relevant, can probably be closed.

closed 2023-09-02 by TD-er