letscontrolit/ESPEasy #3054: Generic HTTP don't work since ESP_Easy_mega-20200204_dev_ESP8266_4M1M.bin

sracing, 2020-05-05:
I am using several Wemos D1 with ESP Easy since quite a while and was doing a firmware update on those ESP Easy devices.

On any of my devices, when using a  firmware 'ESP_Easy_mega-20200204_dev_ESP8266_4M1M.bin' (OR NEWER), I am not able to receive data from the ESP Easy 'Generic HTTP' Controller anymore.
With any firmware 'ESP_Easy_mega-20191208_dev_ESP8266_4M1M.bin' (OR OLDER) it works without any issue.

- Note: A very similar issue was reported on 7 Feb under #2878 - Generic HTTP don't work since ESP_Easy_mega-20200204_normal_ESP8266_4M1M.bin refering to a wrong URL encoding (fixed)
- The log on a Debug dev / Kernel level does not show any failure.
- That behaviour can be reproduced by switching between firmware versions.
- I would not see a relation to rules as some of my Wemos D1 devices which are not working with the newer firmware are operated without using any rules - just using the 'Generic HTTP' Controller to forward values

TD-er, 2020-05-05:
This is communication between ESPEasy nodes?

TD-er, 2020-05-05:
What are you trying to send?
Can you give an example of such an URL?

sracing, 2020-05-05:
Its communication between ESPEasy (Wemos D1 Pro) and my SmarthomeNG controller (similar Domotics or FHEM).

When I capture the HTTP requests, this is what I can see with the different firmware versions:
ESP_Easy_mega-20191208_dev_ESP8266_4M1M.bin shows:
http://192.168.178.44:8384//shNG/ws/items/esp_easy.strom.count/0

ESP_Easy_mega-20200204_dev_ESP8266_4M1M.bin shows:
http://192.168.178.44:8384/%2f%2fshNG%2fws%2fitems%2fesp_easy.strom.count%2f0

TD-er, 2020-05-05:
OK, that last part makes it clear it is for sure related to the fix you linked.

What is the string you use in the controller to generate this URL?

sracing, 2020-05-05:
Under Controller Publish: /shNG/ws/items/%sysname%.%tskname%.%valname%/%value%

TD-er, 2020-05-05:
Check!
I will have a look at it. I think it is really similar to the other fix.
Maybe also other controllers are affected here.

uzi18, 2020-05-05:
it is incompatibility with older release but SmarthomeNG controller should respect url encoded link, so for me it is more like bug there

sracing, 2020-05-05:
Beside the pure url encoding, it looks like there is a specific bug if you precisely compare both outputs.
The 20200204 firmware has http://192.168.178.44:8384/%2f%2fshNG ..... with "/%2f%2f" right after the port definition - which would be encoded as "///" ( 3 x the slash character ) while it should be only 2x slash right after the port definition.

uzi18, 2020-05-05:
@sracing try this: shNG/ws/items/%sysname%.%tskname%.%valname%/%value%

uzi18, 2020-05-05:
@TD-er hint: https://stackoverflow.com/a/43537042/12226030

TD-er, 2020-05-05:
I have looked at the code quite intensively and at least in the URL encode functions in ESPEasy, there is nothing wrong so it seems.

```

244382: SYS : 4.00,15640.00,33.89,0.00
244398: //shNG/ws/items/ESP_Easy.test.uptime/4.00
244401: //shNG/ws/items/ESP_Easy.test.freeheap/15640.00
244405: //shNG/ws/items/ESP_Easy.test.load/33.89
244408: //shNG/ws/items/ESP_Easy.test./0.00
244470: HTTP : C008 written to client (160/160)
244470: HTTP : C008 closing connection
244492: HTTP : C008 connecting to nuc.lan:80
```
This is in the log.

I will now look at the link @uzi18 posted...

TD-er, 2020-05-05:
What we're doing is the "method of choice" according to that link.

Or at least that's what is being created as can be seen in the logs.
I has troubles getting the data from the Apache logs, so I can't see what is actually being sent, but at least the data processed in the controller is generated with the correct encoding.

uzi18, 2020-05-05:
it can be some differents between esp/esp32
but first we should compare generic http advanced controller to generic http

TD-er, 2020-05-05:
Ah finally found the logs....
```
192.168.10.65 - - [05/May/2020:23:04:25 +0200] "GET //shNG/ws/items/ESP_Easy.test./0.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:26 +0200] "GET //shNG/ws/items/ESP_Easy.test.uptime/17.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:26 +0200] "GET //shNG/ws/items/ESP_Easy.test.freeheap/15640.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:26 +0200] "GET //shNG/ws/items/ESP_Easy.test.load/25.10 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:26 +0200] "GET //shNG/ws/items/ESP_Easy.test./0.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:27 +0200] "GET //shNG/ws/items/ESP_Easy.test.uptime/17.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:27 +0200] "GET //shNG/ws/items/ESP_Easy.test.freeheap/15640.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:27 +0200] "GET //shNG/ws/items/ESP_Easy.test.load/25.10 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
192.168.10.65 - - [05/May/2020:23:04:27 +0200] "GET //shNG/ws/items/ESP_Easy.test./0.00 HTTP/1.1" 403 452 "-" "ESP Easy/20106/May  5 2020 11:14:39"
```

So just to be sure it isn't something that may have been fixed in the last few days, I will make a test build of this code.

TD-er, 2020-05-05:
Can you test [issue3054_ESP_Easy_mega_20200505_dev_ESP8266_4M1M.bin](https://www.dropbox.com/s/napbhe8orr7d7nj/issue3054_ESP_Easy_mega_20200505_dev_ESP8266_4M1M.bin?dl=0)

Tom-Bom-badil, 2020-05-05:
@sracing:
Side note - running ESPEasy here on a bunch of nodes with UDP communication to shNG, no problems since April 2018 (besides the reboot issues ESPEasy had in between, but that's a different story).

Details [here](https://knx-user-forum.de/forum/supportforen/smarthome-py/1189828-einbindung-esp-mit-espeasy-firmware?p=1218035#post1218035), if of interest.

Probably not solving the problem, but a workaround which is more straight than untilizing the more complex http protocol.

uzi18, 2020-05-05:
have looked into sources of shNG and it supports mqtt, it is far more reliable protocol

uzi18, 2020-05-05:
@Tom-Bom-badil did you use generic udp protocol? screenshots are unavailable for external users... so hard to guess

TD-er, 2020-05-05:
Well other protocol options aside, the described issue does seem like an encoding problem with the HTTP controller and since I cannot reproduce it on my node here based on the current code base, I would really like to know if this test build is working as expected or not.

If that's working as expected, you can all help into finding the best suitable protocol, as I do agree http is probably not the best one for the job.
But first let's make sure there is no hidden bug in it.

sracing, 2020-05-05:
Nearly a fix, but ..

1 - the %valname% now seems to be a combination of the device 'Unit Name' (esp_easy) and 'Unit Number' (1)
2 - root only has 1 slash where the 'old '' versio had 2x the slash

Output for issue3054_ESP_Easy_mega_20200505_dev_ESP8266_4M1M.bin is:
http://192.168.178.44:8384/shNG/ws/items/esp_easy_1.strom.count/0

Output for ESP_Easy_mega-20191208_dev_ESP8266_4M1M.bin was:
http://192.168.178.44:8384//shNG/ws/items/esp_easy.strom.count/0

TD-er, 2020-05-05:
That is part of the %sysname%.  You can see it is in front of the `.`
That's configurable in the settings. (Config => Append Unit Number to hostname:)

sracing, 2020-05-06:
Makes a lot of sense, but it somewhat looks that the 'Append Unit Number to hostname:' switch was not considered in %sysname% for firmware 20191208 and older.

However, the firmware fix now works fully (I just need to disable that 'Append Unit Number to hostname:' switch after the update).

Great work. Thanks for doing this in such short time.

closed 2020-05-06 by TD-er