letscontrolit/ESPEasy #3056: controller publish eps32

tiruds76, 2020-05-06:
hello everyone I am a beginner and I installed a PIR sensor HC-SR501 a wemos mini d1 esp32, everything works with espeasy everything works well for the moment, on the other hand to add it to my box domotic jeedom, it is well recognized also with espeasy plugin but in espeasy mega it does not want to register my controller publish command which is: device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%,
when i click on sumit it only keeps: device=%sysname% i don't know why.
Thank you in advance for your help.

TD-er, 2020-05-06:
Where are you placing this string?

And what build are you using?
It sounds like a report I had about a week ago where it appeared someone was using a really old build for ESP32.
So please check you're using one of the most recent builds (N.B. 20200424 does not work for ESP32, please try one older)

tiruds76, 2020-05-06:
I can only flash with the epseasy 32 R20100 version.
The command is placed in controller setting then controller publish as requested in the tutorials: https://projetsdiy.fr/plugin-espeasy-iot-jeedom/

I am on a wemos d1 mini esp32 the flash with the esp8266 versions does not work it always says flash failed.

TD-er, 2020-05-06:
That R20100 version is indeed the same ancient version as in a similar report of about a week ago.
It has been fixed in the newer builds.

I made you a new build based on the current state of the mega branch, so even newer than the last nightly builds.

- [ESP_Easy_mega_20200506_test_ESP32_4M316k-factory.bin](https://www.dropbox.com/s/siuh9iyrq8reu7j/ESP_Easy_mega_20200506_test_ESP32_4M316k-factory.bin?dl=0)
- [ESP_Easy_mega_20200506_test_ESP32_4M316k.bin](https://www.dropbox.com/s/z2i4fxim7l7svaw/ESP_Easy_mega_20200506_test_ESP32_4M316k.bin?dl=0)

The `-factory.bin` file is one that must be flashed on blank units, or at least one that hasn't had a recent version installed.
The offset to write should be set to 0 and no other bin files are needed to flash it.

The .bin file without `-factory` is the kind of file you need to flash via OTA (via the web interface)
Right now, you should only use the factory bin, as you don't have anything flashed on the node.

Please note that this build also has a different flash layout, so all settings will be lost when you flash this build.

tiruds76, 2020-05-07:
hello and thank you for your work I did a reset via arduino of my wemos d1 but I can't flash with your version which software used for falsh? I tried by putting the file in the folder of: ESPEasy_mega-20200426 but it does not want and the same with ESPEasy32_R20100

TD-er, 2020-05-07:
You can use the tool linked [here](http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/)
Just use the file with -factory.bin and set the offset (right colomn) to 0.

tiruds76, 2020-05-07:
thank you for your help I managed to flash apparently but on the other hand I do not have access to the web interface it does not activate the wifi at the first start like the R20100 version how to access the interface? I tried the ip of the device that I had assigned before but nothing.

tiruds76, 2020-05-07:
everything is ok thank you very much. on the other hand it is no longer recognized in inclusion mode on the espeasy plugin with jeedom I can't find why but I'm looking for it. esp easy web interface has evolved I may have forgotten to check a box but I can not find the reason however everything is as before.

TD-er, 2020-05-08:
Graag gedaan :)

closed 2020-05-08 by TD-er