letscontrolit/ESPEasy #3059: explanation of BUILD_GIT usage

fmuntean, 2020-05-07:
@TD-er: can you please explain the usage of BUILD_GIT='"${sysenv.TRAVIS_TAG}"' in the build process?
I would like to track the git check-in however when building inside the VSCode it  is empty and I endup always overriding it manually.

TD-er, 2020-05-07:
You can also define it in the Custom.h if you like.

What this does is, it takes the Git tag we set when making a nightly build in Travis.
So just these steps:
- Create changes summary file
- Commit
- Create tag in Git
- Push to `mega` branch (including tag)

When pushing a new tag, the build process in Travis is triggered to make a new nightly build.

The `${sysenv.TRAVIS_TAG}` is bash-like syntax for getting the currently set environment variable `TRAVIS_TAG`, which is set in our scripts as the result of asking Git for the latest tag.

fmuntean, 2020-05-07:
maybe this can be ported into python pre hook using the new build process?

TD-er, 2020-05-07:
Sure, but not right now.
I will try to remember to do when I continue on the build process.

First let's see how many more issues will be reported by users regarding the Python 3.8 update.

fmuntean, 2020-05-07:
I have been using Python 3.8 for few months now with ESP8266 with no issues on my side.
Had to change the firmware signature for the CRC32 only.

tonhuisman, 2022-05-07:
This seems outdated, so can probably be closed

closed 2022-05-07 by TD-er