letscontrolit/ESPEasy #3065: Pulse Counter shows many false triggers

caco3, 2020-05-13:
I have a rain gauge with a reed contact. The reed contact is connected to D5 and has an pull resistor.

I use the pulse counter device and in general it seems to work fine (I can let down water and see it counting up).

How ever I see many false pulse counts, even it is not raining at all.

For testing purpose I added a switch device to the same pin and let both sending me MQTT messages (see below).

I configured the pulse counter to use a Debounce Time of 300 ms and to (check if it is a debouncing problem) set the switch Debounce Time to 0ms. Nevertheless the pulse counter sees false triggers where as the switch shows only the real pulses.

### System configuration
Hardware: nodeMCU V3
ESP Easy version: mega-20200410

### Rules or log data
The pulse counter `Wassermesser` sends its events with a sending interval of 720s, whereas the switch `Wassermesser2` sends its message immediately.
The pulse counter total increases by 0.33 per pulse, where as the switch toggles between 0 and 1.
```
26204	|	2020-05-13	07:46:03	|	Wetterstation/Wassermesser2/Position	|	1					
26222	|	2020-05-13	07:56:56	|	Wetterstation/Wassermesser/Total	|	0.33					
26251	|	2020-05-13	08:15:26	|	Wetterstation/Wassermesser2/Position	|	0					
26260	|	2020-05-13	08:20:56	|	Wetterstation/Wassermesser/Total	|	0.66					
26351	|	2020-05-13	09:44:56	|	Wetterstation/Wassermesser/Total	|	5.94	=> many	false	pulses	triggered
26538	|	2020-05-13	12:36:22	|	Wetterstation/Wassermesser2/Position	|	1					
26540	|	2020-05-13	12:37:31	|	Wetterstation/Wassermesser/Total	|	0.33					
26749	|	2020-05-13	16:01:31	|	Wetterstation/Wassermesser/Total	|	0.66	=> 1	false	pulse	triggered
```

# configuration:
```
{
   "System":{
      "Build":20105,
      "Git Build":"mega-20200410",
      "System Libraries":"ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support",
      "Plugin Count":46,
      "Plugin Description":"[Normal]",
      "Local Time":"2020-05-13 18:43:25",
      "Unit Number":0,
      "Unit Name":"Wetterstation",
      "Uptime":414,
      "Last Boot Cause":"Manual reboot",
      "Reset Reason":"Exception",
      "Load":7.20,
      "Load LC":2895,
      "CPU Eco Mode":"false",
      "Heap Max Free Block":18120,
      "Heap Fragmentation":8,
      "Free RAM":19648
   },
   "WiFi":{
      "Hostname":"Wetterstation",
      "IP Config":"Static",
      "IP Address":"192.168.1.129",
      "IP Subnet":"255.255.255.0",
      "Gateway":"192.168.1.1",
      "STA MAC":"BC:DD:C2:30:CB:49",
      "DNS 1":"192.168.1.1",
      "DNS 2":"(IP unset)",
      "SSID":"RW",
      "BSSID":"74:83:C2:B1:8D:5E",
      "Channel":11,
      "Connected msec":24835057,
      "Last Disconnect Reason":2,
      "Last Disconnect Reason str":"(2) Auth expire",
      "Number Reconnects":0,
      "Force WiFi B/G":"true",
      "Restart WiFi Lost Conn":"true",
      "Force WiFi No Sleep":"true",
      "Periodical send Gratuitous ARP":"false",
      "Connection Failure Threshold":1,
      "RSSI":-77
   },
   "Sensors":[
      {
         "TaskValues":[
            {
               "ValueNumber":1,
               "Name":"Count",
               "NrDecimals":2,
               "Value":0.00
            },
            {
               "ValueNumber":2,
               "Name":"Total",
               "NrDecimals":2,
               "Value":0.66
            },
            {
               "ValueNumber":3,
               "Name":"Time",
               "NrDecimals":2,
               "Value":11853771.00
            }
         ],
         "DataAcquisition":[
            {
               "Controller":1,
               "IDX":0,
               "Enabled":"true"
            },
            {
               "Controller":2,
               "IDX":0,
               "Enabled":"false"
            },
            {
               "Controller":3,
               "IDX":0,
               "Enabled":"false"
            }
         ],
         "TaskInterval":720,
         "Type":"Generic - Pulse counter",
         "TaskName":"Wassermesser",
         "TaskDeviceNumber":3,
         "TaskEnabled":"true",
         "TaskNumber":1
      },
      {
         "TaskValues":[
            {
               "ValueNumber":1,
               "Name":"Temperature",
               "NrDecimals":1,
               "Value":18.6
            },
            {
               "ValueNumber":2,
               "Name":"Humidity",
               "NrDecimals":0,
               "Value":73
            },
            {
               "ValueNumber":3,
               "Name":"Pressure",
               "NrDecimals":0,
               "Value":1010
            }
         ],
         "DataAcquisition":[
            {
               "Controller":1,
               "IDX":0,
               "Enabled":"true"
            },
            {
               "Controller":2,
               "IDX":0,
               "Enabled":"false"
            },
            {
               "Controller":3,
               "IDX":0,
               "Enabled":"false"
            }
         ],
         "TaskInterval":360,
         "Type":"Environment - BMx280",
         "TaskName":"BME280",
         "TaskDeviceNumber":28,
         "TaskEnabled":"true",
         "TaskNumber":2
      },
      {
         "TaskValues":[
            {
               "ValueNumber":1,
               "Name":"Position",
               "NrDecimals":0,
               "Value":1
            }
         ],
         "DataAcquisition":[
            {
               "Controller":1,
               "IDX":0,
               "Enabled":"true"
            },
            {
               "Controller":2,
               "IDX":0,
               "Enabled":"false"
            },
            {
               "Controller":3,
               "IDX":0,
               "Enabled":"false"
            }
         ],
         "TaskInterval":0,
         "Type":"Switch input - Switch",
         "TaskName":"Wassermesser2",
         "TaskDeviceNumber":1,
         "TaskEnabled":"true",
         "TaskNumber":11
      },
      {
         "TaskValues":[
            {
               "ValueNumber":1,
               "Name":"uptime",
               "NrDecimals":0,
               "Value":410
            },
            {
               "ValueNumber":2,
               "Name":"rssi",
               "NrDecimals":0,
               "Value":-77
            }
         ],
         "DataAcquisition":[
            {
               "Controller":1,
               "IDX":0,
               "Enabled":"true"
            },
            {
               "Controller":2,
               "IDX":0,
               "Enabled":"false"
            },
            {
               "Controller":3,
               "IDX":0,
               "Enabled":"false"
            }
         ],
         "TaskInterval":600,
         "Type":"Generic - System Info",
         "TaskName":"SysInfo",
         "TaskDeviceNumber":26,
         "TaskEnabled":"true",
         "TaskNumber":12
      }
   ],
   "TTL":60000
}
```

## Rules
```
On Clock#Time=All,00:00 Do
  ResetPulseCounter,1
  Publish %sysname%/Wassermesser/Total,0
EndOn

on System#Boot do
  Publish %sysname%/Boot,0
endon

 	

on MQTT#Connected do
 Publish %sysname%/status,0
endon

on MQTT#Disconnected do
 Publish %sysname%/status,0
endon
````

tonhuisman, 2020-05-13:
- What does the wiring from NodeMCU to the sensor look like? (shielded or not)
- Have you added an external pull-up resistor to the input? (internal pull-up is usually too weak, so spurious spikes tend to cause false counts, assuming a low-active counter output)

caco3, 2020-05-13:
>     * What does the wiring from NodeMCU to the sensor look like? (shielded or not)
> 
>     * Have you added an external pull-up resistor to the input? (internal pull-up is usually too weak, so spurious spikes tend to cause false counts, assuming a low-active counter output)

I have 40cm unshielded cables, so that's for sure not perfect.
How ever since the normal switch has no false triggers (and I expect it is also sensitive to glichtes), I doubt it is hardware related!

It would surprise me if the Pulse Counter is implemented much different than a switch low level wise.

TD-er, 2020-05-14:
Well 40 cm wire sounds like a nice antenna.
Also a reed contact may have more resistance compared to a normal switch, plus it may also act slightly on mechanical vibrations.

So I would suggest the standard tips for these:
- Twist the wires (lessen the effect of electromagnetic interference)
- Use a pull-up resistor to make the signal more 'strong'. (higher current is needed to flip the state)
- Add a small capacitor over the switch, if the state change is only for longer periods. This filters out short pulses.

caco3, 2020-05-14:
Thanks for your suggestions. I know that it is not ideal. If needed, I will improve it (electronics engineering and MCU programming is my daily business)! 

I would prefer if you could give some insight why a switch (with or without debouncing) works very well, but the Pulse Counter gives false signals from time to time.

Since they are connected to the same GPIO and get triggered by the same ISR, there shouldn't be a difference in it!

TD-er, 2020-05-14:
Are they also using the same plugin?
P001 does use filtering.

caco3, 2020-05-15:
> Are they also using the same plugin?

As expected, they do not.

I do not fully understand how it works, but on plugin level the switch seems to process the pin with 10 Hz:
https://github.com/letscontrolit/ESPEasy/blob/dd4ac70c1128fdad702cb63bd337fca0b2599a56/src/_P001_Switch.ino#L439

Where as the Pulse Counter plugin somehow gets directly triggered by the IRQ:
https://github.com/letscontrolit/ESPEasy/blob/dd4ac70c1128fdad702cb63bd337fca0b2599a56/src/_P003_Pulse.ino#L303

This leads me to the understanding, that the Pulse Counter is able to process faster pin level changes as the Switch plugin.
This is IMO ok, how ever I wish the plugin documentations would briefly mention it!

As for me, I use now a Switch Device and wrote a Pulse Counter rule around it:
<details>
<summary>Simple Rules and Switch Device based Pulse Counter</summary>

```
// Own implementation of a Pulse Counter
// Var1 contains the total per interval (10 minutes)
// Var2 contains the total per day 
// Var3 contains the rain gauge bucket size in mm

on System#Boot do
  let,1,0 // Interval count
  let,2,0 // Day count
  let,3,0.33 // Rain gauge bucket size in mm
endon

// On the GPIO change, increment both variables
on Regenmenge#bucket_tilted do
  let,1,[VAR#1]+[VAR#3]
  let,2,[VAR#2]+[VAR#3]
  logentry,"Bucket full, new interval count: %v1%, new day count: %v2%"
endon

// Every 10 minutes send both variables using MQTT and reset var1
on Clock#Time=All,**:*0 do // Every 10 minutes
  Publish %sysname%/Regenmenge/interval,[VAR#1]
  Publish %sysname%/Regenmenge/day,[VAR#2]
  logentry,"Interval is up reset interval counter (%v1%), day count: %v2%"
  let,1,0 // Reset interval counter

  If %systime% = 00:00
    logentry,"Its midnight, reset day counter as well"
    let,2,0
  endif
endon
```
</details>

### Idea
Add a "safe button" mode to the Pulse Counter so it has the same low pass filtering behaviour as the Switch plugin.

TD-er, 2020-05-15:
> Add a "safe button" mode to the Pulse Counter so it has the same low pass filtering behaviour as the Switch plugin.

That's already been discussed before as there are also some issues (mainly from the maintainer perspective) with the current switch plugin (P001).
That plugin is trying to be a Swiss army knife and thus makes things a lot more complex then it has to be.
So the plan is to split and merge those plugins so they also share the same de-bounce code and options.

It is a work-in-progress.

closed 2020-05-16 by caco3