letscontrolit/ESPEasy #3068: Bug in Plugin 008: RFID - Wiegand

GitCybers, 2020-05-14:
Hello,

i think there is a bug in the plugin 008 RFID - Wiegand.
If I type the Code 1234# in the keypad of my wiegand-device, ESPEasy shows me 4660. This is an hex/dec-error. It seems that the Keypad input is in hex and the ESPEasy converts it in decimal.
Is there the possibility to change the output in hex or pehaps a checkbox between hex/dec?

Thank, Sascha

TD-er, 2020-05-14:
Indeed, 0x1234 = 4660 dec.

Right now plugins/tasks don't support HEX output, but maybe we can convert the input value?
That part is probably done in the code anyway.

TD-er, 2020-05-14:
I was thinking, maybe you could look at HEX conversion in the rules?
See [documentation](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=hex#strtol)

GitCybers, 2020-05-15:
Thanks, I can try it with rules but it would be much nicer if I could change it with one click. ;)

TD-er, 2020-05-15:
I know, but rules are present in the current build and changes like you want are not and also not (yet) in the next :)

tonhuisman, 2020-05-17:
As far as I can find, these type of keypads send their entered value as hex (most likely because of the number of hex keypads in use). https://ipvm.com/forums/video-surveillance/topics/wiegand-keypad-reader (first reply)
So I'd rather not call this a bug but a feature request. And you can of course use the proposed workaround :-)

tonhuisman, 2020-07-04:
I added an option to enable the conversion you described here

PR #3149 (2020-07-19): [P008] Add option to convert value to 'hex' (#3068), and updated docs
Resolves #3068 

Tag value can optionally be converted to hex by discarding any nibble > A to 0, and treating each nibble as decimal, resulting in value 1234 if 1234 was entered on a keypad, instead of 4660 (0x1234).
closed 2020-07-19 by TD-er