letscontrolit/ESPEasy #3071: Blynkget saving values ​​to Dummy_Device does not work.

matryc182, 2020-05-15:
In newer ESPEasy releases (I mean from 2019 and newer), the Blynkget option does not write value to Dummy_Device. I use ESP8266 (sonoff 4Mb mod flash) and tested many versions of ESPEasy. Going back to old versions of ESPEasy noticed that it worked well until November 2018 - I tested for example ESP_Easy_mega-20180904_test_ESP8266_4096.bin and the log looked like this:
...
Command: blynkget
GET / xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx / get / V9 HTTP / 1.1
 Host: blynk-cloud.com
 Connection: close
HTTP: Success
HTTP: Success
HTTP: Success
HTTP: Success
HTTP: Success
HTTP: Success
Timeout while reading input data!
HTTP: Success
Blynk get - get / V9 => 9
HTTP: closing connection
...
Dummy: value 1: 9.00
...
I have a button set in Blynk with the values ​​Off = 9 On = 10
ESPEasy gets the value of the button and controls the relay in my sonoff.
In newer versions after November 2018, my configuration no longer controls the relay. Checking the problem I looked in the log. It looks like the value in Dummy_Device after the command is set to 0. Even if I manually set "http://IP/control?Cmd=TaskValueSet,1,1,5" and I see in the logs:
...
Dummy: value 1: 5.00
...
then after executing the command "BlynkGet 1,1, get / V9" I see that
...
Dummy: value 1: 0.00
...
The log in newer versions such as ESP_Easy_mega-20190305_test_core_242_ESP8266_4M.bin looks like this:
...
Command: blynkget
BlynkGet 1.1, get / V9
HTTP: C012 connecting to blynk-cloud.com:80
GET / xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx / get / V9 HTTP / 1.1
 Host: blynk-cloud.com
 Connection: close

HTTP: closing connection (012)
...
Dummy: value 1: 0.00
...
nie ma nic w logach w odczytanej wartości.

njozsef, 2021-01-14:
check ack option on controller tab. 
working with wemos mini d1 (4M1M)
Firmware
--
Build:⋄ | 20111 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 45 [Normal]
Build Origin: | Self built
Build Time:⋄ | Jan 14 2021 00:01:36
Binary Filename:⋄ | ESP_Easy_mega_20210114_normal_ESP8266_4M1M
Build Platform:⋄ | Windows-10-10.0.19041-SP0
Git HEAD:⋄ | pygit2_not_installed

[image]

tonhuisman, 2023-09-02:
This seems to be resolved, AFAICS it can be closed.

closed 2023-09-02 by TD-er