letscontrolit/ESPEasy #3073: Ventus W266 Build not working

TheItschi, 2020-05-16:
It looks like all the new releases have the same issue again:  

> @chunter1
> I've tested the updated plugin with mega-pre-release 2018-11-07 and it works like it should.
> Great!
> Thanks for the short response

Hi @chunter1, I'm struggeling with the same issue trying to use recent releases. Data ist received but  no update of the devices. Would it be possible to do the same fix again? Regards, Itschi

_Originally posted by @TheItschi in https://github.com/letscontrolit/ESPEasy/issues/2020#issuecomment-629710530_

tonhuisman, 2020-05-17:
What version of ESPEasy are did you use here?
And what version have you used last that worked as expected?

TheItschi, 2020-05-17:
The version I’m currently using is a custom build I found in the Domoticz forum:

 

Firmware

Build

20100 - Mega (core 2_4_1)

GIT version

(custom)

Plugins

2 [Normal] [Testing] [Development]

Build Md5

4d44355f4d44355f4d44355f4d44355f

Md5 check

fail !

Build time

Apr 21 2018 07:31:07

Binary filename

ThisIsTheDummyPlaceHolderForTheBinaryFilename...

ESP board

ESP Chip ID

7250864 (0x6EA3B0)

ESP Chip Freq:

80 MHz

 

It works,  but there is no update of “Lightning” and some data has no decimal digits like UV-Index, even so it shows one decimal digit on the receiver. That’s why I tried to use a new version. I tested several newer builds and there the “ESP_Easy_mega-20200426_hard_Ventus_W266.bin”. The latest I tried is ESPEasy_mega-20200426.

 

Regards, Yavuz

tonhuisman, 2020-05-18:
Please use a more current version of ESPEasy, as that (custom) build is over 2 years old, and not a representation of the current state of development.
Get the most current release from the releases page: https://github.com/letscontrolit/ESPEasy/releases

TheItschi, 2020-05-18:
I alreday tried the latest version from Github. The last one I tried isRelease mega-20200426 from 26th of April 2020. This one also donn’t work. It shows data coming from the Weather Station in the Log, but the device data is not updated.

 

Regrads, Yavuz

tonhuisman, 2020-05-18:
> And what version have you used last that worked as expected?

Can you answer this question too, please?

chunter1, 2020-05-18:
This is what my system is running on:

```
Build:⋄ | 20102 - Mega
Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version:⋄ | (custom)
Plugins:⋄ | 74 [Normal] [Testing]
Build time:⋄ | Nov 9 2018 
Binary filename:⋄ | Self built!
```

TheItschi, 2020-05-18:
Do you mind to send me the binary?

TheItschi, 2020-05-18:
It’s 

 

Build

20100 - Mega (core 2_4_1)

GIT version

(custom)

Plugins

2 [Normal] [Testing] [Development]

Build Md5

4d44355f4d44355f4d44355f4d44355f

Md5 check

fail !

Build time

Apr 21 2018 07:31:07

chunter1, 2020-05-18:
> Do you mind to send me the binary?

Please use the information provided above to build it yourself.
I'm not into espeasy anymore.

TD-er, 2021-10-19:
Not sure if it is a RAM issue or more like flash build size.
But yes, that's mentioned in the comments :)
Just to be sure, you are using the plugin called "Hardware - Ventus W266 [TESTING]" ?
I can make you a test build if you want.

TD-er, 2021-10-19:
[test build](https://www.dropbox.com/s/70pvdbwaqw5w73i/ESP_Easy_mega_20211019_hard_Ventus_W266.bin?dl=0)

TheItschi, 2021-11-07:
> YES! it works, thank you very much! [image] more ko-fi is on the way ;-)

Beside the UV-INdex issue: Are Updates now comming on a regular basis? I also had the poblem, that it sometimes takes 30min for the next update...

TD-er, 2021-11-07:
What value do you have for the "interval" setting on that task?

mobilegrundy, 2021-11-07:
this plugin does not have an interval, it updates everytime it recieves a data frame from the outside sensor-> and thats the problem, sometimes i get one after 2-3 mins, then nothing for 30 mins, maybe some other 868mhz device sends the same time and breaks it, i have a lot in use. but it updates more often when its closer together.
so if there would be an interval and no new frame is received it should output an error value or just the last value again, dont know whats better. in my software i just have a hole in the plot when nothing is receved, which is not pretty but i can live with that :-) (about 3-4 missing values per day , recording every 10 minutes)

TD-er, 2021-11-07:
You can also store the values in a dummy task and periodically "run" that task from the rules via "taskrun" so it will output the values stored in the dummy task at the interval you like.
By acting on the events sent by this task for the Ventus, you can then immediately copy the values to the dummy task and for the periodical values in your logging system you can create a looptimer.

TheItschi, 2021-11-07:
The problem with the missing values is that I need close to real time data. E.g. I use the wind speed to retract the blinds if it gets too windy. When looking at the Ventus device for me it looks like updateting the values at a very short timeperiod. However the ESP doesn’t seem to decode each packet received. So I belive it’s  an issue with the plugin.

 

Regards, Itschi

tonhuisman, 2021-11-07:
> but it updates more often when its closer together.

> So I belive it’s an issue with the plugin.

Well, I must disagree with your second statement, because if it is working better if you place them closer together, it must be caused by poor signal quality, there's nothing software can do about that.

TheItschi, 2021-11-08:
Hm, but how does that fit to fact that the Receiver-Device updates the data on it‘s display whereas the ESP doesen‘r transmit for a longer time…