letscontrolit/ESPEasy #3074: PWM value - rules

nooa85, 2020-05-18:
hi,
how to set pwm value to Dummy Device in rules.

rules:
On test#state do
   TaskValueSet,3,1,[pwm,1]
endon

fraeggle, 2020-06-20:
Hi.
I am using this for ex. an LED Stripe controlled via broker and MQTT

Device:
MQTT IMPORT Device 11:  switch#Wertxx
//value2: switch#Wert2 --> Ausgang für LED_Stripe
//value3: switch#Wert3 --> PWM Wert

dummy Device 12:  Status#xxxx
//value2:  Status#Power --> LED_Leiste an/aus
//value3:  Status#PWM --> LED Helligkeit

Rules:

on switch#Wert2 do
 if [switch#Wert2] = 1
   PWM,14,[switch#Wert3],1000
   TaskValueSet,12,2,1
   TaskValueSet,12,3,[switch#Wert3]
   Publish,%sysname%/Power/Power,1
 else
   PWM,14,0,1000
   TaskValueSet,12,2,0
   Publish,%sysname%/Power/Power,0
 endif
endon

on switch#Wert3 do
 if [Status#Power]=1
   PWM,14,[switch#Wert3],1000
 endif
 TaskValueSet,12,3,[switch#Wert3]
endon

tonhuisman, 2023-09-01:
This seems to be solved by the PWM PinState GPIO status feature. Can be closed.

closed 2023-09-01 by TD-er