letscontrolit/ESPEasy #3075: ESP_Easy_mega_20200516_minimal_core_242_ESP8266_1M_OTA GPIO 15 = 1 after reboot

fly74, 2020-05-18:
GPIO 15 starts with high after reboot in ESP_Easy_mega_20200516_minimal_core_242_ESP8266_1M_OTA

fly74, 2020-05-18:
Config is relative easy and worked in the release before

[image]
[image]
[image]

tonhuisman, 2020-05-18:
Is this a question, a challenge or a problem?

fly74, 2020-05-18:
It's a problem. When the power supply is short disconnected all switches are on after the power comes back. Nice for christmas trees :), but ...

TD-er, 2020-05-18:
Just curious, what was the last state of the GPIO pin before the reboot?

I've seen a PR/issue a while ago in the esp8266/Arduino repo about GPIO pins toggle at reboot.
So maybe they finally tried to fix it to keep a pin steady at reboot and then here it is seen as a bug ;)

Not sure if that's the reason, but just an idea popping up in my head as regarding GPIO pins I only changed the core lib from 2.6.3 to 2.7.0

fly74, 2020-05-18:
It doesn't matter of it was 1/0 it is alway 1 after reboot. :/

TD-er, 2020-05-18:
Hmm, too bad, now I really have to dig into the code to see what's changed.

fly74, 2020-08-02:
Sadly still the same with ESP_Easy_mega_20200801_minimal_core_242_ESP8266_1M_OTA

fly74, 2020-11-23:
Is there any progress or can you recommend a old version where it worked before?

TD-er, 2020-11-24:
I'm currently working on "loose end issues" to finally put a release label on it.
This does look like such an issue.

Just to be sure about what's happening, no matter what you did before the reboot, GPIO 15 (only that one?) is high after a reboot?

fly74, 2020-11-24:
Correct, it is a device with only one switch (GOSUND), everytime high after reboot. before ESP_Easy_mega_20200516_minimal_core_242_ESP8266_1M_OTA ist was working with low after reboot.

TD-er, 2020-11-24:
And you still used the core 2.4.2 builds? (for as long as they were present in the builds)
The only main change related to GPIO handling I found between the last build before 20200516 and tag "20200515" is about the switch from core 2.6.2 to 2.7.0

There is one minor change I noticed in the scheduler code.
Can you show the settings of the switch task?
How does the unit reboot? Power cycle or a warm reboot (e.g. crash)?

fly74, 2020-11-24:
The Info:

Unit Number: | 0
-- | --
Local Time: | 2020-11-24 12:14:44
Uptime: | 0 days 0 hours 0 minutes
Load: | 100.00% (LC=2)
CPU Eco Mode: | false
Free RAM: | 26976 (37104 - )
Free Stack: | 3712 (3072 - )
Boot: | Cold boot (0)
Reset Reason: | External System
Last Task: | Background Task
SW WD count: | 0
Network ❔
Wifi: | 802.11N (RSSI -44 dB)
IP Config: | DHCP
...
Channel: | 1
Connected: | 18 s
Last Disconnect Reason: | (1) Unspecified
Number Reconnects: | 0
WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Firmware
Build:⋄ | 20110 - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 3 [Normal] [Minimal 1M OTA]
Build Origin: | Travis
Build Time:⋄ | Aug 29 2020 16:07:11
Binary Filename:⋄ | firmware.bin
Build Platform:⋄ |  
Git HEAD:⋄ |  
System Status
Syslog Log Level: | None
Serial Log Level: | Info
Web Log Level: | None
ESP Board
ESP Chip ID: | 10801447 (0xA4D127)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP01_1M
Storage
Flash Chip ID: | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 571 kB (304 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | true
OTA 2-step Needed: | true
SPIFFS Size: | 110 kB (28 kB free)
Page size: | 256
Block size: | 4096
Number of blocks: | 27
Maximum open files: | 5
Maximum path length: | 32

fly74, 2020-11-24:
The Rule:
On System#Boot do
    GPIO,0,0
    GPIO,2,1
    GPIO,15,0
endon

On pon do
 gpio,0,1
 gpio,2,0
 gpio,15,1
endon

On poff do
 gpio,0,0
 gpio,2,1
 gpio,15,0
endon

On SW#State do
if [REL#State] = 0 then
    Event,pon
else
    Event,poff
endif
enddo

fly74, 2020-11-24:
Devices:
[image]

TD-er, 2020-11-24:
And the settings of the switch tasks? 
What if the switch of task 2 tries to set its value and fires an event, that one will then trigger the `On SW#State do` event handler, which switches GPIO 15 to high.

For debugging purpose, can you disable this part in the rules and trigger a reboot?

fly74, 2020-11-24:
Damn you're right it was the rule. SW was inverse so

if [REL#State] = 1 then

thx

closed 2020-11-24 by fly74

TD-er, 2020-11-24:
And another issue bites the dust :)

fly74, 2020-11-24:
:) but https://github.com/letscontrolit/ESPEasy/issues/3388 still open