letscontrolit/ESPEasy #3077: Prevent Deepsleep in Rule

sl6der, 2020-05-18:
**Feature Request**
**_Disable DeepSleep in IF-Condition in Rules._**

Sometimes it´s neccessary and absolutly useful to prevent the DeepSleep state in a conditional state in the rules section.

_**For an example:**_
You has a conditional event, that is triggered from a sensor value. The actor of the event should do the job until another sensor value has been triggered.

**- Is-State:**
The event has been triggered by the sensor value, but the DeepSleep kicks in and the actor do his job until the next awake from DeepSleep has been set.
Or the actor can do his job only for the set awake time and is going to sleep again although the actor hasn´t finished his job.

**- Future-State:**
The sensor value triggered the event successfully after the next awake time. The actor can do his job until the if-condition finished and the event has been closed.
The software return to his sleep mode.

Another explaination in the forum:
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=7754

stefbo27, 2020-05-18:
Doesn't 
`nosleep,0` 
do what you want? I used it successfully to prevent from going to deepsleep. However it is not fully documented. I came across it accidentially...

sl6der, 2020-05-18:
> Doesn't
> `nosleep,0`
> do what you want? I used it successfully to prevent from going to deepsleep. However it is not fully documented. I came across it accidentially...

Really? 
I can't find anything in the code. Probably I looked into every single file if the code.
But eventually you're right. I found this "nosleep" command in Google in context to esp8266.
Tomorrow I will try it and letting you know if this works for me too.