letscontrolit/ESPEasy #3078: ESP_Easy_mega_20200516 version not working on 3/5 ESP8266 modules

blb4github, 2020-05-19:
I update all 5 modules with the latest version:
ESP_Easy_mega_20200516_test_beta_ESP8266_4M1M

2 of them are working fine and do survive a reboot.
The other 3 did come back after reboot, but stop functioning after a short time. 
Resetting or power down/up these modules did not help, sometimes the module came active for a short moment but stopped working after some time again.
At the end I did connect all 3 to pc to upload the former version (ESP_Easy_mega-20200426_test_beta_ESP8266_4M1M.bin) to get them back into service.

TD-er, 2020-05-19:
Can you give the flash chip ID?
It is either available via the flash tool, or the tools -> sysinfo page.

If the vendor ID equals 0x20, then it is an XMC brand flash and then it is a known issue (working on it)

blb4github, 2020-05-19:
You are right, the 2 which functions ok with the latest firmware do have 0xC8, the failing once do have 0x20

bojanul, 2020-06-17:
Is this fixed?

TD-er, 2020-06-17:
It should be.
The last build should be based on the core 2.7.1 which should have fixes for the XMC flash modules

TD-er, 2020-06-28:
This can be closed as it has been fixed in the last build.

blb4github, 2020-06-28:
Hi @TD-er , I have tried again to load the latest version (ESP_Easy_mega_20200608_test_beta_ESP8266_4M1M) on 1 of my modules with Vendor: 0x20 and it's not functioning. Going back to ESP_Easy_mega-20200426_test_beta_ESP8266_4M1M.bin and it works again.

TD-er, 2020-06-28:
That's "test beta", which is using the staging version of the core lib.
So that should be working as it is for sure the latest version of the esp8266/Arduino repo.

Can you try any of these latest test builds I made the last week for some other PRs?

- [ESPEasy_ESP82xx_mega-20200608-92-PR_3134.zip](https://www.dropbox.com/s/iloa6d2yu183599/ESPEasy_ESP82xx_mega-20200608-92-PR_3134.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200608-89-PR_3050.zip](https://www.dropbox.com/s/7xunol3igp81lwq/ESPEasy_ESP82xx_mega-20200608-89-PR_3050.zip?dl=0)

blb4github, 2020-06-29:
@TD-er I've tried test_beta_ESP8266_4M1M version, both are not booting.

TD-er, 2020-06-29:
Can you also test the non-beta versions in those ZIP files?

TD-er, 2020-06-29:
Oh and another thing...
What does seem to be related is that there is a difference between OTA update and flashing via serial.

I just browsed through the XMC-related issues in [esp8266/Arduino](https://github.com/esp8266/Arduino/issues?q=is%3Aissue+xmc+is%3Aclosed) and it has been mentioned a few times that the flags in the boot loader are not always updated when performing an OTA update.
So could you also try to flash the -beta build in those zip files via serial and see if that makes the units boot?
If that works, then you could try to OTA using any build and back to the latest nightly build without issue....
At least if my theory is correct.

blb4github, 2020-06-29:
I have also tested 2 non-beta versions, I did get same result; not booting module.
I have now flashed via serial first and now it's working! I did flash via serial ESP_Easy_mega_20200625_test_beta_ESP8266_4M1M which was successful. After that I did OTA 'update' to ESP_Easy_mega_20200608_test_beta_ESP8266_4M1M and this was successful as well; module is booting successful!
So it seems to get over this issue with Vendor: 0x20 one time flashing via serial is required.

Thanks!

TD-er, 2020-06-29:
Thanks for testing

closed 2020-06-29 by TD-er