letscontrolit/ESPEasy #3080: MQTT issues

TungstenE2, 2020-05-20:
Hi all,

why is setting up MQTT on ESPeasy so complicated compared to Tasmota?

In Tasmota I was done in 2 Min.
In ESPeasy there are several MQTT options in Protocol with several settings.

I choose HomeAssistant MQTT, but even after several attemts I am only able to retrieve data in FHEM as MQTT2_device,
but having same Topic structure as in Tasmota or sending MQTT commands to ESPeasy is just not working for me, after several attempts.

Any advice how to solve this?

Tasmota:
/SmartHome/Steckdosen/%topic%/%prefix%/
in FHEM:
/SmartHome/Steckdosen/Sonoff-S20-01/tele/STATE:.* { json2nameValue($EVENT) }

ESPeasy:
/SmartHome/%tskname%/%valname%/
in FHEM:
ESP_Easy5:/SmartHome/Klimaanlage/settings/:.* { json2nameValue($EVENT) }

e.g.: Why does ESPeasy set the CID in front of the topic?
Why does ESPeays use different namings? I thought 'topic' is a standard convention in MQTT naming?

Never really paid attention, but now I find 'tskname' for the devices in the list of devices also confusing. Task in List of Devices should be renamed to ID
and %tskname% should be %devicename%.

thx

TD-er, 2020-05-21:
Well I agree on the part that the "Devices" tab in the web UI is confusing.

Just a summary of the names used in ESPEasy:
- Plugin: A piece of code in ESPEasy to interact with a specific piece of hardware.
- Task: A selected instance of a specific plugin. 
- Controller: Interface to send sample values from a task to a specific target (e.g. Domoticz MQTT)

So a task has a plugin selected and 0 .. 3 controllers to which the output should be sent.

This means the name `%tskname%` is correct here.
For example if you have a BME280 sensor.
This is supported by the BME280 plugin.
So you make a task for it by selecting the BME280 plugin and for example name your task "bme". (or "garden" or whatever name you give it)
Then `%tskname%` will be "bme" for this one when it sends data to the controller.

So I don't agree the naming should be changed into "device" instead of "task", as you can have multiple tasks all using an instance of the same plugin.
"device" is then a bit too generic and not clear whether it reflects to what we now call "plugin" or "task".

About the host name shown in FHEM.
As far as I know, we don't put the hostname in the topic, but the hostname is used as a unique MQTT client ID. So maybe FHEM does put the MQTT client ID in front of the topic?

For MQTT we have a number of standard names to describe things and as far as I know , we only use the common names for it.
So a topic is also what we call a topic.
You subscribe to a topic to get messages published to it and you can publish yourself to a topic.

I don't know what Tasmota supports for MQTT.
In ESPEasy, a controller is to push data out (publish) and to get data in, we have the MQTT import plugin.
There is a small number of exceptions to this. For a few plugins (e.g. P001_switch) there is support is some controllers to also receive data from the MQTT controller.
But in general, a controller -as we now have implemented it- only outputs data.

TD-er, 2022-04-01:
Not sure if this is still an issue, Please let me know if it still is.
For now, I'll close the issue

closed 2022-04-01 by TD-er