letscontrolit/ESPEasy #3081: mega-20200426 - Rules disappearing

TungstenE2, 2020-05-20:
Hi all,

I found my chicken door not working last night.
This morning I was investigating and it seems like all rules are gone.

It was working yesterday morning fine and I did not do any changes to the rules.

Any idea how this can happen? Seems like the D1 mini rebbots sometimes by itself.
Can this cause a loos of rules? All other settings and devices are still available.

Not sure if I have the latest rules version as backup on my pc.

[image]

IP/rules/backup is just bringing up an empty page.

[image]

thx

TungstenE2, 2020-05-20:
just tried to reinstall mega-20200426 but the Rules page still looks wrong and empty.

On my other devices it still looks good.

[image]

Anybody knows what causes this issue and how to fix?

TungstenE2, 2020-05-20:
Rules Settings/Old Engine was disabled.

This caused Rules to disappear and destroy Rules page.

Is this a bug or a feature?

TD-er, 2020-05-21:
> Rules Settings/Old Engine was disabled.

Hmm that's strange as it isn't even the default setting for it.
That was the only issue here?
So switching it back and it all worked again?

TungstenE2, 2020-05-22:
yes, switching back to old engine fixed the problem.

To be honest, it was me disabeling the setting the day before, not knowing about the function.

Tried to past old rules into the new event window, but always got an error on the name.

So switching back to old engine solved my issue.

TD-er, 2020-05-22:
Thanks for 'confessing' you made the change as it made me a bit worried how it could have changed on its own.

So this one can be closed.

closed 2020-05-22 by TD-er