letscontrolit/ESPEasy #3083: ESP32: disabling ARDUINO_OTA fails the buil

fmuntean, 2020-05-20:
I am trying to disable the ARDUINO_OTA feature for ESP32 board by commenting out the
#define FEATURE_ARDUINO_OTA from inside the ESPEasy_common.h 

The build fails with the following errors:
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino: In function 'String SaveSettings()':
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:263:3: error: 'MD5Builder' was not declared in this scope
   MD5Builder md5;
Compiling .pio\build\MFD_ESP32_4M316k\src\src\Commands\MQTT.cpp.o
   ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:296:3: error: 'md5' was not declared in this scope
   md5.begin();
   ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino: In function 'String LoadSettings()':
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:344:3: error: 'MD5Builder' was not declared in this scope
   MD5Builder md5;
Compiling .pio\build\MFD_ESP32_4M316k\src\src\Commands\Networks.cpp.o
   ^
Compiling .pio\build\MFD_ESP32_4M316k\src\src\Commands\Notifications.cpp.o
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:373:3: error: 'md5' was not declared in this scope
   md5.begin();
   ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino: In function 'String getPartitionType(byte, byte)':
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1263:3: error: 'esp_partition_type_t' was not declared in this scope
Compiling .pio\build\MFD_ESP32_4M316k\src\src\Commands\RTC.cpp.o
   esp_partition_type_t partitionType       = static_cast<esp_partition_type_t>(pType);
   ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1264:3: error: 'esp_partition_subtype_t' was not declared in this scope
   esp_partition_subtype_t partitionSubType = static_cast<esp_partition_subtype_t>(pSubType);
   ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1266:7: error: 'partitionType' was not declared in this scope
   if (partitionType == ESP_PARTITION_TYPE_APP) {
       ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1266:24: error: 'ESP_PARTITION_TYPE_APP' was not declared in this scope
   if (partitionType == ESP_PARTITION_TYPE_APP) {
                        ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1267:10: error: 'partitionSubType' was not declared in this scope
     if ((partitionSubType >= ESP_PARTITION_SUBTYPE_APP_OTA_MIN) &&
          ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1267:30: error: 'ESP_PARTITION_SUBTYPE_APP_OTA_MIN' was not declared in this scope
     if ((partitionSubType >= ESP_PARTITION_SUBTYPE_APP_OTA_MIN) &&
                              ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1268:29: error: 'ESP_PARTITION_SUBTYPE_APP_OTA_MAX' was not declared in this scope
         (partitionSubType < ESP_PARTITION_SUBTYPE_APP_OTA_MAX)) {
                             ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1274:13: error: 'partitionSubType' was not declared in this scope
     switch (partitionSubType) {
             ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1275:12: error: 'ESP_PARTITION_SUBTYPE_APP_FACTORY' was not declared in this scope
       case ESP_PARTITION_SUBTYPE_APP_FACTORY: return F("Factory app");
            ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1276:12: error: 'ESP_PARTITION_SUBTYPE_APP_TEST' was not declared in this scope
       case ESP_PARTITION_SUBTYPE_APP_TEST:    return F("Test app");
            ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1281:7: error: 'partitionType' was not declared in this scope
   if (partitionType == ESP_PARTITION_TYPE_DATA) {
       ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1281:24: error: 'ESP_PARTITION_TYPE_DATA' was not declared in this scope
   if (partitionType == ESP_PARTITION_TYPE_DATA) {
                        ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1282:13: error: 'partitionSubType' was not declared in this scope
     switch (partitionSubType) {
             ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1283:12: error: 'ESP_PARTITION_SUBTYPE_DATA_OTA' was not declared in this scope
       case ESP_PARTITION_SUBTYPE_DATA_OTA:      return F("OTA selection");
            ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1284:12: error: 'ESP_PARTITION_SUBTYPE_DATA_PHY' was not declared in this scope
       case ESP_PARTITION_SUBTYPE_DATA_PHY:      return F("PHY init data");
            ^
D:/git/HomeAutomation/ESPEasy/src/ESPEasyStorage.ino:1285:12: error: 'ESP_PARTITION_SUBTYPE_DATA_NVS' was not declared in this scope
       case ESP_PARTITION_SUBTYPE_DATA_NVS:      return F("NVS");
[... 89 lines cut ...]
   if (_mypartiterator) {
       ^
D:/git/HomeAutomation/ESPEasy/src/WebServer.ino:1157:7: error: '_mypart' was not declared in this scope
       _mypart = esp_partition_get(_mypartiterator);
       ^
D:/git/HomeAutomation/ESPEasy/src/WebServer.ino:1157:50: error: 'esp_partition_get' was not declared in this scope
       _mypart = esp_partition_get(_mypartiterator);
                                                  ^
D:/git/HomeAutomation/ESPEasy/src/WebServer.ino:1168:67: error: 'esp_partition_next' was not declared in this scope
     } while ((_mypartiterator = esp_partition_next(_mypartiterator)) != NULL);
                                                                   ^
D:/git/HomeAutomation/ESPEasy/src/WebServer.ino:1171:34: error: '_mypartiterator' was not declared in this scope
   esp_partition_iterator_release(_mypartiterator);
                                  ^
D:/git/HomeAutomation/ESPEasy/src/WebServer.ino:1171:49: error: 'esp_partition_iterator_release' was not declared in this scope
   esp_partition_iterator_release(_mypartiterator);
                                                 ^
D:/git/HomeAutomation/ESPEasy/src/WebServer_SysInfoPage.ino: In function 'void handle_sysinfo_Storage()':
D:/git/HomeAutomation/ESPEasy/src/WebServer_SysInfoPage.ino:669:24: error: 'ESP_PARTITION_TYPE_DATA' was not declared in this scope      
   getPartitionTableSVG(ESP_PARTITION_TYPE_DATA, 0x5856e6);
                        ^
D:/git/HomeAutomation/ESPEasy/src/WebServer_SysInfoPage.ino:675:24: error: 'ESP_PARTITION_TYPE_APP' was not declared in this scope
   getPartitionTableSVG(ESP_PARTITION_TYPE_APP, 0xab56e6);

This was possible for the ESP8266 boards before.

TD-er, 2020-05-21:
OK, this is clearly a missing `#ifdef` somewhere and for ESP32 probably never tried before by anyone.

fmuntean, 2020-05-22:
From my investigation I found that there are missing includes in ESPEasayStorage.ino

#ifdef ESP32 //MFD: These were missing when not using the ARDUINO_OTA lib
  #include <MD5Builder.h>
  #include <ESP_Partition.h>
#endif

I can add a pull request for this if you want.

TD-er, 2020-05-22:
@fmuntean If you could make a PR for it, that would be great.

fmuntean, 2020-07-27:
@TD-er : this issue seems to be still here in 20200721 release:

This is the environment configuration I was able to build before but can't anymore
 [env:MFD_ESP32_4M316k]
extends                   = esp32_common
lib_ignore                = ${esp32_common.lib_ignore}, ArduinoOTA, Adafruit SGP30 Sensor, Adafruit TCS34725, Adafruit TSL2591 Library, ESP8266_SSD1306, Adafruit ILI9341, Adafruit GFX Library, Adafruit Motor Shield V2 Library, Adafruit MPR121, Adafruit NeoPixel, Adafruit Unified Sensor, AM2320, Blynk, CHT16K33 LED Matrix Library, LiquidCrystal_I2C, RN2xx3 Arduino Library, SparkFun APDS9960 RGB and Gesture Sensor, TinyGPSPlus, MechInputs, SDM, SerialSensors, NewPingESP8266, HLW8012
#lib_deps                  = https://github.com/TD-er/ESPEasySerial.git#v2.0.1, Adafruit ILI9341,  Adafruit GFX Library

lib_deps                  = https://github.com/TD-er/ESPEasySerial.git#v2.0.1, https://github.com/lbernstone/Tone, MFRC522
platform                  = ${esp32_common.platform}
build_flags               = ${esp32_common.build_flags}   
                            -D PLUGIN_BUILD_CUSTOM
                            -D MFD_PLUGIN_SET_PROD 
                            -D USE_CUSTOM_H
                            -D DONT_USE_SERVO 
                            -D ESP32
                            -D WEBSERVER_NEW_UI
                            -D USES_P201
                            -D USES_P129
                            -D CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT
board                     = esp32dev
board_build.flash_mode    = qio
extra_scripts             = ${esp32_common.extra_scripts}
                            pre:tools/pio/pre_custom_esp32.py
                            post:post_build_ESP32.py
upload_speed              = 921600

TD-er, 2020-07-27:
It would help if you also gave the build errors.
And your pre_custom_esp32.py

If you don't like to share it online, you can also email me.

TD-er, 2020-07-27:
I noticed MD5Builder is part of the Arduino_OTA library, so if you ignore that lib then MD5Builder will not work.

I will have a look at it.

TD-er, 2020-07-27:
I will make a commit with a few changes I made.
That commit, including this build env should probably work:
```
[env:MFD_ESP32_4M316k]
extends                   = esp32_common
board                     = esp32dev
platform                  = ${esp32_common.platform}
lib_ignore = ${esp32_common.lib_ignore}, ArduinoOTA, Adafruit SGP30 Sensor, Adafruit TCS34725, Adafruit TSL2591 Library, ESP8266_SSD1306, Adafruit ILI9341, Adafruit GFX Library, Adafruit Motor Shield V2 Library, Adafruit MPR121, Adafruit NeoPixel, Adafruit Unified Sensor, AM2320, Blynk, CHT16K33 LED Matrix Library, LiquidCrystal_I2C, RN2xx3 Arduino Library, SparkFun APDS9960 RGB and Gesture Sensor, TinyGPSPlus, MechInputs, SDM, SerialSensors, NewPingESP8266, HLW8012, LOLIN_EPD
lib_deps = ${esp32_common.lib_deps}, https://github.com/lbernstone/Tone, MFRC522
build_flags = ${esp32_common.build_flags} -DPLUGIN_BUILD_CUSTOM -D MFD_PLUGIN_SET_PROD -D DONT_USE_SERVO -D ESP32 -D WEBSERVER_NEW_UI -D USES_P201 -D USES_P129 -D CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT
board_build.flash_mode = qio
extra_scripts = ${esp32_common.extra_scripts} pre:tools/pio/pre_custom_esp32.py
upload_speed = 921600
```

fmuntean, 2020-07-28:
> It would help if you also gave the build errors.
> And your pre_custom_esp32.py
> 
> If you don't like to share it online, you can also email me.
> 
> Is the build error related to P012_LCD ? I tested with your config and I ran into a build error there which I cannot yet explain.

I don't have a custom pre ESP32 script. This is the one that is part of the ESPEasy project.
The error was around the MD5Builder not being found. 
In my local env I have added #include for this and works. But the question is why do we need this for ESP32 ?

fmuntean, 2020-07-28:
You might have run into a different build error than me due to the fact that you might use different plugins.
I use so many lib_ignore due to the fact that I don't need them as I am not using the plugins that are using those libraries. So no need to slow the process down.
my MFD_PLUGIN_SET_PROD selection is:
`// This selects only the plugins I use in my prod boards
#ifdef MFD_PLUGIN_SET_PROD
    # define PLUGIN_SET_NONE

    
    # define USES_P001 // Switch
    # define USES_P002 // ADC
    # define USES_P003 // Pulse
    # define USES_P014 // SI7021

    # define USES_P017 // PN532 RFID reader

    # define USES_P025 // ADS1115
    # define USES_P026 // SysInfo
    # define USES_P028 // BME280
    # define USES_P033 // Dummy
    # define USES_P069 // LM75A Temperature Sensor

    #define USES_P097   // Touch (ESP32)

    # define USES_P105 //RGBW Light
    # define USES_P123 // SI7013 Temperature Humindity and ADC
// # define USES_P129 // RC522 RFID reader
    # define USES_P200 // ADS1115 Thermistor

    # define CONTROLLER_SET_NONE
    # define USES_C005 // OpenHAB MQTT

// # define NOTIFIER_SET_NONE //MFD: setting this removes the notification page
    # ifdef NOTIFIER_SET_STABLE
        #  undef NOTIFIER_SET_STABLE
    # endif // ifdef NOTIFIER_SET_STABLE
    # ifdef NOTIFIER_SET_TESTING
        #  undef NOTIFIER_SET_TESTING
    # endif // ifdef NOTIFIER_SET_TESTING
    # ifdef NOTIFIER_SET_EXPERIMENTAL
        #  undef NOTIFIER_SET_EXPERIMENTAL
    # endif // ifdef NOTIFIER_SET_EXPERIMENTAL
    # ifdef NOTIFIER_SET_NONE
        #  undef NOTIFIER_SET_NONE
    # endif // ifdef NOTIFIER_SET_NONE
    # define USES_N002  // Buzzer

#endif      // ifdef MFD_PLUGIN_SET_PROD`

TD-er, 2020-07-28:
With the pending PR and the esp32 env for in your .ini file I can build it just fine with this plugin set you showed.

fmuntean, 2020-07-31:
> With the pending PR and the esp32 env for in your .ini file I can build it just fine with this plugin set you showed.

Great: let's have this PR moved forward and I can test it locally.

TD-er, 2020-07-31:
It has been merged.

fmuntean, 2020-07-31:
just got the lastest from mega branch and uploaded to my board.
Still getting the same problem.
here are the logs:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
�U36 : Info  :

INIT : Booting version: MFD-mega-2020_07_21 (ESP32 SDK v3.2.3-14-gd3e562907)
37 : Info  : INIT : Free RAM:294924
37 : Info  : INIT : Cold Boot - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
38 : Info  : FS   : Mounting...
76 : Info  : CRC  : SecuritySettings CRC   ...OK
142 : Info  : INIT : Free RAM:289412
144 : Info  : INIT : I2C
145 : Info  : INIT : SPI Init (without CS)
159 : Info  : MFRC522: Init
160 : Info  : MFRC522: Reset on pin: 25
373 : Info  : MFRC522: Software Version:  = v2.0
430 : Info  : INFO : Plugins: 16 [Normal] (ESP32 SDK v3.2.3-14-gd3e562907)
432 : Info  : EVENT: System#Wake
560 : Info  : WIFI : Set WiFi to STA
748 : Info  : WIFI  : Start network scan
3058 : Info  : WIFI : Connecting Howard attempt #1
3061 : Info  : WIFI : WiFi.status() = WL_DISCONNECTED  SSID: Howard
3065 : Info  : Webserver: start
3079 : Info  : EVENT: System#Boot
3087 : Info  : ACT  : NOTIFY,1
3090 : Info  : Command: NOTIFY
3130 : Info  : WIFI  : Scan finished, found: 60
3132 : Info  : WIFI  : Selected: Linksys 00:1E:E5:2D:4C:87 Ch:1 (-61dBm) WPA2/PSK
3680 : Info  : SYS  : 0.00,0.00,100.00,-1.00
4523 : Info  : EVENT: SYSTEM#Uptime=0.00
4675 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 221096 WiFiStatus WL_DISCONNECTED
4703 : Info  : EVENT: SYSTEM#WiFi=0.00
4775 : Info  : EVENT: SYSTEM#Load=100.00
4847 : Info  : EVENT: SYSTEM#=-1.00
12606 : Info  : WIFI : Disconnected! Reason: '(15) 4way handshake timeout' Connected for 9441 ms
12723 : Info  : WIFI : Set WiFi to STA
12836 : Info  : WIFI  : Start network scan
15546 : Info  : WIFI : Connecting Linksys attempt #2
15548 : Info  : WIFI : WiFi.status() = WL_DISCONNECTED  SSID: Linksys
15560 : Info  : WIFI  : Scan finished, found: 54
15562 : Info  : WIFI  : Selected: Linksys 00:1E:E5:2D:4C:87 Ch:1 (-59dBm) WPA2/PSK
15591 : Info  : EVENT: WiFi#Disconnected
20188 : Info  : WIFI : Set WiFi to AP+STA
20288 : Info  : WIFI : Set WiFi to AP+STA
20388 : Info  : WIFI : Set WiFi to AP+STA
20488 : Info  : WIFI : Set WiFi to AP+STA
20588 : Info  : WIFI : Set WiFi to AP+STA
20688 : Info  : WIFI : Set WiFi to AP+STA
20788 : Info  : WIFI : Set WiFi to AP+STA
20888 : Info  : WIFI : Set WiFi to AP+STA
20988 : Info  : WIFI : Set WiFi to AP+STA
21088 : Info  : WIFI : Set WiFi to AP+STA
21188 : Info  : WIFI : Set WiFi to AP+STA
21288 : Info  : WIFI : Set WiFi to AP+STA
21388 : Info  : WIFI : Set WiFi to AP+STA
21488 : Info  : WIFI : Set WiFi to AP+STA
[... 58 lines cut ...]
  #42 0x4011e587:0x3ffb0920 in setAP(bool) at src/ESPEasyWifi.cpp:676
  #43 0x4011f5eb:0x3ffb0940 in WiFiConnected() at src/ESPEasyWifi.cpp:141
  #44 0x4011daa7:0x3ffb0970 in NetworkConnected() at src/ESPEasyNetwork.cpp:38
  #45 0x400f63f1:0x3ffb0990 in syslog(unsigned char, char const*) at D:/git/HomeAutomation/ESPEasy/src/Misc.ino:2584
  #46 0x400f6601:0x3ffb09f0 in addToLog(unsigned char, char const*) at D:/git/HomeAutomation/ESPEasy/src/Misc.ino:2584
  #47 0x400f665a:0x3ffb0a20 in addToLog(unsigned char, String const&) at D:/git/HomeAutomation/ESPEasy/src/Misc.ino:2584
  #48 0x4011e3e1:0x3ffb0a40 in setWifiMode(wifi_mode_t) at src/ESPEasyWifi.cpp:676
  #49 0x4011e587:0x3ffb0a90 in setAP(bool) at src/ESPEasyWifi.cpp:676
  #50 0x4011f5eb:0x3ffb0ab0 in WiFiConnected() at src/ESPEasyWifi.cpp:141

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
�U36 : Info  :

fmuntean, 2020-08-02:
tested with latest and now builds succesful.

PR #3052 (2020-06-08): [LittleFS] Make switching between SPIFFS and LittleFS easy to do
Just add "_LittleFS" to the PIO env label and it will compile with LittleFS instead of SPIFFS.

First impression:
- Load speed of web page is just WOW! (average 170 msec)
- Saving settings is terribly slow (1...2 seconds)

Fixes: #3083
commit 25459bb4 (2020-06-08): [ESP32] Fix Disabling ARDUINO_OTA fails the build (#3083)

Fixes: #3083
closed 2020-08-02 by fmuntean