letscontrolit/ESPEasy #3092: mqtt 1.4.14  on Pi3

CAHbKA-A, 2020-05-25:
Good day!
Tell me, please, with which version of mqtt broker mosquitto does easyEsp work?

I tried using
EasyESP: Release mega-20200515
protocoll HomeAssistent (openHAB) MQTT

RaspberryPI3.
(mosquitto version 1.4.14 -mosquitto is already the newest version.)

with this configuration, esp does not publish the value in Mqtt.

When using the release of ESPEasy_R120 (2016). MQTT is working.

CAHbKA-A, 2020-05-25:
ESP_Easy_mega-20181129_dev_ESP8266_4096.bin working

CAHbKA-A, 2020-05-25:
ESP_Easy_mega-20190202_dev_ESP8266_4096.bin working

TD-er, 2020-05-25:
OK, I think I need a bit more information to understand what you're asking here.

It looks like you're quoting some text of somewhere (no idea where it comes from) and I also really don't know what your question is.

ESPEasy does not differentiate between ESP-12F or NodeMCU boards.
So I have no clue what the problem is here.

Also what entry do you want to delete? 
What question is closed?

CAHbKA-A, 2020-05-28:
I have no question.  Close this issue,  plz

closed 2020-05-28 by CAHbKA-A