letscontrolit/ESPEasy #3101: Feature request : Controller plugin for probe request protocol.

happytm, 2020-06-05:
I have been looking for cheap and simple ways to reduce power consumption for my battery powered ESP8266  & ESP32 devices for quite long time.I tried lot of different hardware modification and software protocols like IR, RF, LORA & ESPNow. There are always complexities in hardware implementation
due to lack of GPIO pins on ESP8266  or simply hardware is too expensive. So I decided to leverage built in feature called probe request in ESP software to create a simple standalone IOT sensor network in star topology.  

I have been testing this code for at least last 2 month with one ESP8266 Gateway and 8 ESP8266 remote battery powered devices and they are working fine without any issues.The remote devices take only 50 to 75 milliseconds to send sensor data to gateway and go back to sleep. which is way faster than any other protocol I tried. If for some reason if you want to control the remote devices via MQTT there is duplex mode option selection in remote device code. Duplex mode require around 140 milliseconds to communicate between Gateway and remote device in two way fashion. The gateway device have MQTT broker built in so there is no need to connect to any MQTT broker.

This is the simplest and least expensive way I know to implement low power standalone IOT sensor network.

I would like to request one of the developer who understand ESPEasy plugins to take a look at the code linked below and try to implement as a controller plugin if possible.:

ESP8266 Gateway code ( With built in MQTT broker - no need to connect to any MQTT broker):

https://github.com/happytm/BatteryNode/blob/master/examples/ProbeRequestGatewayESP8266.ino

ESP32 Gateway code ( With built in MQTT client - connection may be needed to MQTT broker):

https://github.com/happytm/BatteryNode/blob/master/examples/ProbeRequestGatewayESP32.ino

ESP8266 Remote code (this is for battery powered devices.These devices are not connected directly to home router or internet. They are only communicating with the gateway without connecting to gateway. For standalone local network connection to internet is not required):

https://github.com/happytm/BatteryNode/blob/master/examples/ProbeRequestRemoteEsp8266.ino

Thanks.

uzi18, 2020-06-05:
Brilliant idea to use mac address as a communication medium, but it is also limited for data length.

TD-er, 2020-06-05:
Without having looked at your code (yet), I assume you are using ESP-now for sending data.

I'm also working on a (rather huge) addition to ESPEasy using ESP-now, as you can see here: https://github.com/TD-er/ESPEasy/tree/feature/esp-now-controller

I've also overcome the limited data length as mentioned by @uzi18 by implementing a data splitter and data merger.

That part is working fine and I have a setup here routing roughly 3 (large) MQTT messages a second from several ESP nodes via ESP-now to a MQTT broker.
These messages are 700 bytes in size (topic + message) and thus split in 3 packets and merged on the receiving end.

It is also possible to make hops, so you can forward these messages to a connected gateway node.
Right now, I've run tests over 5 hops covering 200+ meter, but the success rate then reduces.

It can use ESP32 and ESP8266 mixed.

So I guess it would be a very good idea to have some very small builds that just send the data and then use these gateway nodes (always connected, always on) to forward the data to other nodes or to the MQTT broker or wherever you need them to be.

I will now have a look at your code, to see what would be the easiest way to integrate these.

TD-er, 2020-06-05:
Ah, now I get it, you just encode your sensor type in the Mac address of the node.
And by using the `onSoftAPModeProbeRequestReceived` callback you simply get your sensor announcement.

That's rather brilliant :)

It is at least something I would like to test with my implementation of the ESP-now discovery so a sensor can set its MAC to some discovery specific MAC and if a node gets scanned by it, it can help the other node to map the network.

happytm, 2020-06-05:
>Brilliant idea to use mac address as a communication medium, but it is also limited for data length.
 

@uzi18  Yes it is limited data of six bytes per probe request. I am transferring 24 bytes of data using 4 probe requests within 140 milliseconds (3 probe request from remote and 1 probe request from gateway). For most use cases it could be enough information for  gateway to collect to effectively monitor and control the remote device.But you can use more probe requests as required to exchange more data.

> I'm also working on a (rather huge) addition to ESPEasy using ESP-now, as you can see here: https://github.com/TD-er/ESPEasy/tree/feature/esp-now-controller

> So I guess it would be a very good idea to have some very small builds that just send the data and then use these gateway nodes (always connected, always on) to forward the data to other nodes or to the MQTT broker or wherever you need them to be.

@TD-er  Thank you for testing the code. I will test your code. I strongly believe ESPEasy need some plugin for battery powered wireless remote sensor devices which is simple and inexpensive. There are lot of use cases around the house like door or window sensors , outdoor weather station sensor, water or oil tank level sensor, leak sensor & even temperature and humidity sensors inside the house where mains power is not available easily.

The beauty of this code is it has MQTT broker built in for ESP8266. The uMQTT broker library is not ported to ESP32 devices yet so I had to use MQTT client for ESP32 gateway. So in theory you can create a network of remote devices for unlimited range as long as there is one gateway device (always on) every few meters. They will work as MQTT repeaters in the area they are present and will connect to nearby gateway to exchange data. This is kind of simple mesh network.

Thanks.

bccrew, 2020-06-06:
Is it possible to use secure MQTT with this code?

happytm, 2020-06-06:
@bccrew  Yes it is possible. Please look at following code in the ESP8266 gateway file.
  
      virtual bool onAuth(String username, String password) {
      Serial.println("Username/Password: " + username + "/" + password);
      return true;`

bccrew, 2020-06-07:
Actually I meant using a (public) certificate for MQTTS.

happytm, 2020-06-07:
No. (public) certificate is not implemented in uMQTT library as far as I know.

The library is here:  https://github.com/martin-ger/uMQTTBroker if you need more information.

Thanks.