letscontrolit/ESPEasy #3106: MQTT import issues? (please test)

TD-er, 2020-06-08:
As stated here: https://github.com/letscontrolit/ESPEasy/commit/15c34eea022efaeede7151c7ccc9d8fdb8dff738
Apparently the MQTT import plugin is causing issues with the current `mega` branch.

> CPU load rising over 40%(!) with MQTTimport (P037) plug-in with this changes. :-(
```
P037 : MQTT Intentional reconnect
P037 : MQTT connection lost
P037 : MQTT Intentional reconnect
P037 : MQTT connection lost
P037 : MQTT Intentional reconnect
P037 : MQTT connection lost
```

Test build with current mega branch: 
[ESPEasy_ESP82xx_mega-20200515-87-g7a353b12.zip](https://www.dropbox.com/s/5xhasjgpoh2yp8f/ESPEasy_ESP82xx_mega-20200515-87-g7a353b12.zip?dl=0)
[ESPEasy_ESP32_mega-20200515-87-g7a353b12.zip](https://www.dropbox.com/s/2cgn5yhlv4mau5n/ESPEasy_ESP32_mega-20200515-87-g7a353b12.zip?dl=0)

Would be nice if someone else could also test it.

Wiki591, 2020-06-08:
At a first glance:

[image]

[image]

Status/behaviour before update:

```

CPU Load avg 17.5%

Copied: "Build:?
20104  - Mega
System Libraries:?
ESP82xx Core bc204a9b, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:?
mega-20191208
Plugins:?
46 [Normal]
Build Time:?
Dec  8 2019 18:04:38
Binary Filename:?
ESP_Easy_mega-20191208_normal_ESP8266_4M1M.bin
" to clipboard!

Copied: "1642559268: IMPT : [Batterie#AZ] : 6712.00
1642559352: EVENT: Batterie#AZ=6712.00
1642559441: IMPT : [BS#LZ] : 92.00
1642559456: EVENT: BS#LZ=92.00
1642559586: IMPT : [Batterie#EC] : 7603200.00
1642559606: IMPT : [Batterie#FC] : 1036800.00
1642559615: IMPT : [BS#BS] : 56.40
1642559636: IMPT : [BS#PP] : 1767.00
1642559641: EVENT: Batterie#EC=7603200.00
1642559754: EVENT: Batterie#FC=1036800.00
1642559836: EVENT: BS#BS=56.40
1642559907: IMPT : [BS#LP] : 250.00
1642559936: EVENT: BS#PP=1767.00
1642560035: EVENT: BS#LP=250.00
1642560061: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t7.txt='92%''     
1642560064: Command: SendToHTTP
1642560124: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t5.txt='56.40V''  
1642560126: Command: SendToHTTP
1642560246: ACT  : publish domoticz/in,'{'idx':138,'nvalue':0,'svalue':'92'}'     
1642560249: Command: publish
1642560254: ACT  : TaskValueset,4,2,6712/60         
1642560255: Command: TaskValueset
1642560259: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t33.txt='Ausgleichszeit:112min.''           
1642560261: Command: SendToHTTP
1642560287: ACT  : TaskValueSet,3,3,1036800/60      
1642560289: Command: TaskValueSet
1642560293: ACT  : TaskValueSet,3,4,7603200/60      
1642560295: Command: TaskValueSet
1642560308: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t34.txt='Volladung:12.00Tage''                    
1642560311: Command: SendToHTTP
1642560355: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t35.txt='Ausgleichsldg.:88.00Tage''                    
1642560357: Command: SendToHTTP
1642575546: Command: publish
1642575551: ACT  : TaskValueset,4,2,6701/60         
1642575553: Command: TaskValueset
1642575557: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t33.txt='Ausgleichszeit:112min.''           
1642575559: Command: SendToHTTP
1642575590: ACT  : TaskValueSet,3,3,1036800/60      
1642575592: Command: TaskValueSet
1642575595: ACT  : TaskValueSet,3,4,7603200/60      
1642575596: Command: TaskValueSet
1642575609: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t34.txt='Volladung:12.00Tage''                    
1642575612: Command: SendToHTTP
1642575654: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t35.txt='Ausgleichsldg.:88.00Tage''                    
[... 137 lines cut ...]
344021: EVENT: Batterie#FC=1036800.00
344113: IMPT : [BS#BS] : 56.40
344685: ACT  : publish domoticz/in,'{'idx':138,'nvalue':0,'svalue':'92'}'
344688: Command: publish
344715: ACT  : TaskValueset,4,2,6238/60
344716: Command: TaskValueset
344720: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t33.txt='Ausgleichszeit:104min.''
344722: Command: SendToHTTP
344759: ACT  : TaskValueSet,3,3,1036800/60
344760: Command: TaskValueSet
344763: ACT  : TaskValueSet,3,4,7603200/60
344765: Command: TaskValueSet
344779: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t34.txt='Volladung:12.00Tage''
344781: Command: SendToHTTP
344817: ACT  : SendToHTTP 192.168.xxx.yyy,80,'/control?cmd=Di,page1.t35.txt='Ausgleichsldg.:88.00Tage''
344819: Command: SendToHTTP
" to clipboard!

```

For me works perfectly, MQTT broker on Raspberry Stretch:

mosquitto version 1.3.4 (build date 2017-05-29 22:25:09+0000)

Wired connection of the broker.

giig1967g, 2020-06-08:
for me is working fine too:
```
152303831: IMPT : Connected to MQTT broker with Client ID=ESPT10_10-Import
152303859: IMPT : [test#Value1] subscribed to OH2Varazze/status/zr5
152303860: IMPT : [test#Value2] subscribed to OH2Varazze/status/zr4
152303861: IMPT : [test#Value3] subscribed to OH2Varazze/status/zn10a
152304044: IMPT : [test#Value1] : 3.00
152304137: EVENT: test#Value1=3.00
152304257: IMPT : [test#Value2] : 2.00
152304337: EVENT: test#Value2=2.00
152304445: IMPT : [test#Value3] : 0.00
152304537: EVENT: test#Value3=0.00
152324041: WD : Uptime 2539 ConnectFailures 0 FreeMem 13680 WiFiStatus 3
152328486: EVENT: Clock#Time=Mon,15:25
152335150: IMPT : [test#Value3] : 1.00
152335237: EVENT: test#Value3=1.00
152343946: IMPT : [test#Value3] : 0.00
152344037: EVENT: test#Value3=0.00
```

v-a-d-e-r, 2020-06-09:
Ok, now it works here, too. I have no clue what was happening here.... :-/ I think the open issue can be closed now.

TD-er, 2020-06-09:
OK, then we can close it.
Thanks all for testing.

closed 2020-06-09 by TD-er