letscontrolit/ESPEasy #3113: Cannot access Rules pages 2,3,4 if unit is password protected

giig1967g, 2020-06-12:
I found a bug in the latest releases:
if you have set an "Admin Password" for accessing the node, you are not allowed to access rules pages 2, 3 and 4. You can only access page rule page 1.

ghtester, 2020-06-15:
Well, just for a reference:
#2980
#2987 
In my environment access to rule pages 2-4 partially works - it looks it depends on web browser as I can't log on to my recently updated ESP node with MS Edge browser at all (/timingstats page which is not protected by password opens normally).

giig1967g, 2020-06-15:
I use Firefox. And when I try to access rules pages 2-4 I get a looping request for password, that does not give access even if using the right credentials

ghtester, 2020-06-15:
At my primary site Firefox  works including all 4 rule pages. Remote access (IP filtering set to Allow All) from another Firefox (from another PC, another OS and Firefox releases) does not work for rule pages 2-4 and MS Edge can't access any password protected page (also repeating logon window).

TD-er, 2020-06-15:
Which Edge version?
The "old" one, or the "new" one based on Chromium?

ghtester, 2020-06-15:
The "Standard" one I suppose so this should be "old" but who knows... :-) 
Microsoft Edge 44.17763.831.0
Microsoft EdgeHTML 18.17763

TD-er, 2020-06-28:
Did some tests.
Chrome works as expected, but these all don't work:

- Edge (Chromium version)
- Firefox
- Opera

So it sure is a problem, but so far I don't have a clue yet where it could be.
Maybe in the JavaScript used?

TD-er, 2020-06-28:
Maybe someone with some JavaScript experience can see what may be wrong here?
I guess the `saveRulesFile()` function is also called when the page is loaded?

```JavaScript
<button id='save_button' class='button' onClick='saveRulesFile()'>Save</button>
<div id='toastmessage'>Saved!</div>
<script defer type='text/JavaScript'>
    function saveRulesFile() {
        document.getElementById("save_button");
        let e = document.getElementById("size")
          , t = document.getElementById("rules").value;
        t = t.replace(/\r?\n/g, "\r\n");
        let n = document.getElementById("set").value
          , l = new File([t],"rules" + n + ".txt",{
            type: "text/plain"
        })
          , a = new FormData;
        a.append("file", l),
        a.append("enctype", "multipart/form-data");
        let o = "/rules" + n + ".txt?callback=" + Date.now();
        fetch(o).then(e=>e.text()).then(l=>{
            if (t === l)
                console.log("nothing to save...");
            else {
                fetch("/upload", {
                    method: "POST",
                    body: a
                }).then(e=>e.text()).then(l=>{
                    let a = "/rules" + n + ".txt?callback=" + Date.now();
                    fetch(a).then(e=>e.text()).then(n=>{
                        t === n ? (toasting(),
                        e.innerHTML = t.length) : console.log("error when saving...")
                    }
                    )
                }
                )
            }
        }
        )
    }
</script>
```

TD-er, 2020-06-28:
I've tried to use an online JavaScript validator, but I keep running into "Unexpected 'let'" on the very first line where it is used.
The most obvious issues have been fixed here:
```JavaScript
/*jslint browser:true */
function saveRulesFile() {
    "use strict";
    let e = document.getElementById("size")
      , t = document.getElementById("rules").value;
    t = t.replace(/\r?\n/g, "\r\n");
    let n = document.getElementById("set").value
      , l = new File([t],"rules" + n + ".txt",{
        type: "text/plain"
    });
    let a = new FormData();
    a.append("file", l),
    a.append("enctype", "multipart/form-data");
    let o = "/rules" + n + ".txt?callback=" + Date.now();
    fetch(o).then(e=>e.text()).then(l=>{
        if (t === l)
            console.log("nothing to save...");
        else {
            fetch("/upload", {
                method: "POST",
                body: a
            }).then(e=>e.text()).then(l=>{
                let a = "/rules" + n + ".txt?callback=" + Date.now();
                fetch(a).then(e=>e.text()).then(n=>{
                    t === n ? (toasting(),
                    e.innerHTML = t.length) : console.log("error when saving...");
                }
                );
            }
            );
        }
    }
    );
}
```

But I really don't know anything about JavaScript.
So if someone else can have a look, that would be great.

TD-er, 2020-06-28:
On another validator site I could get a bit more info and came to this:
```JavaScript
/*jslint browser:true */
/*jslint esversion: 6 */
function saveRulesFile() {
    "use strict";
    let e = document.getElementById("size");
    let t = document.getElementById("rules").value;
    t = t.replace(/\r?\n/g, "\r\n");
    let n = document.getElementById("set").value;
    let l = new File([t],"rules" + n + ".txt",{
        type: "text/plain"
    });
    let a = new FormData();
    a.append("file", l);
    a.append("enctype", "multipart/form-data");
    let o = "/rules" + n + ".txt?callback=" + Date.now();
    fetch(o).then(e=>e.text()).then(l=>{
        if (t === l)
            console.log("nothing to save...");
        else {
            fetch("/upload", {
                method: "POST",
                body: a
            }).then(e=>e.text()).then(l=>{
                let a = "/rules" + n + ".txt?callback=" + Date.now();
                fetch(a).then(e=>e.text()).then(n=>{
                    if (t === n) {
                        toasting();
                        e.innerHTML = t.length;
                    } else {
                        console.log("error when saving...");
                    }
                }
                );
            }
            );
        }
    }
    );
}
```

TD-er, 2020-06-28:
I moved the JavaScript code to the `<head>` part of the page, to make sure it is just declared as a function and not called.
But it still seems to be called when loading the page after a login.

I guess we should check to see if there is an element with the ID's we're looking for and if they do not exist then we should not call the save function.

TD-er, 2020-06-28:
Ah, I may have found the issue here.
With loading the page directly from the URL pointing to the rules set, including login, you have a special use case.

- You have web arguments (the rules set)
- You reload the page with the same rules set number as the last time you tried to load it.

This combined with the bug that we still has code present to support rules upload via POST would delete the rules.

ghtester, 2020-06-29:
Unfortunately I can't still logon using Edge to ESP node just updated with a latest custom build:

Build:⋄ | 20108  - Mega
-- | --
System Libraries:⋄ | ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Jun 28 2020 22:18:18
Plugin Count:⋄ | 32 [Normal]
Build Origin: | Vagrant
Build Time:⋄ | Jun 28 2020 22:20:46
Binary Filename:⋄ | ESP_Easy_mega_20200628_custom_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega

Also when I open (at least) the Tools page from Firefox, the blue box 'Submitted' appears for a while (without clicking Submit button of course).
Maybe it's due to custom build...

TD-er, 2020-06-30:
The "Submitted" box is indeed an issue introduced with the changes I made for this.
But I don't understand why you're seeing these issues with Edge, as I tried it myself with Edge. (and Firefox & Chrome)

ghtester, 2020-06-30:
Well, regarding to Edge - I don't know... FYI I try to access the ESP node from Edge which is remote and there's port forwarding at firewall (on ESP node the Client IP filtering is set to Allow all). But I don't know how this could make the trouble with logon when Firefox works even remotely (and after your last patch Firefox works also on rule pages 2-4).

ghtester, 2020-07-15:
IMHO this issue should be reopened, also #3138

TD-er, 2020-07-15:
Well I cannot re-open a pull request, but I will re-open this issue.

ghtester, 2020-07-27:
It looks that in latest sources the issue with rules pages 2-4 is fixed but the access to ESP node protected by password from Edge is not still possible.
Build:⋄	20109 - Mega
System Libraries:⋄	ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄	My Build: Jul 24 2020 17:33:39
Plugin Count:⋄	33 [Normal]
Build Origin:	Vagrant
Build Time:⋄	Jul 24 2020 17:36:09
Binary Filename:⋄	ESP_Easy_mega_20200724_custom_ESP8266_4M1M
Build Platform:⋄	Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄	mega

ghtester, 2021-01-19:
Unfortunately I am still experiencing this issue even with current custom (Vagrant) build.

Build:⋄ | 20111 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Jan 19 2021 20:59:35
Plugin Count:⋄ | 34
Build Origin: | Vagrant
Build Time:⋄ | Jan 19 2021 20:58:39
Binary Filename:⋄ | ESP_Easy_mega_20210119_custom_IR_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega_4747dd8

TD-er, 2021-01-19:
Just as a work-around... there currently is not a limit on the max. rules size
So you can save 6k or 8k rules if you like (just make sure to keep a copy in some editor when saving, just to be sure)

TD-er, 2021-01-19:
@ghtester "testing" builds have some parts taken out of the code.
Search for `LIMIT_BUILD_SIZE` in the code

ghtester, 2021-01-19:
Thanks a lot for hints, I know the increased rule size limit was mentioned earlier somewhere but I still see the warning in GUI:

Current size: 2158 characters (Max 2048)Filesize exceeds web editor limit!

Should I always import longer rules as a file or is it safe to edit rules > 2048 characters in web editor?

I am not sure what exactly that means> Search for LIMIT_BUILD_SIZE in the code

This was the build result:
    default: Retrieving maximum program size .pio/build/custom_IR_ESP8266_4M1M/ESP_Easy_mega_20210119_custom_IR_ESP8266_4M1M.elf
    default: Checking size .pio/build/custom_IR_ESP8266_4M1M/ESP_Easy_mega_20210119_custom_IR_ESP8266_4M1M.elf
    default: Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
    default: RAM:   [=====     ]  54.1% (used 44292 bytes from 81920 bytes)
    default: Flash: [========  ]  82.4% (used 860576 bytes from 1044464 bytes)
    default: ======================== [SUCCESS] Took 612.81 seconds ========================
    default: Environment             Status    Duration
    default: ----------------------  --------  ------------
    default: custom_IR_ESP8266_4M1M  SUCCESS   00:10:12.807
    default: ========================= 1 succeeded in 00:10:12.807 =========================

TD-er, 2021-01-19:
The rules editing page is now using JavaScript to actually call the file upload feature.
But as it is not tested very well with larger rules files, I kept the warning.
Not sure what the drawbacks can be, but I can imagine on 1M units with very small file systems you may run into file storing issues due to fragmentation.
NO idea whether parsing multiple smaller files or one larger file is taking more resources to process.

What I meant with searching for `LIMIT_BUILD_SIZE` in the code was an answer to your remark:

> It's strange that this (official) build does NOT have the issue... :-/ I'll need to investigate...

The "testing" builds have some parts left out of the build which are wrapped in checks for `#ifndef LIMIT_BUILD_SIZE`

ghtester, 2021-01-19:
OK, thanks again, I am not able to investigate the source codes at this level, I meant to compare the behaviour / configuration differences of 2 ESP nodes with these different builds (and they have also different plugins in use and a different Free RAM so it could be perhaps an issue root... )

Git Build: | My Build: Jan 19 2021 20:59:35
-- | --
Local Time: | 2021-01-19 23:45:49
Uptime: | 0 days 1 hours 26 minutes
Load: | 6.05% (LC=2378)
Free RAM: | 7440
Free Stack: | 3648

-----------------------------------------

Git Build: |  
-- | --
Local Time: | 2021-01-19 23:48:44
Uptime: | 20 days 9 hours 53 minutes
Load: | 5.08% (LC=3762)
Free RAM: | 16032
Free Stack: | 3712

ghtester, 2021-01-20:
Another official build which has NO issue with rules:

Build:⋄ | 20111 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 88 [Normal] [Testing] [Development]
Build Origin: | Travis
Build Time:⋄ | Jan 14 2021 23:21:32
Binary Filename:⋄ | ESP_Easy_mega_20210114_dev_ESP8266_4M1M
Build Platform:⋄ | Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄ | mega-20210114_cdc8a1a

Git Build: |  
-- | --
Local Time: | 2021-01-20 01:34:54
Uptime: | 1 days 1 hours 48 minutes
Load: | 4.29% (LC=4025)
Free RAM: | 14464
Free Stack: | 3728

TD-er, 2021-01-20:
And what if you test with "normal" builds?
"testing" and "development" both include the "testing" part, which does leave out some parts.
So if the "normal" build has issues, then we may have a pointer where to look for the issue at hand.

ghtester, 2021-01-20:
OK, I'll try on another ESP node, which currently has an older Vagrant build and NO rules issue.

Build:⋄ | 20108 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Jun 28 2020 22:18:18
Plugin Count:⋄ | 32 [Normal]
Build Origin: | Vagrant
Build Time:⋄ | Jun 28 2020 22:20:46
Binary Filename:⋄ | ESP_Easy_mega_20200628_custom_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega

Git Build: | My Build: Jun 28 2020 22:18:18
-- | --
Local Time: | 2021-01-20 08:52:57
Uptime: | 0 days 0 hours 5 minutes
Load: | 9.73% (LC=325)
Free RAM: | 17320 (33208 - )
Free Stack: | 3712 (3824 - )

I am going to replace the firmware with latest official Normal build (20210114) and post the result here  a bit later.

TD-er, 2021-01-20:
Older builds had issues with saving the file as they needed to handle it via a POST and thus have it all allocated in memory, causing all kinds of issues.
So that's why we added JavaScript to perform an upload of the file (upload is written directly to the storage, not kept in memory as a whole)

So if you try to compare older builds, please do not use large rules files.

That being said, I don't think this issue is related to the rules size, it probably has something to do with the URL being created and how it is handled.

The only reason I mentioned the option to store larger rules files (in newer builds, not older builds!!!!) is to overcome the need to use multiple rules files and thus act as a work-around for the problem you're now facing.

So please try to keep those aspects separated.
- Bug has probably something to do with URL and how it is being handled.
- Rules size is a work-around to not having to use several files.

ghtester, 2021-01-20:
Thanks for a reply / info. 
I don't plan to use older builds anymore, it was the node already flashed a couple months ago and pulled out from a drawer now.
I also plan to split a single large rule between Rule Sets as the size really makes the trouble described earlier. (try the same with rule size about 1800). I don't know how the URL is being handled so I believe it's related with this somehow... When the rule length is about 1717 characters, switching between Rule Sets usually works but sometimes the issue happen. When the rule length is 1800 or more, the issue happen always (at least in Chromium currently tested but I believe it will be similar in other browsers as well).
In general, I think it's not so serious bug if we know what to do... ;-)

TD-er, 2021-01-20:
Hmm that size of 1700'ish bytes intrigues me.
Is that when loading the page, or when saving it?

ghtester, 2021-01-20:
After opening the (any) Rule Set containing the rule longer than about 1800 characters and then open another Rule Set.
Just opening, without any save attempt.
Rules even longer than 2048 can be saved properly. But as soon as I save rule longer than 1700-1800, starting to have the issue.

uzi18, 2021-01-20:
@ghtester no hint in console log (Developer tools)?

TD-er, 2021-01-20:
I discussed with @Grovkillen the possibility to also load the rules using JavaScript.
I think that will solve this issue once and for all, but I also would like to know WHY this happens.

ghtester, 2021-01-20:
@uzi18 I don't think so... Tried to disable all devices. Rules Set 1 - empty, Rules Set 2 - 231 characters, Rules Set 3 - 1918 characters,  Rules Set 4 empty (0 characters). All rules are commented out (// at start of every line).

Serial Log Level: set to Debug dev

Rebooted ESP, logged on, then cleared the scrollback in Putty serial, opened Rules menu (so empty Rules Set 1 opened), opened Rules Set 2, opened Rules Set 3, opened Rules Set 4 and logon issue appears, saved log...

pm open,type:2 0
                21072 : Debug  : HTML : Request file /index.htm.gz
21103 : Debug  : HTML : Request file /index.htm
21192 : Debug dev : sendcontent free: 18792 chunk size:400
21224 : Debug dev : sendcontent free: 17376 chunk size:400
21227 : Debug dev : sendcontent free: 17376 chunk size:400
21230 : Debug dev : sendcontent free: 17376 chunk size:400
21234 : Debug dev : sendcontent free: 15976 chunk size:400
21238 : Debug dev : sendcontent free: 15976 chunk size:400
21240 : Debug dev : sendcontent free: 15976 chunk size:400
21246 : Debug dev : sendcontent free: 15976 chunk size:400
21249 : Debug dev : sendcontent free: 15976 chunk size:400
21251 : Debug dev : sendcontent free: 15976 chunk size:400
21255 : Debug dev : sendcontent free: 14576 chunk size:400
21259 : Debug dev : sendcontent free: 15976 chunk size:400
21262 : Debug dev : sendcontent free: 15976 chunk size:400
21265 : Debug dev : sendcontent free: 15976 chunk size:400
21300 : Debug dev : sendcontent free: 16776 chunk size:400
21307 : Debug dev : sendcontent free: 16776 chunk size:400
21314 : Debug dev : sendcontent free: 17840 chunk size:400
21323 : Debug dev : sendcontent free: 16520 chunk size:400
21348 : Debug dev : sendcontent free: 17920 chunk size:400
21349 : Debug dev : sendcontent free: 17920 chunk size:0
31917 : Debug dev : Memtrace
                            0: lowest: 14536  loadFromFS-> 21048 loadFromFS-> 21008 sendHeaderBlocking-> 20928 sendWebPage-> 18856 sendContentBlocking-> 18752 sendContentBlocking-> 17336 sendContentBlocking-> 17336 sendContentBlocking-> 17336 sendContentBlocking-> 15936 sendContentBlocking-> 15936 sendContentBlocking-> 15936 sendContentBlocking-> 15936 sendContentBlocking-> 15936 sendContentBlocking-> 15936 sendContentBlocking-> 14536
31918 : Debug dev : 1: lowest: 15936  PluginCall (22)-> 21104 PluginCall (22)-> 21048 PluginCall (5)-> 21048 PluginCall (30)-> 21048 handle_root-> 21048 loadFromFS-> 21048 loadFromFS-> 21008 sendHeaderBlocking-> 20928 sendWebPage-> 18856 sendContentBlocking-> 18752 sendContentBlocking-> 17336 sendContentBlocking-> 17336 sendContentBlocking-> 17336 sendContentBlocking-> 15936 sendContentBlocking-> 15936
31919 : Debug dev : 2: lowest: 15936  PluginCall (22)-> 21048 PluginCall (5)-> 21048 PluginCall (30)-> 21048 handle_root-> 21048 loadFromFS-> 21048 loadFromFS-> 21008 sendHeaderBlocking-> 20928 sendWebPage-> 18856 sendContentBlocking-> 18752 sendContentBlocking-> 17336 sendContentBlocking-> 17336 sendContentBlocking-> 17336 sendContentBlocking-> 15936 sendContentBlocking-> 15936 sendContentBlocking-> 15936
31920 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 19800 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
32807 : Debug  : LoopStats: shortestLoop: 234 longestLoop: 1076756 avgLoopDuration: 3394.45 loopCounterMax: 128205 loopCounterLast: 8811
32807 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 8808/1919/7/79.14
34594 : Debug dev : sendcontent free: 18648 chunk size:400
34599 : Debug dev : sendcontent free: 17272 chunk size:400
34602 : Debug dev : sendcontent free: 17272 chunk size:400
34604 : Debug dev : sendcontent free: 17272 chunk size:400
34608 : Debug dev : sendcontent free: 15872 chunk size:400
34611 : Debug dev : sendcontent free: 15872 chunk size:400
34614 : Debug dev : sendcontent free: 15872 chunk size:400
34621 : Debug dev : sendcontent free: 15872 chunk size:400
34624 : Debug dev : sendcontent free: 15872 chunk size:400
34627 : Debug dev : sendcontent free: 15872 chunk size:400
34631 : Debug dev : sendcontent free: 16752 chunk size:400
34634 : Debug dev : sendcontent free: 16752 chunk size:400
34637 : Debug dev : sendcontent free: 17272 chunk size:400
34639 : Debug dev : sendcontent free: 17272 chunk size:400
34645 : Debug dev : sendcontent free: 15816 chunk size:400
34659 : Debug dev : sendcontent free: 17256 chunk size:400
34666 : Debug dev : sendcontent free: 17840 chunk size:400
34674 : Debug dev : sendcontent free: 17560 chunk size:400
34682 : Debug dev : sendcontent free: 15872 chunk size:400
34716 : Debug dev : sendcontent free: 19192 chunk size:400
34719 : Debug dev : sendcontent free: 17792 chunk size:400
34723 : Debug dev : sendcontent free: 17816 chunk size:334
34724 : Debug dev : sendcontent free: 17816 chunk size:0
rules1.txt 0
rules2.txt 0
rules3.txt 0
rules4.txt 0
40529 : Debug dev : sendcontent free: 18504 chunk size:400
40534 : Debug dev : sendcontent free: 17128 chunk size:400
40536 : Debug dev : sendcontent free: 17128 chunk size:400
40539 : Debug dev : sendcontent free: 17128 chunk size:400
40542 : Debug dev : sendcontent free: 15728 chunk size:400
40546 : Debug dev : sendcontent free: 15728 chunk size:400
40548 : Debug dev : sendcontent free: 15728 chunk size:400
[... 49 lines cut ...]
47474 : Debug dev : sendcontent free: 15728 chunk size:400
47476 : Debug dev : sendcontent free: 15728 chunk size:400
47483 : Debug dev : sendcontent free: 16304 chunk size:400
[... cut ...]
62807 : Debug  : LoopStats: shortestLoop: 234 longestLoop: 1076756 avgLoopDuration: 2805.33 loopCounterMax: 128205 loopCounterLast: 10667
62808 : Debug  : Scheduler stats: (called/tasks/max_length/idle%) 10667/2098/7/92.47
66362 : Info   : EVENT: Clock#Time=Wed,13:13
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug Processing:rules2.txt
     flags CMI  parse output:
RuleDebug Processing:rules3.txt
     flags CMI  parse output:
RuleDebug Processing:rules4.txt
     flags CMI  parse output:

Grovkillen, 2021-01-20:
What uzi is suggesting is the log of the browser, not the ESP.

ghtester, 2021-01-20:
@TD-er  I believe it's easily reproducible if some of developers wish to find the root of the issue... Yes it's strange but not a serious bug from my point of view...

TD-er, 2021-01-20:
> @TD-er I believe it's easily reproducible if some of developers wish to find the root of the issue... Yes it's strange but not a serious bug from my point of view...

This one may be not too serious, but I'm afraid we may run into similar issues which remain unnoticed causing strange issues like corrupted settings.
So I really would like to know why this is happening before declaring its seriousness.

ghtester, 2021-01-20:
I agree. Fortunately this issue is easily reproducible, hopefully somebody else can confirm it...

ghtester, 2021-01-20:
Yeah, I already know. see the comments above... :-) Nothing to share...

ghtester, 2021-01-20:
> "testing" and "development" both include the "testing" part, which does leave out some parts.

Well, it looks it affects rtttl command... Don't want to put an offtopic here but just encountered fresh builds I tried (except Normal) returns Failed when I try to run it, for instance  rtttl,15:d=10,o=6,b=200,c,g,e
Hopefully there's a chance to make the Vagrant build where the rtttl is working?

TD-er, 2021-01-20:
Yep, the "custom" build is not trying hard to limit the build size.
You may need to define `USE_RTTTL` for this.

uzi18, 2021-01-20:
need to find where are my toys - esp(ecially)32 :)

ghtester, 2021-01-20:
@TD-er Thanks a lot, putting #define USE_RTTTL to Custom.h and rebuild enabled the rtttl command working again.

Now back to topic issue - it looks when I select any (but not empty) Rules Set and then another Rules Set, the POST is invoked and visible in browser's Address line. Perhaps there's a POST size limit somewhere... It's strange the issue does not happen when saving a very long Rule Set but just when the another Rules Set is selected (after the long one)...

Grovkillen, 2021-01-20:
That POST must be some leftovers from how it used to be because we now
create the file in the browser memory and upload it just like you would
upload any file manually to the unit. So perhaps that's the issue here...
Thanks for the clue.

ghtester, 2021-05-09:
> After opening the (any) Rule Set containing the rule longer than about 1800 characters and then open another Rule Set. Then the logon window appears but I can't log on properly and in browser's address line the rule content is displayed behind the node IP address.
> Just right after opening the next / another Rule Set, without any save attempt.
> Rules even longer than 2048 can be saved properly. But as soon as I save a rule longer than 1700-1800, I am starting to have the issue. So the "fix" is to save a very long rule always to Rule Set 2-4 because the Rule Set 1 is opened by default when the Rules menu is accessed and it makes the trouble (so only Rule Set 1 can be used in this case).

Unfortunately this strange issue with Rules pages is still here.
<html>
<body>

Firmware
--
Build:⋄ | 20113 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: May 7 2021 20:50:54
Plugin Count:⋄ | 35
Build Origin: | Vagrant
Build Time:⋄ | May 7 2021 20:49:56
Binary Filename:⋄ | ESP_Easy_mega_20210507_custom_IR_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega_52fb53a

</body>
</html>

The behavior is a bit different (logon form does not appear) but in general Rules pages 2 - 4 can't be accessed when the Rules Set 1 is non-empty and exceeds some size limit.

URI: 
Method: GET
Arguments: 0

TD-er, 2021-08-12:
This finally has been solved today and the code has been merged.
So I will close this issue.

PR #3138 (2020-06-28): [Rules] Fix rules handling of set 2...4 when using admin login (#3113)
With loading the page directly from the URL pointing to the rules set, including login, you have a special use case.

- You have web arguments (the rules set)
- You reload the page with the same rules set number as the last time you tried to load it.

This combined with the bug that we still had code present to support rules upload via POST would delete the rules.

Fixes #3113
commit db94b4c6 (2020-06-28): [Rules] Fix rules handling of set 2...4 when using admin login (#3113)

Fixes #3113
closed 2021-08-12 by TD-er