letscontrolit/ESPEasy #3114: Rotary Value can't be changed via taskvalueset

stwa2020, 2020-06-13:
I have used  mega-20200608 and Rotary Switch HW 040 

**Rotary    <->    ESP8266**
GND    <->    GND
"+"       <->   +3.3V
  DT       <->    D7   
CLK     <->    D6

The Rotary Counter can be changed by moving the Rotary to the left or to the right, no problem.

The Value of the Rotary Counter in the Device list can be set to "0" with the following command, 
http://192.168.1.163/?cmd=taskvalueset,1,1,0 

**Problem:**
The Counter Value can be set to "0" but when the Rotary Encoder is moved it will start with the number which was used before it was set to "0"

All tests and screenshots can be found 
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=7738

BR, Stefan

TD-er, 2020-06-13:
I marked it as a bug, since it is indeed counter-intuitive behavior.
But actually it is more like a (very logical) feature request.
`TaskValueSet` was original meant to be used with a Dummy plugin, but it makes sense to have it working on this plugin as well.

Have you tried the command `encwrite` ?
I just looked in the code and this command only accepts a single parameter, so this probably only works on the 1st encoder.
`encwrite,0` will set the counter value to 0.

stwa2020, 2020-06-13:
The _encwrite,0_ will set the value to 0 but this will not change the value in the Device list. I have to use first the _TaskValueSet 1,1,0_ to set the counter in the Device list to 0 and afterwards the _encwrite,0_ to set the value of the counter to 0. Now Device list is 0 and the Counter start from 0.

TD-er, 2020-06-13:
The `encwrite` command does indeed not update the values in the `UserVar` array.
That's for sure something that can be improved.

tonhuisman, 2023-09-02:
This has been changed in code, AFAICS, so can be closed.

closed 2023-09-02 by TD-er