letscontrolit/ESPEasy #3117: P045 - MPU6050 - Calculation of angle

Wiki591, 2020-06-15:
The calculation of MPU6050s angle depending on the accreleration of the axis need some trigonomic calculations which are hard to do using rules.

Is it possible to include the calculation of the angles inside the plugin?

TD-er, 2020-06-15:
Can you give some examples of angles you wish to be computed?
For example XZ or YZ or also the angle Z makes with the XY plane?
I can imagine the placement of the node may not always be perfectly aligned with one of the sensor axis perpendicular to the axis that can be changed.
So then you would probably want to have some rotation defined first.
For example:
- Set rotation of XY plane (or 3D rotation matrix)
- Compute angle of XZ relative to that rotation
- Compute angle of YZ relative to that rotation
- Compute angle of XYZ relative to that rotation

Wiki591, 2020-06-15:
Eeehm, this is too academic for my small, old mind. To be precise and frankly: no idea what you are talking about.

I may be wrong but I don't think that ESPEasy will ever be used for robots or drones.

My idea is to get just a plain angle of one axis, e.g. to get the elevation of a satellite dish (what I want to do in my RV-project) using the X-axis. My small project showed me: the sensor never is in static mode, there are constantly some different readings and it is almost impossible to place the sensor in zero-position. So there has to be a kind of calibration / adoption / averaging to the project. Means: one point defined as 0° (horizontal position), and from there on the angle of one axis.

TD-er, 2020-06-15:
Well, just to explain it more in real-life examples.
Let's assume you have a motor home (purely hypothetical ;) )
You will park the home and then have to level the unit.
I assume you can crank up some supports on each corner of the vehicle.

But even then, what do you consider a rotation axis?  (X-axis or Y-axis)
Is the X-axis from left to right and the Y-axis from back to front?
Or maybe in this (hypothetical) use case, it would make sense to have one axis from left-back to right-front and the other from left-front to right-back.
With the axis placed "diagonally" across the floor of the RV, you can just look at the tilt angle of a single rotation axis and just focus on the 2 cranks associated with that rotation axis to make sure the Z-direction points to what you consider to be level.

That would make it a lot easier to operate.
However, that would require you to mount the sensor also in this specific orientation, which may not be practical at all.

So what I could do in software is just mathematically rotate the axis of the sensor so their output resembles the most practical rotation axis.
This way you can mount the sensor any way you like in the RV and you can still get 2 angles to help you adjust the cranks in each corner of the RV to level the floor.

In mathematics you rotate these coordinates simply by using a rotation matrix, which is probably next to impossible to do in rules, but quite do-able in C++ code.

As a matter of fact, it would even allow to have these virtual axis (from corner to corner of your RV) just any line. So they don't have to be perpendicular to each other. 
After all a RV is not a square so the lines connecting opposite corners are not crossing each other at a 90 degree angle.

But like I said, it is just a hypothetical example :)

Wiki591, 2020-06-15:
OK, this way you explained it, at least I am able to understand what you were talking about - I am a complete noob concerning this kind of trigonometrie and sensors, I am am 1D, not 2D nor 3D. I will book this to my account of education. Thank you, really.

In a (theoretical) application to determine the levelling of (only in theory) a RV there would be no need to go that deep into the detail because you could have (theoretical) a bubble level as relation value for the desired levelling. The sensor could be fixed in almost x/y correct direction, the rest could be adjusted using correction values (I think, someone has done this already successfully somewhere this way).

What I am talking about would be a sensor, mounted in _almost_ correct direction concerning the x/y axis, one of the axis more or less in 100% relation to the elevation direction expected. So the moved object would influence jut one axis by being moved. The readings should only be done by one direction, either x or y. The resulting values should be comparable with freely accessible information or data, e.g. recommended elevation of a satellite dish in relation to your geographic position. There are several apps which are telling you depending on your actual geo position the elevation and direction for yout satellite dish. But how the hell you get the actual elevation of the dish, if you are standing inside of the RV trying to adjust it by moving the antenna mast up or down (one usual system) or by turning a crank handle right / left (another usual system). Would be helpful to know at least the elevation, so you only have to turn the mast to the correct direction until you get a signal.

Eeehm, did I explain, what I mean?

TD-er, 2020-06-15:
For dish orientation you probably only need to have the angle between the Z-axis and the horizontal plane (XY-plane)
The direction (North/South/East/West) could be done using a compass (sensor)
If it is sensitive enough, you could even do it automatically :) 
N.B. a compass sensor often gets "dirty" (by lack of a better term) in which is looses sense of direction due to external influences.
On a mobile phone you can help to re-calibrate the sensor by making 8-turns using your phone.
But that may be a bit difficult with an RV :)

Wiki591, 2020-06-15:
North/South... doesn't matter for me, only the elevation of the dish. Compass is an absolute nogo in my RV, I am using too much magnets for fixing whatever, but isn't needed because there is always at least an idea of the direction having a look in the sky. No smartphone, not even an old magnetic compass was able up to now to get a direction out of inside my RV. I suppose, most of the RVers made the same experience, that the compass doesn't work inside their homes, too many I have seen with their smartphones outside the vehicle standing in the pouring rain.

I don't exactly know what the z-axis is ment for, I have never seen an z-axis on my living room table. I want to calibrate the sensor once to 0° referring to the surface of the planet, and to forget this at once, and to get the elevation of the dish while turning the crank with a sensor more or less accurate attached on its back referring to the 0°. No xy plate, no roll, no pitch, no nothing. Just an angle. Thats what the whole bunch of tools are referring to - uselessly for everyone except the one who gets a reading of the dish elevation.

Wiki591, 2020-06-15:
Oh, probably I should explain: once you've got the correct elevation of the dish its a matter of seconds to direct the dish into the right direction using a satellite finder. This tiny small thing gives an alarm, if it receives the proper signal - correct elevation as a prerequisit.

Wiki591, 2020-06-16:
Just an example, not used by me:

[image]

Nice, but useless. Nobody knows the elevation of his dish except the one who is sitting on the roof of his house and has a look at the markings of his sat dish.

Imaging you are standing inside the RV, the mast in your hand, you are able to turn the mast, you have an idea where 316.6° roughly is (still: there is the sun to get an orientation....), you know, turning the mast will force the satfinder to beep if there is a signal, but you havn't any idea which elevation yout dish has - you're lost. Only painful experiences lets you know, turning the dish up to maximum and returning it 2 1/2 rounds down is roughly the elevation you need (but still an estimation), turning the mast from left to right and right to left, altering the elevation a lilltle bit up, turning again, altering a little bit down, turning again..... its a pain in the a**, believe me. But its the way RVers do it manually, believe, too.

Calibrating for me would work as follows: the named elevation may have nothing in common with the angles needed for your dish but getting the elevation out of one of these useless apps, fixing the dish manually to the correct elevation (will take some time, as always when you moved the RV), take the reading of the angle the MPU-plugin tells you and insert the corresponding correction factor/value into the rule you are using or - most desireable - into the plugin configuration. Going this way you will have a reference for other geographical locations, then using the up to now useless apps.

I hope I explained what I am talking about. If this could be implemented into the plugin - or realized in any other way, I'm open for solutions. If implemented however and wherever I would be pleased to write a detailed documentation how to use it and it could be a breakthrough in the RV-scene. Systems which are doing this automatically are far >1.000€. It is paid for to avoid the painful process of finding the desired satellite but there are a lot of RVers like me with old, manual systems and without money. And you know, I'm talking about the most worthy thing: the WAF.

TD-er, 2020-06-16:
Well just to explain what the Z-axis is.
Consider the width of your (leveled) table the X-direction.
The length of the table is the Y-direction.  (with a round or square table you may chose for yourself :) )
So the table-top is the XY-plane.
Now put a bottle of beer on the table such that the beer doesn't fall out.
Then the opening of that bottle is pointing in the Z-direction.

So what you want to know is the angle between the XY-plane (the table) and the Z-direction.
This needs a command or button (or both) to tell the task "consider this to be level in the XY-plane" and the task should then compute a rotation matrix so the read angles can be converted into the orientation practical for your use case.
So even if the motor home is parked at an angle, you should be able to see at what elevation angle the dish has compared to the earth.

I think it is clear what you want.

Wiki591, 2020-06-16:
Oh wow, I think now after your third try I've got it. Thank you for your patience and for my first 3D-lesson and sorry for my stupidity. Handling of a beer bottle is kind of common for me.

Yes, Thats what I need.

TD-er, 2020-06-16:
The only stupid thing is not asking what you don't understand, so to me this does not appear stupid at all.

And to continue with the beer analogy...
Imagine you rotate the beer bottle in such a way that there will be some beer pouring out (keep a glass ready).
Then the table, the bottle and the pouring stream of beer form a triangle.
The angle you're looking for is the angle the beer bottle makes with the table.  :)

Now imagine the table is not level, but at an angle.
You still have to tilt the bottle at the same angle (compared to the horizon) to start pouring out some beer.
But the imaginary triangle between the pouring beer, the bottle and the table has changed and will change even further when you point your bottle in another direction.
That compensation has to be done in software for the sensor that will probably be mounted in such a way that it can be compared to the tilted table.
So what you want is regardless the direction of your bottle/satellite dish, you want to get the angle compared to the horizon and not to the table.
This also means that if the mounting pole is slightly tilted, you may need to correct the angle of the satellite dish a bit again if you orient it to the right north/east/etc. direction.
But with such a sensor that should be easy to do to make reception even better.

So first set it to the correct angle with the horizon, then rotate to the correct (north/east/etc) direction and then finalize the perfect dish orientation by tweaking the last angle offset compared to the horizon. (by using the reported angle)

Wiki591, 2020-06-16:
Yes, that is exactly want I want to do (and what I am doing now everytime I am parked for the night with my estimation of the needed elevation) and why I asked for the possibility of including the angle calculation into the plugin.

Wiki591, 2020-11-08:
Is there any "roadmap" when a solution might come?

Wiki591, 2021-01-28:
With the newly added trigonometric functions there is no need anymore to include angle calculation inside the plugin, now it is possible to do this using rules.

closed 2021-01-28 by Wiki591