letscontrolit/ESPEasy #3118: SDS011/018/198 keeps running when task disabled

azsystem, 2020-06-17:
### Checklist
- [ x] The title of this issue is "to the point" and descriptive.
- [ x] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

When the SDS task is disabled from rules, the SDS fan keeps running. Perhaps the laser is also running

### Expected behavior

When the SDS task is disabled, I would like the sensor to stop taking samples, and the fan should stop

### Actual behavior

The fan keeps running

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2020-06-17:
So the `taskdisable` should call the `PLUGIN_EXIT` function and
`taskenable` should call `PLUGIN_INIT`.

azsystem, 2020-06-17:
I think that would be a solution, but another solution would be to be able to call the PLUGIN_EXIT and PLUGIN_INIT from the rules

TD-er, 2020-06-17:
Those commands also toggle the enabled/disabled (or actually that's the only thing they toggle right now) state.
Calling PLUGIN_INIT will not work if the task is not set to enabled.

azsystem, 2020-06-17:
How do I call the  PLUGIN_INIT /  PLUGIN_EXIT command from rules?

TD-er, 2020-06-17:
You don't.
PR #3119  made by @tonhuisman should add it.

azsystem, 2020-06-18:
It would be very nice to have that feature!

Just a little side information:
I use the SDS011 to detect a nearby BBQ, so I can close the windows and shut down ventilation before the whole house smells. For that detection, I need data with a high sample rate. But I only need the data during the BBQ timeframe. Due to the limited lifespan of the SDS011 sensor, I don't want the sensor taking samples 24/7.

TD-er, 2020-06-18:
For that I have added the option to set the interval timer of the SDS011.
So it will operate on its own, starting/stopping autonomously.
But I never thought of the use case you mention :)

PR #3119 (2020-06-18): [Tasks] Call PLUGIN_INIT/PLUGIN_EXIT using TaskEnable/TaskDisable from rules
Resolves #3118
closed 2020-06-18 by TD-er