letscontrolit/ESPEasy #3121: [Rules] Modulo operator '%' didn't work

sebastien247, 2020-06-21:
Hi,
I try to use modulo operator '%' like this in my rules:

```
on DistributionCount#State do
  Let,1,%v1%+1
  if %v1%%2=0
    SendToHTTP 192.168.31.215,8080,/json.htm?type=command&param=udevice&idx=2&nvalue=0&svalue=%v1%
  endif
endon
```
or directly like this `4%2=0` but nothing appear

I use this build on Wemos D1 mini:

```
Build:⋄
20107  - Mega
System Libraries:⋄
ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20200608
Plugin Count:⋄
46 [Normal]
Build Time:⋄
Jun  8 2020 21:21:57
Binary Filename:⋄
ESP_Easy_mega_20200608_normal_ESP8266_4M1M
Build Platform:⋄
Linux-5.0.0-1031-gcp-x86_64-with-glibc2.2.5
Git HEAD:⋄
HEAD
```

Regards

TD-er, 2020-06-21:
The only place where we support modulo is in the cron plugin.
So I'll consider this a feature request. :)

sebastien247, 2020-06-21:
But on the release `Release mega-20181214` we can read that in the changelog:
`[Rules] Modulo operator '%' only works on int`
https://github.com/letscontrolit/ESPEasy/commit/9b1151baa676f30e5db4cfb4983b329d3950bf29

tonhuisman, 2020-06-22:
The `%` operator is both documented and in the code, so I'd expect it to work. Seems something is a bit broken here, and _my_ first suspect is the processing of %v1%
Does it help if you change the if statement like below? (sorry, no testing environment available, atm)
```
on DistributionCount#State do
  Let,1,%v1%+1
  if [int#1]%2=0
    SendToHTTP 192.168.31.215,8080,/json.htm?type=command&param=udevice&idx=2&nvalue=0&svalue=%v1%
  endif
endon
```

TD-er, 2020-06-22:
Maybe we should change it to a mod command?

sebastien247, 2020-06-22:
> The `%` operator is both documented and in the code, so I'd expect it to work. Seems something is a bit broken here, and _my_ first suspect is the processing of %v1%
> Does it help if you change the [...]

Didn't work too

tonhuisman, 2020-06-23:
I did some digging in the code, and the case here is that the if statement in rules only supports 'simple' values on either side of a conditional operator, not expressions.
The way to make your code work would be like this:
```
on DistributionCount#State do
  Let,1,%v1%+1
  Let,2,%v1%%2
  if %v2%=0
    SendToHTTP 192.168.31.215,8080,/json.htm?type=command&param=udevice&idx=2&nvalue=0&svalue=%v1%
  endif
endon
```
It might be useful to implement expression support in the if statement.

tonhuisman, 2023-09-02:
This seems to be resolved, so can be closed.

closed 2023-09-02 by TD-er