letscontrolit/ESPEasy #3123: Stack usage of  Generic HTTP Advanced is way too high

TD-er, 2020-06-22:
Controller 11 (Generic HTTP Advanced) uses quite a large struct to interact with the settings.
This struct is declared on the stack, sometimes needing upto 1200 bytes of stack when calling the function.

```c++

#define C011_HTTP_METHOD_MAX_LEN          16
#define C011_HTTP_URI_MAX_LEN             240
#define C011_HTTP_HEADER_MAX_LEN          256
#define C011_HTTP_BODY_MAX_LEN            512

struct C011_ConfigStruct
{
  void zero_last() {
    HttpMethod[C011_HTTP_METHOD_MAX_LEN - 1] = 0;
    HttpUri[C011_HTTP_URI_MAX_LEN - 1] = 0;
    HttpHeader[C011_HTTP_HEADER_MAX_LEN - 1] = 0;
    HttpBody[C011_HTTP_BODY_MAX_LEN - 1] = 0;
  }

  char          HttpMethod[C011_HTTP_METHOD_MAX_LEN] = {0};
  char          HttpUri[C011_HTTP_URI_MAX_LEN] = {0};
  char          HttpHeader[C011_HTTP_HEADER_MAX_LEN] = {0};
  char          HttpBody[C011_HTTP_BODY_MAX_LEN] = {0};
};
```

Related:
#2993 
#1536
#2227 (???)
#2301 (???)

pkoster, 2020-07-31:
Recent builds do not properly save HTTP URI, HTTP Header, HTTP Body. The fields stay empty and the plugin does not trigger any http calls on the network. Both IE and Chrome do indicate 'submitted'. Also tried clean nodemcu.

ESPEasy_ESP82xx_mega-20200703
ESPEasy_ESP82xx_mega-20200721
(ESPEasy_mega-20200328 did not suffer from this)

TD-er, 2020-08-01:
I will re-open this based on your reply.

TD-er, 2020-08-02:
Fixed it, as it was a very silly mistake.
When loading/saving settings of this controller, the `sizeof` a pointer was used, not the `sizeof` the structure that needs to be loaded or saved.

tonhuisman, 2023-09-02:
As the problem is already fixed, this issue can be closed.

PR #3134 (2020-06-27): [C011] Generic HTTP Advanced Reduce stack usage (#3123)
Fixes: #3123 

ControllerSettings and (large) struct for the controller config were active at the same time while it is not needed.

Also the host to send data to was connected twice, which may also lead to strange results.
commit 93ad4168 (2020-06-27): [C011] Generic HTTP Advanced Reduce stack usage (#3123)

Fixes: #3123 

ControllerSettings and (large) struct for the controller config were active at the same time while it is not needed.

Also the host to send data to was connected twice, which may also lead to strange results.
closed 2023-09-02 by TD-er