letscontrolit/ESPEasy #3135: cant update sonoff ch4 R2

userosos, 2020-06-26:
Hi everyone! I use sonoff ch4 R2 with old firmware:

Build | 20102  - Mega
-- | --
Libraries | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version | mega-20180606
Plugins | 47 [Normal]
Build Md5 | b44aa2c49badf0e2832b4dc7b98cbb
Md5 check | passed.
Build time | Jun  6 2018 02:13:33
Binary filename | ESP_Easy_mega-20180606_normal_ESP8285_1024.bin

Now i download latest release - ESPEasy_ESP82xx_mega-20200608
Unzipped from one - ESP_Easy_mega_20200608_normal_ESP8285_1M.bin
Goto web-face to Tools/Firmware/Load and select the ESP_Easy_mega_20200608_normal_ESP8285_1M.bin and press the Update button. After ~5 secs i see Update error: ERROR[4]: Not Enough Space 
What is this?

Logs
1149109: SYS  : 19.00
1149112: EVENT: uptime#Uptime=19.00
1176669: WD   : Uptime 20 ConnectFailures 0 FreeMem 15688
1195111: EVENT: Clock#Time=Sat,00:37
1206669: WD   : Uptime 20 ConnectFailures 0 FreeMem 15688
1209109: SYS  : 20.00
1209112: EVENT: uptime#Uptime=20.00
1236673: WD   : Uptime 21 ConnectFailures 0 FreeMem 15760
1255377: EVENT: Clock#Time=Sat,00:38
1266673: WD   : Uptime 21 ConnectFailures 0 FreeMem 15760
1269375: SYS  : 21.00
1269378: EVENT: uptime#Uptime=21.00
1296673: WD   : Uptime 22 ConnectFailures 0 FreeMem 15760
1315377: EVENT: Clock#Time=Sat,00:39
1326673: WD   : Uptime 22 ConnectFailures 0 FreeMem 15640
1329377: SYS  : 22.00
1329380: EVENT: uptime#Uptime=22.00

tonhuisman, 2020-06-27:
Your flash size is only 1 MB, and that's too small to do a single-step update, so you'll need to go the 2-step route:
First upload a .bin file with 'minimal' and 'OTA', and in your case also '8285', in the name, that should leave enough free space to upload, in the second step hence the 2-step name, the .bin you mentioned above.

userosos, 2020-06-27:
> Your flash size is only 1 MB, and that's too small to do a single-step update, so you'll need to go the 2-step route:
> First upload a .bin file with 'minimal' and 'OTA', and in your case also '8285', [...]
I used ESP_Easy_mega_20200608_minimal_core_242_ESP8285_1M_OTA.bin
And it not work too - Update error: ERROR[4]: Not Enough Space

tonhuisman, 2020-06-27:
Ah, my mistake, I've never used the 2-step upload myself, sorry.

You need the ESPEasy_2step_UploaderMega_1024.bin to be flashed first, then, after the reboot, upload the .bin file you need. The OTA I referred to in the name is the Arduino OTA feature that is available in those builds, not related to the ESPEasy upload feature.

userosos, 2020-06-27:
> Ah, my mistake, I've never used the 2-step upload myself, sorry.
>
> You need the **ESPEasy_2step_UploaderMega_1024.bin** to be flashed first, then, after the reboot, upload the .bin file you need. The [...]

Hmmm. Sorry but it not work too. Update error: ERROR[4]: Not Enough Space 
Where i can see more info about it maybe?

userosos, 2020-06-27:
Or i can do an old update. Witout the problem? My's firmware is very old now (2018 years).

tonhuisman, 2020-06-27:
You can of course upload a newer firmware by disconnecting the sonoff from power, and connect to your computer like you originally installed ESPEasy on it. General advice is to use the latest, as that's the greatest.

userosos, 2020-06-27:
> 
> 
> You can of course upload a newer firmware by disconnecting the sonoff from power, and connect to your computer like you originally installed ESPEasy on it. General advice is to use the latest, as that's the greatest.

For update will i need use UART everytime? it is not ok for me.

userosos, 2020-06-27:
May be can make an firmware has into only tools for upgrade the firmware? For upgrade only.

TD-er, 2020-06-27:
> May be can make an firmware has into only tools for upgrade the firmware? For upgrade only.

That's what the 2-step OTA bin file is made for.
So you flash that build, then you load a build with OTA in the name.

It could be that an old build didn't have the right check for the size, but recent builds do check with the size of the small 2-step OTA in mind.

Wiki591, 2020-06-28:
No, Sirs, you are all wrong. Reading before posting helps sometimes. Just have a look at what userosos has documented in his first post:
> 
> 
> [....]
> ESP_Easy_mega-20180606_normal_ESP8285_1024.bin
> [....]

@userosos: you will never be able to update this device using OTA. The devil is in detail:

The "normal" build is too big to allow even a 2-step-OTA-upgrade. If you want to be able to upgrade a 1M device like the Sonoff CH4 R2 you have to use the _minimal_ builds. I'm sorry, but you have to flash the device once again using cables. This is pretty well documented in the wiki of ESPEasy. And don't blame the developers, ESPEAsy is too mighty to fit into every flash size. Works as designed - and as documented.

userosos, 2020-06-28:
> No, Sirs, you are all wrong. Reading before posting helps sometimes. Just have a look at what userosos has documented in his first post:
>
>
> @userosos: you will never be able to update this device [...]

Ok. Thx! What different between minimal_ builds and normal builds?

TD-er, 2020-06-28:
Some features have been disabled and there is only a minimal set of controllers and plugins included.

userosos, 2020-06-28:
> 
> 
> Some features have been disabled and there is only a minimal set of controllers and plugins included.
> 
> See:
> 
> https://github.com/letscontrolit/ESPEasy/blob/ce317f7207f98eb148322bb50114abf40b2ce141/src/define_plugin_sets.h#L248-L355

Thx! Now i flashed it! But i has some problem. If i use the reulse from old firmware the LEDs is blink and relays on/off too everytime ~1 Hz. And if i delete the rules LEDs 2 and 3 is on everytime. Reboot not fix it. If i pressed the buttons into the sonoff the LEDs not triggired on it.
[rules1.txt](https://github.com/letscontrolit/ESPEasy/files/4842210/rules1.txt)
[image]
[image]
[image]

Wiki591, 2020-06-28:
For me the configuration lokks a little bit mixed up. I do have a Sonoff 1CH device which I had to configure in another way.
You'll find  configuration instructions for the 4CH including rules [here](https://smarthome.kuklin.de/2019/04/25/sonoff-4ch-pro-mit-espeasy-firmware-in-homematic-system-einbinden/)
I'm sorry, its in German but the pictures you need to follow the config are self explaining, I think.

userosos, 2020-06-28:
> For me the configuration lokks a little bit mixed up. I do have a Sonoff 1CH device which I had to configure in another way.
> You'll find configuration instructions for the 4CH including rules [...]

Good! Thx!
I have sonoff ch4 r2 . Have it different between GPIO?
[image]
[image]
[image]

Wiki591, 2020-06-29:
Not as far as I know, should be no difference

berbergh, 2022-07-03:
This is of course an ancient issue, but still relevant to me, as I have this device on a very remote place.
Up to some years ago, it was possible to remotely update the Sonoff, but I have not seen an OTA file for a long time.
Does this mean, that I have to effectively phase out the Sonoff?

tonhuisman, 2022-07-03:
> Does this mean, that I have to effectively phase out the Sonoff?

Depending on what your intent is here, you either keep using it as-is, or update to the latest ESPEasy version via the serial route. (disconnect from mains first!) Updates via OTA, both ESPEasy OTA and Arduino OTA, are not currently possible for the 1MB Flash units.
But if it works as intended, why would you want to change anything? What new feature of ESPEasy do you need on your device?

berbergh, 2022-07-03:
Thank you for your reply.
It works as intended, but I have learned to allways update to latest versions. Not only for new goodies, but also for bugs.
As fas as I can see, the OTA version was removed from the repository with previous notice.
In the mean time I have been asking for future plans regarding the Sonoff once in a while. I never got a clear answer.

TD-er, 2022-07-03:
The update button was probably disabled because the currently installed firmware version was larger than 600k.
That's the largest bin size which allows for 2-step OTA as the 2-step OTA bin + 600k + filesystem size = 1M
If you would update the 2-step OTA anyway, it would destruct the file system and thus all settings.