letscontrolit/ESPEasy #3141: UTC Time settings

ingoiot, 2020-07-02:
Hi,

this issue refers to this [discussin in the forum:](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=7828)

it seems like the UTC+- setting is only activated when DST is active.
Thats a bit confusing especially when I dont want to use DST. Maybe its better placed under "Location Settings".

Maybe one could merge "NTP Settings" and "DST Setting" to "Time settings", and add there the UTC settings too.

TD-er, 2020-07-20:
I finally came to test this issue.
But I cannot find anywhere in the code where it may be wrong.

```c++
void ESPEasy_time_zone::applyTimeZone(uint32_t curTime) {
  int dst_offset = Settings.DST ? 60 : 0;
```

This offset is used to compute the start of the DST (CEST, typically the last Sunday of March)
This function to apply the time zone is also called regardless the state of the DST checkmark.

I also tested it here with the timezone offset of 60 and both states of the DST checkmark appear to give me the expected time.

So unless you can prove me otherwise, it does not seem to be a bug.

ingoiot, 2020-07-20:
seems like i got a bit confused,
it works no matter if ticked or not.
only the placing under dst settings seems a bit weird.
sry for the effort.

TD-er, 2020-07-20:
I did move the timezone to the location settings:
[image]

closed 2020-07-20 by TD-er

datenheim, 2023-06-14:
Just a note to others:
changing the Timezone Offset and DST settings are applied after a system reset, not immidiately.
Tested on several mega-20230508 builds.

TD-er, 2023-06-14:
Ehh nope, just tested it on a node here.
I did update the DST and timezone offset and on the next reload of either the main page or sysinfo page showed immediately the new 'local time'.

TeeTrizZz, 2023-08-26:
Hey TD-er! I too had to restart my Wemos D1 Mini (mega-20230822) after changing the Timezone Offset setting. I initially assumed that this setting was not working, as the main page still showed the old local time. After reading datenheim's comment and restarting my board, it finally worked.

TD-er, 2023-08-26:
That's really strange.
I will have a look at it.
Maybe the timezone parameters should be recalculated whenever we also clear the cache as it is essentially a cache to keep this calculated.

PR #3148 (2020-07-20): [WiFi] Reduce wifi reset calls and call full wifi init at reset (#3147)
Fixes: #2931
Fixes: #1258
Fixes: #3155
Fixes: #3141
Fixes: #3111 

Connected state is now based on the state reported by the internal WiFi object.
Frequent calls to `reset()` of WiFi are now rate limited to make sure we don't end up in a loop.