letscontrolit/ESPEasy #3146: P095 plugin Display TFT - ILI9341 missing ?

llevet, 2020-07-03:
I try to use ILI9341 but don't find it in any release.

I have enable SPI Interface in "Hardware".
But plugin is not present in "Devices".

I try : 

ESP_Easy_mega_20200516_custom_ESP8266_4M1M
ESP_Easy_mega_20200516_normal_ESP8266_4M1M
ESP_Easy_mega_20200516_test_beta_ESP8266_4M1M
ESP_Easy_mega_20200516_dev_ESP8266_4M1M

ESP_Easy_mega_20200608_custom_ESP8266_4M1M
ESP_Easy_mega_20200608_normal_ESP8266_4M1M
ESP_Easy_mega_20200608_test_beta_ESP8266_4M1M
ESP_Easy_mega_20200608_dev_ESP8266_4M1M

Thanks.

Ludovic.

TD-er, 2020-07-03:
Indeed, it is now not included in a build.
See [search for "USES_P095"](https://github.com/letscontrolit/ESPEasy/search?q=USES_P095&unscoped_q=USES_P095)

Can you make a build on your own?

llevet, 2020-07-03:
I did not know that all plugin was not on some build.

Maybe adding a column in this doc ( https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html ) to reflect the status of each plugin: 
Present in build : Normal / Test / Beta /  Dev / Custom / Minimal / None.

I will take a look how to build it by myself.

Thanks.

TD-er, 2020-07-03:
Well as you may have seen in one of the sticky topics: https://github.com/letscontrolit/ESPEasy/issues/2590

It is really needed to have a good look at how to re-distribute the plugins over builds

llevet, 2020-07-04:
Ok.
Build by myself.
Pull requests #3150 and #3151 make it usable without problem.
Working on AZ-Touch ESP (https://www.hwhardsoft.de/english/projects/arduitouch-esp/)
with NodeMCU Amica.

eXrshock, 2021-02-19:
hi, i am currently testing this with an esp32 with the AZ-Touch ESP.

Does not work the display remains dark.

I also think that the adafruit lib is missing in the lib folder?
I use ESP_Easy_mega_20210114_display_ESP32_4M316k-factory.bin

tonhuisman, 2021-02-19:
Have you powered the backlight? (Can normally be connected to VCC with a wire)

tonhuisman, 2021-02-19:
> I also think that the adafruit lib is missing in the lib folder?

The PlatformIO build tools download libraries on demand. The libraries that _are_ included in the source usually have either adjustments that prevented it from working as intended/required 'out of the box', or have improvements applied that are not available in the publicly available library (and not every maintainer accepts pull requests).

llevet, 2021-02-19:
To enable led backlight you must put Pin mode GPIO-15 (D8) to "Output High" in Hardware tab.

Take a look on schematic : 
https://www.hwhardsoft.de/app/download/11868165697/AZ-Touch+MOD+schematic+V01-03-01.pdf

llevet, 2021-02-19:
you can also put this in : Rules Set 1 
on System#Boot do
  gpio,15,1
  tftcmd,clear,black
  tftcmd,on
endon

eXrshock, 2021-02-19:
> To enable led backlight you must put Pin mode GPIO-15 (D8) to "Output High" in Hardware tab.
> 

Unfortunately, I didn't work, I'll build everything again on a breadboard.

> To test :
> you can also put this in : Rules Set 1
> 
> on System#Boot do
> gpio,15,1
> tftcmd,clear,black
> tftcmd,on
> endon
> on Clock#Time=All,**:** do
> tft,txtfull,100,1,2,%sysday_0%/%sysmonth_0%/%sysyear_0%
> tft,txtfull,115,30,3,%syshour_0%:%sysmin_0%
> endon

It does not work.
The problem is probably with the hardware.
Many thanks for your help

Hardy4711, 2021-03-05:
Set on  AZ-Touch Mod Version 01-03-01  GPI015 to LOW:

on System#Boot do
gpio,15,**0**
tftcmd,clear,black
tftcmd,on
endon

llevet, 2021-03-05:
Yes and no ...
If you use ESP32 ok, gpio 15 must be low to have backlight (via PNP transistor).
For wemos or MCU, the gpio 15 must be in the high state (direct connection to the screen / do not cross the PNP transistor).
And this for all PCB versions. 

In your case, you use a ESP32,  so the code of Hardy is correct.

closed 2022-11-09 by TD-er