letscontrolit/ESPEasy #3147: Rebooting  mega_20200703 via webpage or serial gives rolling reboot

2bontb, 2020-07-04:
This morning I loaded (OTA) ESP_Easy_mega_20200703_normal_ESP8266_4M1M on a Wemos, and all seems OK.
Previous release was : mega-20200426
But if I manual reboot the Mega, it starts a rolling reboot.
If I powercycle the wemos, the system comes back up.

There is the serial log:
61886 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 20952 WiFiStatus 3
91886 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 20952 WiFiStatus 3
110595 : Info  :  Webserver args: 0: 'index' length: 1 1: 'page' length: 1
121885 : Info  : WD   : Uptime 2 ConnectFailures 0 FreeMem 20888 WiFiStatus 3
142814 : Info  :  Webserver args: 0: 'cmd' length: 6
142816 : Info  :      : Rebooting...

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
⸮U71 : Info  : 

INIT : Booting version: mega-20200703 (ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
72 : Info  : INIT : Free RAM:33360
74 : Info  : INIT : Warm boot #1 Last Task: Const Interval timer, id: 3 Last systime: 1593861459 - Restart Reason: Software/System restart
75 : Info  : FS   : Mounting...
100 : Info  : FS   : Mount successful, used 75802 bytes of 957314
117 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
125 : Info  : CRC  : SecuritySettings CRC   ...OK 
236 : Info  : INIT : Free RAM:30592
238 : Info  : INIT : I2C
239 : Info  : INIT : SPI not enabled
321 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
426 : Info  : WIFI : Set WiFi to STA
531 : Info  : WIFI : Connecting IOTdown attempt #0
537 : Info  : Webserver: start
539 : Info  : Time set to 1593861459.000
543 : Info  : Current Time Zone:  DST time start: 2020-03-29 02:00:00 offset: 120 min STD time start: 2020-10-25 03:00:00 offset: 60 min
548 : Info  : Local time: 2020-07-04 13:17:39
653 : Info  : WIFI : Connecting IOTdown attempt #1
1759 : Info  : WIFI : Connected! AP: IOTdown (76:83:C2:2D:9E:EB) Ch: 1 Duration: 26 ms
1760 : Info  : WIFI : DHCP IP: 10.2.1.240 (SensorFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
1777 : Info  : WIFI  : WiFiConnected() out of sync
1778 : Info  : WIFI  : WiFiConnected() out of sync
1778 : Info  : WIFI  : WiFiConnected() out of sync
1778 : Info  : WIFI  : WiFiConnected() o
Exception (28):
epc1=0x4026de70 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000b depc=0x00000000

If you need the stack let me know.
It is reproducible!

TD-er, 2020-07-04:
Can you see using the (Python) flash tool what flash chip you have?

2bontb, 2020-07-04:
One note. I just rebooted another MCU and same problem on this on on release mega-20200426
Never noticed before.

The flash id of the MCU is on mega-20200426:
esptool --port COM5 flash_id
esptool.py v2.8
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:87:53:a1
Uploading stub...
Running stub...
Stub running...
Manufacturer: e0
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

The flash on the mega-20200703 of the Wemos 
esptool --port COM4 flash_id
esptool.py v2.8
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: dc:4f:22:11:28:6e
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

TD-er, 2020-07-04:
Manufacturer 0x20 is XMC.

But since you also have it on the other node, it may have something to do with the rather frequent WiFi reset calls.

I will make a small change and make a test build for you. It will be ready soon.

TD-er, 2020-07-04:
Hmm it will take a bit more time.
My intended fix made it a lot easier to reproduce these crashes...

TD-er, 2020-07-04:
Can you try this test build: [ESP_Easy_mega_20200704_normal_ESP8266_4M1M.bin](https://www.dropbox.com/s/mw7rdtb8uy1ak02/ESP_Easy_mega_20200704_normal_ESP8266_4M1M.bin?dl=0)

2bontb, 2020-07-04:
TD-er,
This fix works for my Wemos (see below) Did not try it on the MCU, assuming  it will be ok.
But I observed someting with this build. However I did not check it on the old builds. 
When setting the loglevel to info, the DHCP info lines comes very often!! Let me know if I sould make seperate ticket on this.
Thanks for the fast fix. I assume in one of the next builds it will be available?

62730 : Info  : Command: reboot

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
⸮U72 : Info  : 

INIT : Booting version:  (ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
73 : Info  : INIT : Free RAM:33536
73 : Info  : INIT : Warm boot #2 Last Task: Background Task Last systime: 1593873446 - Restart Reason: Software/System restart
74 : Info  : FS   : Mounting...
100 : Info  : FS   : Mount successful, used 75802 bytes of 957314
117 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
124 : Info  : CRC  : SecuritySettings CRC   ...OK 
236 : Info  : INIT : Free RAM:30752
238 : Info  : INIT : I2C
239 : Info  : INIT : SPI not enabled
323 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
429 : Info  : WIFI : Set WiFi to STA
533 : Info  : WIFI : Connecting IOTdown attempt #0
540 : Info  : Webserver: start
542 : Info  : Time set to 1593873446.000
546 : Info  : Current Time Zone:  DST time start: 2020-03-29 02:00:00 offset: 120 min STD time start: 2020-10-25 03:00:00 offset: 60 min
551 : Info  : Local time: 2020-07-04 16:37:26
656 : Info  : WIFI : Connecting IOTdown attempt #1
1661 : Info  : WIFI : Connected! AP: IOTdown (76:83:C2:2D:9E:EB) Ch: 1 Duration: 24 ms
1662 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
1682 : Info  : SYS  : 0.00,-47.00,100.00
1786 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
1901 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2016 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2142 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2256 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2371 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2383 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 22376 WiFiStatus 6
2486 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2601 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2716 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2830 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
2956 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3071 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3186 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3300 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3414 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3529 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3644 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3770 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms
3884 : Info  : WIFI : DHCP IP: 10.2.1.120 (SensorWemosFree1-120) GW: 10.2.1.138 SN: 255.255.255.0   duration: 27 ms

PR #3163 (2020-07-09): [Syslog] Fix crash with syslog enabled on wifi (re)connect (#3147 #3072)
Fixes: #3147
Fixes: #3072

It appeared the log handling function was sometimes processing flash strings or progmem strings and thus crashing with an exception.

Crashes like these:
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

0x40275929 in String::operator=(char const*) at ??:?
0x4022cab9 in process_serialWriteBuffer() at ??:?
0x4023532b in addToSerialBuffer(char const*) at ??:?
0x402756a0 in String::copy(char const*, unsigned int) at ??:?
0x40238bd3 in addToLog(unsigned char, char const*) at ??:?
0x40275cac in Stri [... cut]
closed 2020-07-09 by TD-er